evcc-io/evcc #16142: Add GoodWe EM

andig, 2024-09-15:
In diesem Internet:

https://downloads.vodnici.net/uploads/wpforo/attachments/483/4349-1GoodweModbusPROTOCOLHybrid-ENV1-3EMseries.pdf
https://forum.iobroker.net/assets/uploads/files/1666333836822-goodwe_modbus_protocol_hybrid_et_eh_bh_bt__arm205-hv__v1.7-_-read-only_20200226-1.pdf

_Originally posted by @pohlinkzei in https://github.com/evcc-io/evcc/discussions/16028#discussioncomment-10653064_

andig, 2024-09-16:
@pohlinkzei mir fehl aktuell die Kapazität. Wenn Du es schaffst, aus https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/goodwe-hybrid.yaml oder https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/goodwe-dt.yaml einen Vorschlag mit neuen Registern zu erstellen (jeweils für Grid/ PV/ Battery) würde das den Prozess deutlich beschleunigen.

pohlinkzei, 2024-09-16:
Ich bin gerade dabei mich in das Thema einzuarbeiten, hab aber (noch) keine Ahnung von Netzwerkzeugs. Mein EM-Wechselrichter hat auf erste Annäherungsversuche per RS485/UDP@Wifi nicht brauchbar reagiert.
Die Implementation vom HA für den EM habe ich auch gefunden, das habe ich aber noch nicht nachstellen können.

https://github.com/marcelblijleven/goodwe/blob/master/goodwe/es.py

andig, 2024-09-18:
Du musst einfach nur ein `custom` meter konfigurieren. Gebaut werden muss da nix.

pohlinkzei, 2024-09-18:
Das hatte ich erfolglos versucht. In der Implementierung zum EM-WR stand außerdem, dass die Kommunikation per WiFI (?) mit einem 0xAA 0x55 0xC0 0xF7 Header läuft umd UDP (?) ähnlicher zu werden... Ich habe keine Ahnung von TCP/UDP und so. Ich bin eher der PHY/HAL-Typ. Aber hier mal mein custom-Versuch:
`- type: custom
  name: GoodWe-EM-PV
  usage: pv
  power:
    source: calc
	add:
	  source: calc
	  mul:
      - source: modbus
	    id: 247
	    uri: 192.168.178.66:8899
	    timeout: 3s
	    register:
	      address: 1280
	      type: holding
	      decode: uint16
	  - source: modbus
	    id: 247
	    uri: 192.168.178.66:8899
	    timeout: 3s
	    register:
	      address: 1281
	      type: holding
	      decode: uint16
	  source: calc
	  mul:
      - source: modbus
	    id: 247
	    uri: 192.168.178.66:8899
	    timeout: 3s
	    register:
	      address: 1283
	      type: holding
	      decode: uint16
	  - source: modbus
	    id: 247
	    uri: 192.168.178.66:8899
	    timeout: 3s
	    register:
	      address: 1284
	      type: holding
	      decode: uint16
  
- type: custom
  name: GoodWe-EM-BAT
  usage: battery
  power:
	source: calc
	mul:
    - source: modbus
	  id: 247
	  uri: 192.168.178.66:8899
	  timeout: 3s
	  register:
	    address: 1286
	    type: holding
	    decode: uint16
	- source: modbus
	  id: 247
	  uri: 192.168.178.66:8899
	  timeout: 3s
	  register:
	    address: 1290
	    type: holding
	    decode: uint16
	  
  soc: # Batterie SOC (%)
    source: modbus
	id: 247
	uri: 192.168.178.66:8899
	timeout: 3s
	register:
	  address: 1294
	  type: holding
	  decode: uint16`

andig, 2024-09-18:
Und funktoinierts? Die Einrückung brauchts...

pohlinkzei, 2024-09-19:
Leider nicht. Wie geschrieben, ich glaube es braucht einen anderen Header für die UDP-Modbus-Übertragung, wie hier (https://github.com/marcelblijleven/goodwe/blob/master/goodwe/protocol.py#L469-L479) beschrieben. Bei Homeassistant scheint der WR ja zu funktionieren.

andig, 2024-09-19:
So wird das nicht. Deine Config ist unvollständig. Ich mache hier mal zu bis es eine vollständige Config gibt die überhaupt testfähig wäre.

closed 2024-09-19 by andig