evcc-io/evcc #16148: Improve visualization when In&Out doesn't match

maf-soft, 2024-09-16:
[image]

Wie man sieht, passt hier was nicht zusammen - obwohl nur 1,3 kW produziert werden und fast kein Netzbezug stattfindet, wird mit 3,8 kW das Auto geladen und auch noch etwas die Hausbatterie. Das liegt daran, dass mir die Daten der 2. PV fehlen, also die Produktion ist falsch. Das sollte ich natürlich ändern, aber es funktioniert - der Überschuss landet im Auto, wie er soll, und alle Werte werden so dargestellt, wie evcc sie bekommt. Nur die Skalierung der Balken ließe sich doch verbessern - oben müsste eine "unbekannte Produktion" hinzukommen, vielleicht dargestellt mit Sonne+Fragezeichen, damit die Auto-Klammer richtig skaliert werden kann.

Wenn tatsächlich gerade etwas eingespeist wird, sieht es so aus:
[image]

maf-soft, 2024-09-16:
Hintergedanke ist auch, dass man m.E. im UI direkt sehen sollte wenn die Werte nicht zusammenpassen, anstatt dass auf den ersten Blick alles korrekt aussieht, wie in meinem ersten Screenshot. Dass was nicht stimmt, sieht man ja erst wenn man die Werte vergleicht.

sicheste, 2024-09-17:
> Woher willst du denn wissen, welcher der Messwerte richtig oder falsch ist? Wie sieht der Algorithmus den Du zur Korrektur vorschlägst der alle Messwerte ausgehend vom API berücksichtigt?

Die Frage an sich ist berechtigt. Aber kennst du ein Setup, bei dem jemand zwei parallel geschaltete Anschlüsse ans Netz hat? Der grid meter, der Einspeisung/Bezug meldet, der ist doch eigentlich immer korrekt*. Nur die PV-Erzeugung, Nutzung vom Speicher usw. ist unbekannt. Das heißt die Differenz kann man immer auf unbekannte Erzeugungsanlagen aufschlagen. Oder nicht?

* Im Sinne von "man muss nicht zwei Werte addieren", am korrekten Ort und korrekt verkabelt muss er natürlich sein.

naltatis, 2024-09-17:
Das ist das gleiche Thema wie hier: https://github.com/evcc-io/evcc/issues/15847#issuecomment-2325255272
In so einem Fall passen auch die Preis und %-Sonne Berechnungen leider nicht mehr.

> Aber was mir erstmal nur vorschwebt ist eine automatische Skalierung der Balken anhand des größten Wertes. Im oberen Bild müsste rechts einfach nur 4,6 sein und nicht 1,4. Der Fehlende Bereich könnte leer sein, wie ja auch schon im 2. Bild, oder besser ein Fragezeichen oder sowas, dann sieht es nicht kaputt aus.

Bzgl. UI kann ich mal schauen, ob wir da etwas Pragmatisches hinbekommen.

Automatisches Ausgleichen auf Datenbasis sollten wir nicht anfangen. Da können wir, ohne zusätzliche Einstellungsmöglichkeiten, keine zuverlässigen Annahmen treffen.

DavoBe16, 2024-09-19:
Situation ist bei mir auch so, weil ich einen einphasigen alten Wechselrichter der keinerlei Daten liefern kann zusätzlich zur neuen Anlage habe. Die neue Anlage erfasst dessen Produktion(genauer gesagt: dessen Überschuss) aber über ihr Gridmeter mit.Dieses Setup ist gar nicht mal so selten da draußen. Solang der gesamte Überschuss als Regelvariable für meine Wallbox genutzt wird, ist das alles für mich in Ordnung, auch wenn der unbekannte Anteil im der Balkenvisualisierung nicht gekennzeichnet ist. Aber man könnte trotzdem doch eine Regel bilden in der Art: „Wenn Input < Output  (da ein unbekannter Input da ist der nicht von der erfassten PV Produktion,Batterieentladung oder Netzbetug kommt) dann betrachte die Differenz als unbekannten Input oder PV Produktion von unbekannten Drittgeräten. und weise den Wert aus. Sungrow löst das in Ihrer Visualisierung übrigens so, dass negativer Hausverbrauch angezeigt wird. Oder denke ich zu kurz? 

[image]
[image]

naltatis, 2024-09-20:
Ja, wie oben beschrieben. Den Mismatch in der Visualisierung darzustellen, halte ich für eine gute Idee. Alles was aber Richtung Statistik und Auswertung geht (% Sonne, Kosten, ...) können wir aber nicht zuverlässig ohne das komplette Datenbild lösen. Da fangen wir schnell das Raten an und produzieren dann zwar gegebenenfalls richtigere, aber immer noch keine richtigen Zahlen.

Wenn das wichtig ist, führt hier kein Weg an einem zusätzlichen Zähler für bspw. die alte PV-Anlage vorbei.

@DavoBe16 den Sungrow-Case habe ich hier auch. Ich find das mit dem negativen Hausverbrauch maximal unintuitiv und die daraus resultierenden Zahlen sind dann auch nur noch teilweise brauchbar. :D

naltatis, 2024-09-23:
Hier gibts jetzt ne Lösung: https://github.com/evcc-io/evcc/pull/16289

Erstmal etwas konservativ, sodass der unbekannte Bereich nur bei großen Diskrepanzen (> 10%) angezeigt wird.

closed 2024-09-25 by andig

snowdd1, 2024-10-06:
Das passiert (kurzfristig) auch wenn man beim grid meter und beim Inverter einen gleitenden Durchschnitt verwendet - da die Loadpoints Momentanwerte verwenden passen die Werte bei Last/Erzeugungswechseln nicht zusammen.

PR #16289 (2024-09-25): Visualization: unknown power
fixes #16148

Introduce a question-mark when input and output don't match up.

- ⬇️ imported power to low: show a question mark bracket at the top
- ⬆️ imported power to high: show a question mark bracket ad the bottom and add a gray box to the color segments in the middle.
- 🚧 only show unknown areas if mismatch between in and out is above 10% to avoid flickering during normal operation due to meter reading delays.

**For later:**
Maybe add a label that explains the question mark and gray box.

[image]

[image]

[image]