evcc-io/evcc #16155: FYI: Kia API broken again

RustyDust, 2024-09-16:
### Describe the bug

Same with HomeAssistant, doesn't work there anymore either.
As said in the title: just for information.

### Steps to reproduce

- (re)start evcc 
- attach Kia BEV to wallbox

### Configuration details

```yaml
vehicles:
- name: niro
  type: template
  template: kia
  title: Kia Niro EV
  user: user
  password: password
  capacity: 64
  mode: pv
  icon: car
```

### Log details

```text
Sep 16 21:08:22 sru-pi evcc[1933209]: [main  ] ERROR 2024/09/16 21:08:22 creating vehicle niro failed: cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
```
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.11

emcepe, 2024-09-17:
same applies to Hyundai

🤮🤮🤮

I HATE these silly and braindead cat & mouse game by those companies.

VolkerK62, 2024-09-17:
https://github.com/Hyundai-Kia-Connect/kia_uvo/issues/932

0x3d13f, 2024-09-17:
To me it looks like a server problem, not a change in the API. I logged out of the Kia Connect App and also there is a login unsuccessful now.

RustyDust, 2024-09-17:
> Login in the Kia Connect App is working again for me. evcc can also get the status of my EV6 again.

Can confirm this. Web / App / evcc login is working again. Closing.

closed 2024-09-17 by RustyDust

MrPhtevens, 2024-10-21:
Having same issue in Australia with Hyundai. Bluelink working fine here

[main ] ERROR 2024/10/21 17:48:02 creating vehicle I6 failed: cannot create vehicle type 'template': cannot create vehicle type 'hyundai': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
[main ] FATAL 2024/10/21 17:48:02 failed configuring loadpoints: failed configuring loadpoint: not found: vehicle_1