evcc-io/evcc #16166: Huawei LUNA Battery - Set Charging Power

MarcelMertens, 2024-09-17:
**Is your feature request related to a problem? Please describe.**
When configuring "Charge Home Battery" on low grid prices, the charging is started, but only with 2500W. 2500W (dis)charge is the limit per battery pack. As i am running 2 battery packs 5.000W charge is possible.

**Describe the solution you'd like**
Add config parameter to set charging power?
In the reference 
https://docs.evcc.io/docs/devices/meters#sun2000-with-sdongle--power-sensor
`  storageunit: 1 # optional
  capacity: 50 # Akkukapazität in kWh (optional)`
is listed as optional parameter, but no documentation if this affect the charging power.

mucki12, 2024-09-17:
Hier war die Frage, ob Huawei über Modbus größere Angaben als Maximalwert interpretiert:
https://github.com/evcc-io/evcc/issues/15647#issuecomment-2308652809

Wenn ja (ich war wie man sieht technisch nicht in der Lage dieses mit einem Modbus Tool zu testen), dann würde man nicht einmal einen neuen Parameter benötigen.

andig, 2024-09-17:
Just guessing- but if you have 2 storage units (?) you'll need to configure 2 meters.