evcc-io/evcc #16180: evcc may crash with a segfault when switching cars on a loadpoint

jennes, 2024-09-17:
### Describe the bug

When switching from a supported car (a Hyundai in this case) to the "guest" car, evcc may crash with a segfault
(tested with a binary built from source)

### Steps to reproduce

1. Choose the Hyundai in the UI
2. Switch at the right time (presumably shortly after the update request to Hyundai has been triggered) to the "guest" car
3. Watch evcc crash when it tries to update the SoC of the guest car (which seems to have a _nil_ pointer to the SoC struct)

### Configuration details

```yaml
circuits:
  - name: main
    title: Hausanschluss
    maxCurrent: 25 # 25A
    meter: grid1
chargers:
  - name: mywallbox
    type: custom
    enabled:
      source: http
      uri: ****
      method: GET
      timeout: 1s
    enable:
      source: http
      uri: ****
      method: POST
      timeout: 1s
      headers:
        - content-type: application/json
      body: ${enable:%v}
    maxcurrent:
      source: http
      uri: ****

vehicles:
  - name: ioniq
    type: template
    template: hyundai
    title: ****
    user: ****
    password: ****
    capacity: 38
    language: de
    phases: 1
    minCurrent: 6
    maxCurrent: 32
    priority: 2
loadpoints:
  - title: Stellplatz
    charger: mywallbox
    mode: pv
    circuit: main
```

### Log details

```text
[site  ] INFO 2024/09/17 09:34:39 site config:
[site  ] INFO 2024/09/17 09:34:39   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/09/17 09:34:39     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/09/17 09:34:39     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/09/17 09:34:39     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/09/17 09:34:39   vehicles:
[site  ] INFO 2024/09/17 09:34:39     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[site  ] INFO 2024/09/17 09:34:39     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/09/17 09:34:39 loadpoint 1:
[lp-1  ] INFO 2024/09/17 09:34:39   mode:        pv
[lp-1  ] INFO 2024/09/17 09:34:39   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/09/17 09:34:39   meters:      charge ✗
[lp-1  ] INFO 2024/09/17 09:34:39 vehicle updated: unknown -> *** ioniq
[lp-1  ] INFO 2024/09/17 09:34:39 car connected
[lp-1  ] INFO 2024/09/17 09:34:47 vehicle updated: *** ioniq -> unknown
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x40 pc=0x2623401]

goroutine 364 [running]:
github.com/evcc-io/evcc/core/soc.(*Estimator).RemainingChargeEnergy(...)
	github.com/evcc-io/evcc/core/soc/estimator.go:96
github.com/evcc-io/evcc/core.(*Loadpoint).publishSocAndRange(0xc001e66408)
	github.com/evcc-io/evcc/core/loadpoint.go:1574 +0x7c1
github.com/evcc-io/evcc/core.(*Loadpoint).Update(0xc001e66408, 0x408645999999999a, {0xc00132e008, 0x72, 0x74}, 0x0, 0x0, 0x0, 0xc000a1d200, 0xc000a1d208)
	github.com/evcc-io/evcc/core/loadpoint.go:1701 +0x82a
github.com/evcc-io/evcc/core.(*Site).update(0xc001e34200, {0x38214b0, 0xc001e66408})
	github.com/evcc-io/evcc/core/site.go:882 +0x8b8
github.com/evcc-io/evcc/core.(*Site).Run(0xc001e34200, 0xc001e682a0, 0x37e11d600)
	github.com/evcc-io/evcc/core/site.go:994 +0x217
github.com/evcc-io/evcc/cmd.runRoot.func4()
	github.com/evcc-io/evcc/cmd/root.go:288 +0x25
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:287 +0x1e13
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc 0.130.11 (3eb1ae0c)

andig, 2024-09-17:
Interesting. It shouldn't be able to do that since there should be not context switching at this time (and `socEstimator == nil` is being checked before). Wondering how that can come about...

jennes, 2024-09-18:
The Hyundai Bluelink API is rather slow, so it might be, that while the update is still in progress, one could switch the car to the default car (`==nil`). If that happens, `lp.socEstimator` may change after the check.

`LoadPoint.setActiveVehicle` uses `LoadPoint.vmu` but I am uncertain if this would be a good solution for `publishSocAndRange`, too, since it also sends to `uiChan` which may block.

PR #16194 (2024-09-18): Loadpoint: fix race condition accessing soc estimator (#16180)
Fix #16180

It might not be ideal since a now soon-to-not-exist socEstimator may be updated, but the alternative would have been to have mutexes.
Since the bug seems to be rather esoteric, the risk of showing "old" information for a short period of time seems to be acceptable. It should be better than a hard crash in any case.

With that patch applied I couldn't provoke a crash anymore but since I don't fully understand the code paths of evcc (yet)…
closed 2024-09-18 by andig