evcc-io/evcc #16181: BYD Dolphin

Oberreiter, 2024-09-17:
Ich habe einen BYD Dolphin. Kann man den mit der generischen Unterstützung anbinden?
Wenn nein gibt es eine Liste/Zeitplan wann BYD Modelle unterstützt werden?

mdkeil, 2024-09-17:
#15353
also aktuell nicht.. es benötigt halt eine beschriebene (API) Schnittstelle..

closed 2024-09-17 by andig

Oberreiter, 2024-09-19:
Geht mit Einschränkungen doch. Man kann zwar keine Ladung planen aber wenn man das Fahrzeug anschließt lädt es im eingestellten Modus. D.h. im PV Modus den möglichen Überschuss. Da der Dolphin immer bis 100% geladen werden kann ist das so ok. Man kann auch ein Custom Vehicle konfigurieren und beim ensprechenden Ladepunkt als default hinterlegen dann wird der Name angezeigt.

swingnjazz, 2024-10-01:
Ich habe meinen BYD Atto 3 als Offline Vehicle definiert. Damit kann ich eine Ladung für eine Fahrt planen mit einer bestimmten zuzuladenden Energiemenge, z.B. 30 kWh (50%). So bleibt noch genügen Kapazität für das Laden mit PV-Strom nach der Fahrt übrig.

Meine Konfiguration in der evcc.yaml:

vehicles:
- type: template
  template: offline 
  title: BYD Atto 3  
  icon: car  
  capacity: 60.5  
  phases: 3  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev4