evcc-io/evcc #16182: Home battery capacity permanently limited to 80% after blocking home battery for quick charge once.

wiwimaster, 2024-09-17:
### Describe the bug

Our infrastructure consists of a Tesla S, Tesla Powerwall and a PV installation (SMA, not relevant here). Details see evcc.yaml. 

Under "Home Battery", I switched on "Verhindere Entladung im Schnell-Modus ..." to avoid depletion of the battery. During activation, this setting created a "minimum charge" of the powerwall which - as planned - prevented the home battery from depletion.

However, this setting never goes back to 0%, but remains at a minimum of 20%, even when no car is charging or even when the setting is switched off. 

It would be great if you could have a look what causes this, as this effectively reduces our battery capacity by 20%, as we can't use this power anymore.

This shows the initial setting with the "direct charge" setting switched on:
[image]

This results in the powerwall to be limited to 40% of own use, effectively blocking the powerwall of being used:
[image]

once the setting is switched off however, the limit goes up to 80%, but not 100%, reducing the effective capacity of the powerwall by -20%:
[image] 

As you can see on this image here, the 20% "safety net" were kept the whole day
[image]

### Steps to reproduce

See descript

[EVCC-logs.txt](https://github.com/user-attachments/files/17033600/EVCC-logs.txt)

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 30s # control cycle interval

# lifetime sponsortoken
sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: tesla-powerwall 
  usage: grid  
  host: 192.168.1.102  
  password: '00786'  
  refreshToken: xxx
  name: grid1
- type: template
  template: tesla-powerwall 
  usage: pv  
  host: 192.168.1.102  
  password: '00786'  
  refreshToken: xxx 
  name: pv2
- type: template
  template: tesla-powerwall 
  usage: battery  
  host: 192.168.1.102  
  password: '00786'  
  refreshToken: xxx  
  name: battery3

chargers:
- type: template
  template: tinkerforge-warp 
  host: 192.168.1.50  
  port: 1883  
  topic: wallbox/warp3  
  timeout: 30s  
  name: wallbox5
  energymanager: wallbox/warp3

vehicles:
- type: template
  template: tesla 
  title: JansTeslaS  
  accessToken: xxx  
  refreshToken: xxx
  vin: 5YJSA7E29HF193920  
  capacity: 75  
  name: ev4

loadpoints:
- title: Carport
  charger: wallbox5
  mode: pv
  #phases: 0 #automatische umschaltung von 1 zu 3 phasen

site:
  title: Wörthsee
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

tariffs:
  co2:
    type: grünstromindex # Beispiel GrünstromIndex (nur Deutschland)
    zip: 82237 # Deine Postleitzahl
  currency: EUR
  grid:
    type: fixed
    price: 0.3057 # 30,57ct/kWh
    zones:
      - days: Mo-So
        hours: 0-5
        price: 0.1651 # 16,51ct/kWh
```

### Log details

```text
see attachment under "Steps to reproduce", was too big for pasting
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.11

andig, 2024-09-17:
/cc @GrimmiMeloni

GrimmiMeloni, 2024-09-17:
This is the expected default behavior. You can override this by setting `minSoc` in the configuration. You can find the example in the docs when showing the advanced settings.

closed 2024-09-17 by GrimmiMeloni

andig, 2024-09-18:
Ahh, das fehlte mir. Die Anzeige entspricht 100-minsoc?

GrimmiMeloni, 2024-09-18:
>  Die Anzeige entspricht 100-minsoc?

Exakt.

wiwimaster, 2024-09-18:
Tut mir leid, aber ich finde in der Dokumentation (https://docs.evcc.io/docs/Home) kein Beispiel, wie man in der evcc.yaml dieses Setting für die Hausbatterie anwendet, die Suche ergibt nur drei Treffer zur MQTT Steuerung, und hier bezieht sich minsoc nach meinem Verständnis auf das Fahrzeug, nicht die Hausbatterie.
@GrimmiMeloni könntest du mir einen Link auf das genannte Beispiel geben, das würde mir die Einstellung sehr erleichtern.

Was ist denn der Grund für dieses "Expected Behaviour", also dass nach "an-aus" des Settings plötzlich die Batterie nicht mehr vollständig nutzbar ist, und der Nutzer dies dann manuell korrigieren muss?

GrimmiMeloni, 2024-09-18:
Docs siehe [hier](https://docs.evcc.io/docs/devices/meters#tesla-powerwall) (auf show advanced settings klicken)

> Aber macht es Sinn, dass minsoc von 0 auf 20 dauerhaft geändert wird, und der Nutzer dies dann manuell wieder auf 0 korrigieren muss?

Der Nutzer muß nicht manuell korrigieren. Wenn 0 gewünscht ist, einfach minSoc 0 in die Konfig eintragen.

Nochmal erläutert fürs bessere Verständnis:
Wir müssen _irgendeinen_ Wert setzen, da die Powerwall keine "Entladesperre" in der Form an/aus bietet. Wir simulieren eine solche Sperre daher in dem wir den _Mindest_ladestand auf den _aktuellen_ Ladestand setzen. Das hat den gleichen Effekt. 

Sich dabei den vorherigen Mindestladestand dynamisch in evcc zu merken funktioniert nicht, weil diese Werte flüchtig sind (z.B. durch einen Reboot von evcc). Daher müssen User in Ihrer Konfig den Wert den sie als minSoc haben wollen explizit konfigurieren. Den Default von 20% habe ich damals gewählt, weil es der empfohlene Wert in der Tesla App ist. Es kann aber jeder seinen individuellen Wert einstellen.

wiwimaster, 2024-09-18:
Super, danke dir, die Docs hab ich gefunden und erklären die Settings perfekt.

Die Funktionsweise die du beschreibst macht absolut Sinn und ist eine elegante Lösung, die PW zu sperren. Ich verstehe auch, dass sich der Wert nicht so einfach "merken" lässt.

Wir haben noch ein recht altes Setup aus 2x Powerwall 2 sowie Gateway 1 - das Setup hat keine Notstromfunktion, weshalb ein Default von 20% hier keinen Sinn macht, sondern lediglich die effektive Kapazität der Powerwall verringert. 

Für mich ist das minsoc setting jetzt ne super Lösung. Falls das Thema nochmal bei anderen Nutzern aufkommt könnte man ja überlegen, das minsoc setting während des Einrichtungsprozesses abzufragen, oder vor der erstmaligen Aktivierung der Entladesperre auszulesen und in der datenbank zu speichern. Ich kann zwar kein Go programmieren, stehe aber gerne mit Rat & Tat zur Seite ;-)

Danke nochmal für die gute Erklärung, es ist wirklich toll, was ihr mit evcc geschaffen habt :-)

wiwimaster, 2024-09-23:
Guten Morgen, jetzt ist mir noch etwas aufgefallen @GrimmiMeloni : mit dem minsoc-setting fällt die Powerwall nun zuverlässig auf diesen default-wert zurück, wenn das Fahrzeug geladen wird. 

Das minsoc-Setting scheint evcc aber bei Stromtarifen nicht zu greifen: ich habe von 0-5 Uhr einen anderen Tarif wie auf https://docs.evcc.io/docs/features/dynamic-prices beschrieben (siehe evcc.yaml unten) hinterlegt. In dieser Zeit wird in der Powerwall wieder der Reservewert hochgesetzt, und um 5 Uhr wieder auf 0% (wie mit minsoc hinterlegt) abgesenkt. Hier handelt es sich übrigens nicht um Fahrzeugladen sondern um den ganz normalen Hausverbrauch.

Ich habe es leider nicht geschafft herauszufinden, warum genau in der definierten Zone der minsoc-Wert automatisch hochgesetzt wird, und wie ich das verhindern kann. In der Powerwall selbst ist kein Zeittarif hinterlegt.

Habt ihr eine Idee, wie ich auch beim Stromtarif das minsoc-Setting durchsetzen kann?

Danke

Hier sieht man die Batterieeinstellungen kurz nach Mitternacht
[image]

Und hier den Hausverbrauch ab Mitternacht, um 5 Uhr wird die Powerwall wieder genutzt:
[image]

Hier die evcc.yaml:

```
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: yyy

interval: 30s # control cycle interval

# lifetime sponsortoken
sponsortoken: yyy

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: tesla-powerwall 
  usage: grid  
  host: 192.168.1.102  
  password: '00786'  
  refreshToken: yyy 
  name: grid1
- type: template
  template: tesla-powerwall 
  usage: pv  
  host: 192.168.1.102  
  password: '00786'  
  refreshToken: yyy  
  name: pv2
- type: template
  template: tesla-powerwall 
  usage: battery  
  host: 192.168.1.102  
  password: '00786'  
  refreshToken: yyy  
  name: battery3
  minsoc: 0

chargers:
- type: template
  template: tinkerforge-warp 
  host: 192.168.1.50  
  port: 1883  
  topic: wallbox/warp3  
  timeout: 30s  
  name: wallbox5
  energymanager: wallbox/warp3

vehicles:
- type: template
  template: tesla 
  title: JansTeslaS  
  accessToken: yyy  
  refreshToken: yyy  
  vin: 5YJSA7E29HF193920  
  capacity: 75  
  name: ev4

loadpoints:
- title: Carport
  charger: wallbox5
  mode: pv
  #phases: 0 #automatische umschaltung von 1 zu 3 phasen

site:
  title: y
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

tariffs:
  co2:
    type: grünstromindex # Beispiel GrünstromIndex (nur Deutschland)
    zip: 82237 # Deine Postleitzahl
  currency: EUR
  grid:
    type: fixed
    price: 0.3057 # 30,57ct/kWh
    zones:
      - days: Mo-So
        hours: 0-5
        price: 0.1651 # 16,51ct/kWh

```

andig, 2024-09-23:
> In dieser Zeit wird in der Powerwall wieder der Reservewert hochgesetzt, und um 5 Uhr wieder auf 0% (wie mit minsoc hinterlegt) abgesenkt. 

Nur zur Klarheit: 0% kannst Du nicht konfigurieren. Dann gelten weiter die eingestellten 20%.

wiwimaster, 2024-09-23:
Hi @andig danke dir für die schnelle Rückmeldung. Ich hab alle Logs angeschaut, das scheint in der Tat zu Problemen zu führen. Gibt es außer 0 noch Werte, die man nicht konfigurieren kann? Ich würde gerne einen möglichst niedrigen Wert nehmen, um die Kapazität des Stromspeichers vollständig nutzen zu können.

andig, 2024-09-23:
Nein, nur 0 ist "leer" und überschreibt daher keinen Default.

GrimmiMeloni, 2024-09-23:
> Habt ihr eine Idee, wie ich auch beim Stromtarif das minsoc-Setting durchsetzen kann?

Das könnte ein unerwünschter Nebeneffekt vom Planer sein. Aber um das wirklich beurteilen zu können, braucht es davon mal ein Log.

wiwimaster, 2024-09-23:
Sehr gerne, siehe anbei, da hatte ich das minsoc setting noch auf 0.

[EVCC-log.csv](https://github.com/user-attachments/files/17096429/EVCC-log.csv)

GrimmiMeloni, 2024-09-23:
```
DEBUG 2024/09/22 00:00:16 set battery mode: charge
```

Wie vermutet, wird hier von evcc auf charge mode gestellt, und deshalb der MinSoc auf 100% gesetzt. Sieht für mich nach einem Seiteneffekt vom Netzladen für Hausakkus aus.

@andig ich habe mich mit dem Charge feature noch nicht beschäftigt. Ist die Präsenz eines dynamischen Stromtariffs auch automatisch der Auslöser dafür, Hausbatterien in den Netzladungsmodus zu versetzen?

andig, 2024-09-23:
Wenn eine Preis/Co2 Grenze dafür gesetzt ist?