evcc-io/evcc #16191: Unable to restart after publishing then refreshing sponsor token

jivesinger, 2024-09-18:
### Describe the bug

Refreshing the `sponsortoken` does not remove the error.
Removing the `sponsortoken` key from yaml does not remove the error

`FATAL 2024/09/17 07:10:51 sponsortoken: rpc error: code = Unknown desc = token blocked due to public exposure, please generate new token or contact info@evcc.io for support
`
I've ticked the box saying that I've tried nightlies, but running the home assistant addon, I'm not aware of a way I can do this, so that is actually false.
My apologies for this.

### Steps to reproduce

1. Accidentally publish sponsor token
2. Upgrade 0.130.10 to 0.130.11
3. Fails to start with error: `sponsortoken: rpc error: code = Unknown desc = token blocked due to public exposure`
4. Refresh sposnortoken multiple times
5. Fails to start with error: `sponsortoken: rpc error: code = Unknown desc = token blocked due to public exposure`
6. Remove sponsortoken key from yaml
7. Fails to start with error: `sponsortoken: rpc error: code = Unknown desc = token blocked due to public exposure`
8. Donwgrade to 0.130.10
9. Fails to start with error: `sponsortoken: rpc error: code = Unknown desc = token blocked due to public exposure`
10. 
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: homeassistant.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
#sponsortoken: notThisTimeBuddyBoy;)
# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
#telemetry: true

# log settings
log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
  ocpp: trace
[... 239 lines cut ...]
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/config/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/09/18 16:11:56 evcc 0.130.11
[main  ] INFO 2024/09/18 16:11:56 using config file: /config/evcc.yaml
[main  ] INFO 2024/09/18 16:11:57 listening at :7070
[main  ] FATAL 2024/09/18 16:11:57 sponsortoken: rpc error: code = Unknown desc = token blocked due to public exposure, please generate new token or contact info@evcc.io for support
[main  ] FATAL 2024/09/18 16:11:57 will attempt restart in: 15m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.10 / 0.130.11

Chris45671, 2024-09-18:
If i am correct, you need to remove # at the Line with the Token. They dont Work with # at Start.

jivesinger, 2024-09-18:
> If i am correct, you need to remove # at the Line with the Token. They dont Work with # at Start.

I deliberately commented out that line while fault finding.
It should have made the error go away.
A valid and non-published token should also make the error go away.
They don't.
That's the bug 😉

0x3d13f, 2024-09-18:
Maybe you have more than one evcc.yaml and edit the wrong one?

Chris45671, 2024-09-18:
You can Change the Token via GUI.
Did you Check If in your Database the Same Token AS you want use in yaml?

jivesinger, 2024-09-18:
> Maybe you have more than one evcc.yaml and edit the wrong one?

Only one evcc.yaml

jivesinger, 2024-09-18:
> You can Change the Token via GUI. Did you Check If in your Database the Same Token AS you want use in yaml?

I didn't know that was a thing.
I'll check it out.

If the token is stored in the db, shouldn't it be updated when the yaml is updated ?

Chris45671, 2024-09-18:
> > You can Change the Token via GUI. Did you Check If in your Database the Same Token AS you want use in yaml?
> 
> I didn't know that was a thing. I'll check it out.
> 
> If the token is stored in the db, shouldn't it be updated when the yaml is updated ?

In my Installation it dont Work. I need to Mark the Line for the Token with # then Start EVCC and Change in the GUI the token

jivesinger, 2024-09-18:
> In my Installation it dont Work. I need to Mark the Line for the Token with # then Start EVCC and Change in the GUI the token

Well I'll be horn-swoggled. That worked ! Thanks @Chris45671 

Seems as though if your token is made invalid while set in yaml, the error can only be resolved by removing the yaml and setting a new token in the GUI.

Now I'm not sure if this a bug 🤷‍♂️

closed 2024-09-18 by andig