evcc-io/evcc #16195: EVCC Config test successfull - service start however fails

Saphirim, 2024-09-18:
### Describe the bug

I followed the debian install guide putting EVCC on a LXC debian container hosted on Proxmox.

I created a config file using the evcc cli.

I sucessfully tested the creatd config with `evcc -c evcc.yaml`

After moving the evcc.yaml file to /etc and starting the evcc service, the UI, Logs and `systemctl status evcc` show the following error: 

```
[main ] FATAL 2024/09/18 07:58:02 failed configuring loadpoints: missing loadpoints
[main ] FATAL 2024/09/18 07:58:02 will attempt restart in: 15m0s
```

The used components are:
- Fronius Gen24 Plus 6.0
- Fronius WattPilot 11 J 2.0

This is the config as created by the `evcc configure` command:

```# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 34f4821cc3eca0de0841c025e918bc427780bc3adc59058adcd1cdd06e55ab48

interval: 30s # control cycle interval

meters:
- type: template
  template: fronius-gen24 
  usage: grid  
  host: 192.168.1.XXX  
  port: 502  
  name: grid1
- type: template
  template: fronius-gen24 
  usage: pv  
  host: 192.168.1.19  
  port: 502  
  name: pv2
- type: template
  template: fronius-gen24 
  usage: battery  
  host: 192.168.1.19  
  port: 502  
  name: battery3

chargers:
- type: template
  template: fronius-wattpilot 
  host: 192.168.1.XXX  
  password: ----------  
  name: wallbox4

loadpoints:
- title: Einfahrt
  charger: wallbox4
  mode: pv

site:
  title: Ada18
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Steps to reproduce

1. Create config with `evcc configure`
2. Test config with `evcc -c evcc.yaml ` while the evcc file is located in the home directory
3. Move the config file as guided by the docs to /etc
4. Start the evcc service via systemctl

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 34f4821cc3eca0de0841c025e918bc427780bc3adc59058adcd1cdd06e55ab48

interval: 30s # control cycle interval

meters:
- type: template
  template: fronius-gen24 
  usage: grid  
  host: 192.168.1.XXX  
  port: 502  
  name: grid1
- type: template
  template: fronius-gen24 
  usage: pv  
  host: 192.168.1.19  
  port: 502  
  name: pv2
- type: template
  template: fronius-gen24 
  usage: battery  
  host: 192.168.1.19  
  port: 502  
  name: battery3

chargers:
- type: template
  template: fronius-wattpilot 
  host: 192.168.1.XXX  
  password: ----------  
  name: wallbox4

loadpoints:
- title: Einfahrt
  charger: wallbox4
  mode: pv

site:
  title: Ada18
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```
```

### Log details

```text
[main ] INFO 2024/09/18 07:58:00 evcc 0.130.11
[main ] INFO 2024/09/18 07:58:00 using config file: /evcc.yaml
[db ] INFO 2024/09/18 07:58:00 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2024/09/18 07:58:00 listening at :7070
[main ] FATAL 2024/09/18 07:58:02 failed configuring loadpoints: missing loadpoints
[main ] FATAL 2024/09/18 07:58:02 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.130.11

VolkerK62, 2024-09-18:
das Log sagt dir, welche yaml benutzt wird.
`[main ] INFO 2024/09/18 07:58:00 using config file: /evcc.yaml`

Saphirim, 2024-09-18:
Danke für den Hinweis!

Ist das empfohlene Vorgehen, die gewünschte evcc.yaml unter / zu verschieben?
Oder kann ich meine EVCC-Instanz so konfigurieren, dass aut /etc/evcc.yaml zugegriffen wird?

jennes, 2024-09-18:
In `/` hat die evcc.yaml ganz sicher nichts verloren, es sei denn, Du arbeitest in einem Docker-Container oder so.
Ich gehe davon aus, dass Du evcc per systemd oder so startest? Dann musst Du der systemd-Unit auch sagen:
`ExecStart = /pfad/zu/evcc -c /etc/evcc.yaml`

Saphirim, 2024-09-18:
> In / hat die evcc.yaml ganz sicher nichts verloren,...

Geht mir genauso.

Solved the issue by editing the service definition file located under `/usr/lib/systemd/system/evcc.service` like so:

`ExecStart=/usr/bin/evcc -c /etc/evcc.yaml`

-> `systemctl daemon-reload`
-> `systemctl start evcc`

Still wondering however why the default location was below `/`, seems like an odd out of the box setting.

closed 2024-09-18 by Saphirim

jennes, 2024-09-18:
> Solved the issue by editing the service definition file located under `/usr/lib/systemd/system/evcc.service` like so:
>
> `ExecStart=/usr/bin/evcc -c /etc/evcc.yaml`
>
> -> `systemctl daemon-reload` -> [...]

_evcc_ sucht beim Start im aktuellen Verzeichnis. Beim Start aus _systemd_ heraus ist das eben `/`, was aus verschiedenen Gründen beim Systemstart tatsächlich sinnvoll ist.