evcc-io/evcc #16213: Hausbatterie wird zum fälschlicherweise zum Laden des Autos verwendet

walburgf, 2024-09-19:
### Describe the bug

ich habe einen Ladeplan eingestellt, der mein Auto ID.4 bis morgens um 7 auf 80% laden soll. Ich habe Tibber und er sollte den günstigsten Strompreis nehmen. Das hat er auch gemacht.
Weiterhin habe ich eingestellt, dass die Batterie bei geplantem Laden sich nicht entladen soll. Dies hat nicht funktioniert. Als das Auto geladen wurde, wurde zuerst die Batterie leer gesaugt, dann wurde erst aus dem Netz geladen.
[image]
[image]

Im Log sieht man, dass der lp-1 um 0 Uhr startet und gegen 2 Uhr fertig ist. Unter Site ist zu sehen, dass die Batterie mit ca. 10kW entladen wird.
[evcc.log](https://github.com/user-attachments/files/17055919/evcc.log)

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc-pve.fritz.box:7070
network:
  schema: http
  host: evcc-pve.fritz.box # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  lp-3: trace
  script: trace
  cache: error

database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db

# unique installation id
plant: xxxx
telemetry: true
sponsortoken: xxxxx
interval: 20s # control cycle interval, default is 30s

meters:
- name: grid1
  type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.1.246

- name: BYD_battery
  type: template
  template: sma-hybrid
  usage: battery
  modbus: tcpip
  id: 3
  host: 192.168.1.231
  port: 5022
#  host: 192.168.1.247
#  port: 502

- name: SMA_Hybrid_WR
  type: template
  template: sma-hybrid
  usage: pv
  # Modbus TCP
  modbus: tcpip
  id: 3
  host: 192.168.1.247 # Hostname
  port: 502 # Port 

- name: AC_Thor_9s_Meter
  type: custom
  power:
    source: modbus
    id: 1
    uri: 192.168.1.231:5021  # IP-Adresse oder Hostname 
    timeout: 2s
    register:
      address: 1000
      type: holding # Holding Register lesen
      decode: uint16
  soc:
    source: http
    uri: http://192.168.1.248/data.jsn
    method: GET # default HTTP method
    headers:
      - content-type: application/json
    insecure: false # set to true to trust self-signed certificates
#    jq: "[.temp1, .temp2, .temp3] | map(. | select (. != null)) | min" # parse response json, get the lowest temperature
    jq: "[.temp1, .temp2, .temp3] | map(. | select (. != null)) | add / length" # parse response json, get the median temperature
    scale: 0.1 # floating point factor applied to result, e.g. for kW to W conversion
    timeout: 10s # timeout in golang duration format, see https://golang.org/pkg/time/#ParseDuration

vehicles:
- name: vw.id.4
  title: VW ID.4
  type: template
  template: vw
  user: xxxx
  password: xxxx
  capacity: 77
  timeout: 10s
  cache: 20m

- name: Warmwasser
  title: Warmwasser
  type: custom
  icon: waterheater
  phases: 3
  soc:
    source: http
    uri: http://192.168.1.248/data.jsn
    method: GET # default HTTP method
    headers:
      - content-type: application/json
    insecure: false # set to true to trust self-signed certificates
#    jq: "[.temp1, .temp2, .temp3] | map(. | select (. != null)) | min" # parse response json, get the lowest temperature
    jq: "[.temp1, .temp2, .temp3] | map(. | select (. != null)) | add / length" # parse response json, get the median temperature
    scale: 0.1 # floating point factor applied to result, e.g. for kW to W conversion
    timeout: 10s # timeout in golang duration format, see https://golang.org/pkg/time/#ParseDuration

site:

- title: Salierstr. 37
  meters:
    grid: grid1
    pv:
    - SMA_Hybrid_WR
    battery:
    - BYD_battery
  residualPower: 100

chargers:
- name: Charger_rechts
  type: template
  template: easee
  user: xxxx
  password: xxxx
  charger: EHJZ7EM3
  timeout: 20s # optional

- name: Charger_links
  type: template
  template: easee
  user: xxxx
  password: xxxx
  charger: EHA4RSE2
  timeout: 20s # optional

- name: AC_Thor_9s_waterheater
  type: custom
  icon: waterheater
  power:
    source: modbus
    id: 1
    uri: 192.168.1.231:5021  # IP-Adresse oder Hostname 
    timeout: 2s
    register:
      address: 1000
      type: holding # Holding Register lesen
      decode: uint16
  wakeup:
    source: script
    cmd: /home/evcc/AC_Thor_9s/wakeup.sh
    timeout: 5s
  enabled: # charger enabled state (true/false or 1/0)
    source: script
    cmd: /home/evcc/AC_Thor_9s/get_enabled.sh #0 or 1
    timeout: 5s
  enable: # set charger enabled state (true/false or 0/1)
    source: script
    cmd: /home/evcc/AC_Thor_9s/set_enable.sh ${enable}
    timeout: 5s
  maxcurrent:
    source: script
    cmd: /home/evcc/AC_Thor_9s/set_maxcurrent.sh ${maxcurrent}
    timeout: 5s
  status: # charger status A..F
    source: script
    cmd: /home/evcc/AC_Thor_9s/get_status.sh #B or C always connected
    timeout: 5s
  features: [integrateddevice, heating]

loadpoints:
- title: Garage_rechts # display name for UI
  charger: Charger_rechts # Wallbox rechts Referenz
  vehicle: vw.id.4 # Referenz auf Standardfahrzeug
  mode: pv # charge mode (off, now, minpv, pv)
  priority: 6

- title: Garage_links # display name for UI
  charger: Charger_links  # Wallbox links Referenz 
  vehicle:  # Referenz auf Standardfahrzeug
  mode: pv # charge mode (off, now, minpv, pv)
  priority: 4

- title: AC Thor 9s
  charger: AC_Thor_9s_waterheater
  vehicle: Warmwasser
  mode: pv
  meter: AC_Thor_9s_Meter
  priority: 2
  soc:
    poll:
      mode: charging
      interval: 5m
    estimate: true
  enable:
    threshold: -100
    delay: 10s
  disable:
    threshold: -50
    delay: 60s

modbusproxy:
- port: 5021
  uri: 192.168.1.248:502  # IP des AC Thor 9s
- port: 5022
  uri: 192.168.1.247:502  # IP des SMA Wechselrichters
```

### Log details

```text
Log ist als Datei angehängt
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.130.11

mdkeil, 2024-09-19:
Was genau wird denn für ein Battery/Hybrid-Wechselrichter eingesetzt..? battery-hold wurde zumindest zeitgerecht gesendet, der WR hat aber darauf nicht reagiert..

`Sep 19 00:03:48 evcc-pve evcc[339]: [site  ] DEBUG 2024/09/19 00:03:48 set battery mode: hold`

`Sep 19 01:50:48 evcc-pve evcc[339]: [site  ] DEBUG 2024/09/19 01:50:48 set battery mode: normal`

walburgf, 2024-09-19:
SUNNY TRIPOWER 10.0 SE

mdkeil, 2024-09-19:
#11567 
#11580

Ob final ein fix gefunden wurde, kann ich nicht bewerten

/CC @premultiply