evcc-io/evcc #16216: HomeAssistant - MQTT - EVCC - GoodWe Wechselrichter - Easee Charge Lite Wallbox - Fehlersuche

PLiNNoX, 2024-09-19:
Hallo zusammen,

ich möchte das Überschussladen mit meiner PV-Anlage gerne über EVCC/HomeAssistant realisieren.
 - Wechselrichter : GoodWe Hybrid 
 - Batteriespeicher: BYD 
 - Wallbox: Easee Charge Lite
 - Auto: Nissan leaf 
 

Über MQTT und Node-Red werden die benötigten Signale bereits an EVCC gesendet.

[image]
 
Allerdings sieht die EVCC Oberfläche nicht schlüssig aus. EVCC öffnet sich auch nur sehr unregelmäßig, oft hängt er auch einfach fest und es kommt nur die Meldung " The add-on ist starting". Wenn die Oberfläche sich allerdings öffnet, ist der Grüne Balken falsch bzw. gar nicht aufgeteilt. Mein Hausverbrauch wird nicht angezeigt und auch der Netzbezug und die Einspeisung werden nicht dargestellt.

[image]

Kann mir jemand bei dieser Problematik weiterhelfen? 
Ich weiß leider absolut nicht weiter :(

Vielen Dank schon mal im voraus! Ich bin über jede Idee oder jeglichen Lösungsansatz dankbar!

Gruß Timo

mdkeil, 2024-09-19:
Die Daten laufen nicht sauber von HomeAssistant (über mqtt) ein.. sieht man am timeout beim grid-meter. Was ich nicht verstehe, wo der http-request herkommt-- vermutlich kommt das als Fehler-Rückmeldung aus Node-Red und wird über das entsprechende topic gegeben.. Ich würde da auf jeden Fall ansetzen.

PLiNNoX, 2024-09-19:
Hallo @mdkeil , 
ich habe mal ein debug-flow hinzugefügt. nach Neustart von HomeAssistant leider immernoch die selben Fehler.

[image]

Oder habe ich etwas falsche gemaht bei dem debug-flow? Bin ein Laie :/

mdkeil, 2024-09-19:
Der Debug-Flow ist nicht die Lösung-- zeigt dir aber im Debug-Fenster, was du aus den "blauen" Kästen bekommst.. ich denke mal darüber wird die Home-Assistent-API angezapft.. aber das scheint nicht so zu funktionieren, wie es soll.
Du kannst aber auch in Home-Assistent eine Automation erstellen welche Änderungen direkt via mqtt publisht.

Hier mal ein sehr einfachen Beispiel (kann man ja auch über die UI zusammenklicken):

```
alias: MQTT_e208_plugged
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.e208_plugged
condition: []
action:
  - data:
      retain: true
      topic: hass/e208/plugged
      payload: "{{ states('sensor.e208_plugged') }}"
      qos: "1"
    action: mqtt.publish
mode: single
```

mdkeil, 2024-09-19:
Alternativ, dein WR wird auch in evcc direkt unterstützt.. wenn du mit evcc und hass parallel zugreifen möchtest, gehst Du notfalls einfach über den in evcc-integrierten modbus-proxy-- ist natürlich abhängig davon, wie du den WR in Home-Assistent integriert hast.

mdkeil, 2024-09-19:
Was auch sein kann, wenn du Home-Assistent neu startest, natürlich in Node-Red nichts auslesen werde kann und der Fehler direkt an evcc-weitergeleitet wird.. das könnte schon zu Problemen führen, in Verbindung damit, dass auch innerhalb des timeout-Fensters keine korrekten Daten kommen.. Wen ich es richtig sehe, wird an den blauen Flows links unten der aktuelle Wert angezeigt.. beim battery-SOC steht z.B. "running" was natürlich auch kein korrekt erwarteter Wert für evcc ist.

RenatusRo, 2024-09-19:
um das outdated zu vermeiden, kannst du die werte auch um die x sekunden publishen: timestamp->curtent_state->mqtt

[image]

PLiNNoX, 2024-09-19:
@mdkeil kann ich meine Problem Beschreibung verschieben, oder muss ich ein komplett neues aufmachen?

mdkeil, 2024-09-19:
Das macht schon ein Dev für dich später.. 
such mal in der Auswahlliste nach "inject".

PLiNNoX, 2024-09-19:
> Alternativ.. dein WR wird auch in evcc direkt unterstützt.. wenn du mit evcc und hass parallel zugreifen möchtest, gehst Du notfalls einfach über den in evcc-integrierten modbus-proxy-- ist natürlich abhängig davon, wie du den WR in Home-Assistent integriert hast.

@mdkeil , wie kann ich am einfachsten nach schauen wie ich meinen Wechselrichter in Homeassistant integriert habe?

mdkeil, 2024-09-19:
Du wirst ja sicherlich irgend eine Integration nutzen.. dort wird dann ja die Verbindungsart eingestellt, wie die Kommunikation zum WR hergestellt wird.

PLiNNoX, 2024-09-19:
@mdkeil , sagt das hier etwas darüber aus? 
[image]

mdkeil, 2024-09-19:
Wenn dein WR diesbezüglich keine Limitierung hat, sind parallele Zugriffe möglich.. sieht nach Modbus TCP/UDP aus.. ist aber nur Mutmaßung ;) ..probieren geht über studieren ;)

PLiNNoX, 2024-09-19:
@mdkeil müsste ich die evcc.yaml umschreiben um die Daten auf einem anderen Weg an EVCC senden zu können? 
Falls ja, weißt du wie dieser Text dann aussehen müsste?

mdkeil, 2024-09-19:
https://docs.evcc.io/docs/devices/meters#goodwe

PLiNNoX, 2024-09-19:
@mdkeil 
Den Weg über Modbus hatte ich mal ausprobiert. Da ich mich damit allerdings nicht richtig auskenne und keine 1:1 Anleitung gefunden habe, habe ich es auf einem anderen Weg probiert, der zwar auch nicht funktioniert aber ich zumindest schonmal EVCC gestartet bekommen habe.

PLiNNoX, 2024-09-19:
Hallo nochmal,
die genannte Alternative von @RenatusRo hat bei mir funktioniert. Ich habe den timestamp jetzt auf 1x die Sekunde gesetzt, falls das zu häufig sein sollte, gerne Bescheid geben, aber so funktioniert es erstmal. Vielen lieben Dank dafür !!!!
[image]

closed 2024-09-19 by PLiNNoX

mdkeil, 2024-09-19:
1s muss nicht.. bei evcc ist ein interval von 30s eingestellt, 10-15s ist also locker ausreichend ;)

PLiNNoX, 2024-09-19:
@mdkeil alles klar, ich habe den Intervall jetzt auf 10s eingestellt.