evcc-io/evcc #16219: Add a "Solar+low tarif" strategy

jaytonic, 2024-09-19:
**Is your feature request related to a problem? Please describe.**
The current mode to charge the car are not optimal for my house. There is no bugs, but the off/solar/min+solar/fast strategy are not optimal.

**Describe the solution you'd like**
My house is equipped with solar panel and we have 2 different tariff(one for the the day, and one betwee 21h and 7h). The day tarif is 50% more expensive.
I would like a new mode: "Solar+low tarif".
This mode would charge during the day only with the solar surplus.
Then if the target is not met, it would wait the best tariff zone to charge to top-up the car. That would allow to maximize the solar production while still costing less.

**Describe alternatives you've considered**
Solar-only: WIll not top-up my car
Min-Solar: Will charge my car for a few hours before it is ready

**Additional context**
Add any other context or screenshots about the feature request here.

jaytonic, 2024-09-19:
SolarManager has a similar mode:
https://www.solarmanager.ch/fr/integration-de-la-station-de-recharge-de-voiture/
https://www.solarmanager.ch/einbindung-ladestation/

--> "Solar & Tarifoptimiert" or "Solaire & tarif optimisé"

VolkerK62, 2024-09-19:
Is still no problem.
You must define you tariff
https://docs.evcc.io/en/docs/features/dynamic-prices#time-dependent-electricity-tariff

and then you can set a price limit
https://docs.evcc.io/en/docs/features/dynamic-prices#cheap-grid-charging

So the car will be charged during the day with pv-surplus and (if the charge limit ist not reached) during night within the low tariff.

jaytonic, 2024-09-19:
Well, in fact, I think I misunderstood the "cheap grid charging" feature, it seems to work fine. sorry.

closed 2024-09-19 by jaytonic