evcc-io/evcc #16230: Own tesla proxy server option

toleabivol, 2024-09-20:
_Not sure if feature request or bug because I see some implementation already in place but not fully done_

**Is your feature request related to a problem? Please describe.**
Tesla integration into evcc now requires us to use Fleet API and that requires owning a key pair, a tesla dev app and installing the pub key on the vehicle. EVCC has done that by using the proxy server tesla provides. This means each car that uses evcc shares the same key and the central proxy is a security concern - e.g. one with access to this server could capture the traffic and get the access token and then send commands to the car.

EVCC already has a property that theoretically allows the change of proxy server domain 

> commandProxy: https://tesla.evcc.io/ # When using a TWC3 (or other 'dumb' charger not capable of control), evcc can manage the charge directly by communicating with the vehicle through a Command Proxy. By default, the proxy provided by evcc is used. With this parameter, you set the base URL of a custom Command Proxy to use instead of the default evcc one. See for example https://github.com/wimaha/TeslaBleHttpProxy for a proxy sending commands via bluetooth. (optional)

But setting this is not enough to switch:

1. evcc tesla proxy client is hardcoded to use US Fleet API [`FleetAudienceNA`](https://github.com/evcc-io/tesla-proxy-client/blob/4168b3759701f370cfb673ddd6489911a21dd4f8/client.go#L35) endpoint .
2. `TESLA_CLIENT_ID` is needed during build of evcc
3. (not sure) Tesla Proxy URL is hardcoded and the config is not  applied
4. Documentation missing on the whole workflow

**Describe the solution you'd like**
As a security guideline it is better that each evcc instance runs it's own proxy server with own keypair.
As not all workflow is part of evcc, documentation on how to do it with links to 3rd parties should be provided.

**Describe alternatives you've considered**
[For HA users there's an addon for tesla proxy.](https://github.com/llamafilm/tesla-http-proxy-addon)

**Additional context**
In the end I was able to run my own proxy server on an AWS EC2 t2.micro, create my developer app in Tesla, install the public key on the car and build evcc so that it works with my client id, eu fleet api base url and own proxy url. ATM I am facing the issue that the refresh url expires in 1d but not sure if it's something again missing on EVCC like `TESLA_CLIENT_SECRET` env var or just me playing with tokens too much on different devices that got it invalidated. This is where some doc might help again.

I see  `TESLA_CLIENT_SECRET` in codebase of evcc but not sure why it would need one if it has the CLIENT ID and refresh token ?

andig, 2024-09-20:
Thanks for documenting the process. At this time there are no plans to make this more configurable. I believe for tech-savvy users that's already possible while for end users it provides no additional value.

andig, 2024-09-20:
Regarding TESLA_CLIENT_SECRET I'm no longer sure why it was added. It's part of the Oauth2 config.

closed 2024-09-20 by andig

toleabivol, 2024-09-20:
> for end users it provides no additional value.
Added security benefit. Not sure if end user understands what they do when they install that key to the vehicle and use the central proxy.

andig, 2024-12-04:
Sure, think you can:

> I believe for tech-savvy users that's already possible while for end users it provides no additional value

vtml, 2024-12-04:
What I am concern, is whether the current usage with EVCC's hosted proxy, have enough allocation with Tesla's pricing.
If I setup EVCC with the current GitHub sponsor token, will the hosted EVCC proxy ask users to pay more in the future because the Tesla billing towards EVCC is no longer financially viable ?