evcc-io/evcc #16237: charger logic error: current mismatch (got 0.016A, expected 6A)

shellfisch, 2024-09-20:
### Describe the bug

Aufgrund der gleichlautenden Diskussion wie im obigen Titel, sollte ich den Fehler nochmals hier beschreiben. Zur Umgebung: 
EVCC-Version: 0.130.11
Sungrow Hybrid WR
Sungrow Wallbox AC011E-01
Auto Smart #1
Nach ähnlich lautenden Fehlerbeschreibungen bin ich der Handlungsempfehlung gefolgt und habe die control cycle intervals von 30 auf 60 und auf 120s ausprobiert habe, aber der Fehler blieb und ist auch nach wie vor existent. 

komplettes trace.log:
[evcc-20240920-135349-trace.log](https://github.com/user-attachments/files/17075174/evcc-20240920-135349-trace.log)
OCPP-Trace.log:
[evcc-20240920-135316-trace.log](https://github.com/user-attachments/files/17075181/evcc-20240920-135316-trace.log)

### Steps to reproduce

1.Empfehlung von coderph0x aus den discussions: Hast du EVCC selbst kompiliert? Mir ist dasselbe passiert, als ich nach einem git pull einfach den letzten Commit kompiliert hab. Beim Checkout des Commits aus dem Release 0.130.11 war der Fehler dann verschwunden.

2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  ocpp: trace

# unique installation id
plant: cad72b4092b041bc61aba40f41d7eea0afef97d674ad49ae6ce67499f69149c0

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.42  
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 10s  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.42  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 10s  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.42  
  port: 502  
  usage: battery  
  modbus: tcpip  
  timeout: 10s  
  name: battery3

chargers:
- type: template
  template: ocpp 
  stationid: A2292802444  
  name: wallbox5
#  remotestart: true

vehicles:
- type: template
  name: Smart
  template: smart-hello 
  title: 'Smart #1'  
  user: *****  
  password: *****  
  capacity: 66  
#  name: ev4

loadpoints:
- title: Carport
  charger: wallbox5
  vehicle: Smart # Referenz auf Standardfahrzeug
  mode: pv

tariffs:
  grid:
    type: tibber
    token: "*****" # access token

influx:
  url: http://192.168.178.86:8086
  database: evcc # InfluxDB v2.x uses term `bucket` but for compatibility still named `database` here
  token: *****
  org: *****

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 100
```

### Log details

```text
siehe log-files
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.11 (42fa9d26)

mfuchs1984, 2024-09-20:
Laut ocpp trace sieht es auf den ersten Blick eigentlich aus, als würde es funktionieren. Evcc gibt 6A vor und die wallbox reported für jede Phase 5.XA. Dennoch erscheint die Fehlermeldung im Log. Falsch ist jedoch, dass die wallbox 16A als Current.Offered meldet.

andig, 2024-09-20:
Thank you for the detailed bug report. You can test updated nightly in 30 minutes. Issue should be resolved.

andig, 2024-09-20:
> Falsch ist jedoch, dass die wallbox 16A als Current.Offered meldet.

Um das zu lösen bitte noch den fehlerhaften Wert entfernen, dafür:

    metervalues: -Current.Offered

damit evcc den ignorieren kann. Feedback of das funktioniert wäre Klasse, dann können wir für die Wallbox noch ein Template anlegen.

/cc @premultiply

closed 2024-09-20 by andig

shellfisch, 2024-09-21:
Der Bug ist doch noch nicht raus. Ich hänge deshalb nochmal die logs rein. 
OCPP-log
[evcc-20240921-122350-trace.log](https://github.com/user-attachments/files/17083743/evcc-20240921-122350-trace.log)
Debug-log
[evcc-20240921-122226-debug.log](https://github.com/user-attachments/files/17083746/evcc-20240921-122226-debug.log)
komplettes trace-log
[evcc-20240921-122124-trace.log](https://github.com/user-attachments/files/17083747/evcc-20240921-122124-trace.log)

Viel Erfolg und tausend Dank für eure tolle Arbeit.

mfuchs1984, 2024-09-21:
Hast du denn die aktuellste Nightly installiert?

andig, 2024-09-21:
Sicht nicht :)

shellfisch, 2024-09-21:
Doch das aktuelle Nightly istinstalliert

andig, 2024-09-21:
Könnten wir mal bitte ein Log haben mit unwichtigem Kram wie modbus, sma etc raus gefiltert? Ab Start? Was ist jetzt noch das Problem?

shellfisch, 2024-09-21:
Er bringt immer noch den Fehler: 
`[lp-1 ] WARN 2024/09/21 14:28:46 charger logic error: current mismatch (got 0.016A, expected 6A) `
Ich hoffe das passt: 
[evcc-20240921-143207-debug.log](https://github.com/user-attachments/files/17084053/evcc-20240921-143207-debug.log)
und OCPP
[evcc-20240921-143048-trace.log](https://github.com/user-attachments/files/17084054/evcc-20240921-143048-trace.log)

andig, 2024-09-21:
> Könnten wir mal bitte **ein** Log haben

?

> Ab **Start**? 

?

shellfisch, 2024-09-21:
[evcc-20240921-150753-debug.log](https://github.com/user-attachments/files/17084138/evcc-20240921-150753-debug.log)
[evcc-20240921-150840-trace.log](https://github.com/user-attachments/files/17084141/evcc-20240921-150840-trace.log)
 Ich bin super daran interessiert, das wir gemeinsam die Sache weiter entwickeln. Ich hoffe das passt jetzt. Leider bin ich jetzt die nächsten 2-3 Std. außer Haus. 
Good luck

mfuchs1984, 2024-09-22:
Deine Nightly ist zu alt

[main  ] INFO 2024/09/21 15:01:17 evcc 0.130.11 (42fa9d26)
Der zugehörige commit ist vom 16.09
https://github.com/evcc-io/evcc/commit/42fa9d263ebfe99787c372d446402bf227e0b9e8

Der entscheidende commit kam erst am 20. rein
https://github.com/evcc-io/evcc/commit/3d642004482dc33cd5939912e55c9e043ee6c722

Dazu dann noch 
https://github.com/evcc-io/evcc/commit/69f6bd1a47c8f5a4f6d58df337bec5b6a0f4e1c3

Also aktuelles nightly nehmen und `metervalues: -Current.Offered` in die config und es sollte laufen.

Wobei im neuesten Log wieder etwas ungewöhnlich ist, die Wallbox meldet bei Freigabe von 6 Ampere und mehr einen Strom zwischen 1 und 2 Ampere, was eigentlich nicht sein kann. In früheren logs waren es knapp 6.

Beispiele
`[lp-1  ] DEBUG 2024/09/21 15:06:20 charge currents: [1.8 1.7 1.7]A`

`[2,"87","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-09-21T13:06:33Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"16.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"1.8"},{"measurand":"Current.Import","unit":"A","phase":"L2","value":"1.7","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L3","value":"1.7","context":"Trigger"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"231.4","context":"Trigger"},{"measurand":"Voltage","unit":"V","phase":"L2","value":"231.0","context":"Trigger"},{"measurand":"Voltage","unit":"V","phase":"L3","value":"229.9","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"1071560.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"1238.0","context":"Trigger"}]}],"transactionId":1726923679}]`

premultiply, 2024-09-22:
> die Wallbox meldet bei Freigabe von 6 Ampere und mehr einen Strom zwischen 1 und 2 Ampere, was eigentlich nicht sein kann.

Warum nicht? Das hängt von Fahrzeug und dessen Zustand und ggf. auch interner Konfiguration ab.
Den Messwerten muss man schon trauen.

mfuchs1984, 2024-09-22:
Hast natürlich recht, gibt ja auch Fahrzeuge, die in bestimmten Zuständen weniger nehmen.

andig, 2024-09-22:
Alle Fahrzeuge- wenn der Akku fast voll ist!

mfuchs1984, 2024-09-22:
Hier hat das Auto laut log allerdings 84%. Aber vllt. ist ja ein internes Limit aktiv. Ist aber langsam OT, am Ende ist entscheidend, eine aktuelle Nightly version zu verwenden

shellfisch, 2024-09-22:
Bei mir scheint die aktuelle nightly-installation nicht zu funktionieren, denn nach 
`curl -1sLf 'https://dl.evcc.io/public/evcc/unstable/setup.deb.sh' | sudo -E bash`
kommt nur das 
`root@EVCC:~# evcc -v
evcc version 0.130.11 (42fa9d26)
root@EVCC:~# `
das würde einiges erklären

Zusatzfrage: wo kommt das `metervalues: -Current.Offered` hin? So:
`chargers:
- type: template
  template: ocpp 
  stationid: A2292802444  
  name: wallbox5
  metervalues: -Current.Offered`

sorry stecke nicht so tief drin

shellfisch, 2024-09-22:
Übrigens ja, es war ein internes Ladelimit aktiv: 85%

mfuchs1984, 2024-09-22:
`sudo apt update` und `sudo apt --only-upgrade install -y evcc` ausgeführt?

mfuchs1984, 2024-09-22:
Zu den metervalues, ja, siehe https://github.com/evcc-io/evcc/issues/16237#issuecomment-2363954689

shellfisch, 2024-09-22:
Danke, nightly hat jetzt funktioniert: `evcc version 0.130.11 (92e4f46e)`

aber mit den metervalues hilft mir das nicht weiter; weiß nicht was ich machen soll

mfuchs1984, 2024-09-22:
Genauso wie du es oben beschrieben hast soweit ich verstehe 

```
chargers:
  - type: template
    template: ocpp
    stationid: A2292802444
    name: wallbox5
    metervalues: -Current.Offered
```

andig, 2024-09-22:
Das sollte egtl. funktionieren.

shellfisch, 2024-09-22:
Es funktioniert, ganz großen Respekt vor Eurer Leistung; keine Fehler mehr im Log. 
sorry, dass ich beim dem NIghtly die falschen Knöpfe gedrückt und daher etwas von eurer Lebenszeit beansprucht habe. Wünsche euch noch ein schönes Restwochenende. Danke