evcc-io/evcc #16256: Fronius Symo Gen24 - Kein Preisbasiertes Netzladen des Hausakkus möglich

neupertd, 2024-09-21:
### Describe the bug

Sobald die Preisgrenze für das preisbasierte Laden des Hausakkus unterschritten wird erfolgt dieses nicht.
Die Anzeige im UI sagt aber, das es durchgeführt wird.

Was ich bereits versucht habe:
* Nutzen der Solarapi per http
* Nutzen von modbus mit den Templates sunspec-inverter und fronius-gen24

Mit fronius-gen24 und der Einstellung für Modbus am Wechselrichter auf float läuft die preisbasierte Fahrzeugladung ohne Probleme. Nur Hausakku geht in keiner Kombination

### Steps to reproduce

1. Wechseln der Preisgrenze über den aktuellen Preis
2. Im Log und im UI wird das korrekt bestätigt
3. Prüfung im WR zeigt, das keine Ladung erfolgt.
4.
...

### Configuration details

```yaml
Auszug des Wechselrichter-Bereiches

meters:
- type: template
  template: fronius-gen24 
    #template: sunspec-inverter
    #template: fronius-solarapi-v1
  usage: grid  
  host: 192.168.1.30  
  port: 502  
  name: grid1
    #modbus: tcpip
    #id: 200
    #user: technician
    #password: XXX
- type: template
  template: fronius-gen24 
    #template: sunspec-inverter
    #template: fronius-solarapi-v1
    #modbus: tcpip
    #id: 1
  usage: pv  
  host: 192.168.1.30  
  port: 502  
  name: pv2
    #user: technician
    #password: XXX
- type: template
  template: fronius-gen24 
    #template: sunspec-inverter
    #template: fronius-solarapi-v1
  usage: battery  
  host: 192.168.1.30  
  port: 502  
  name: battery3
    #modbus: tcpip
    #id: 1
    #user: technician
    #password: XXX
```

### Log details

```text
Sep 21 09:16:39 evcc evcc[372]: [site  ] DEBUG 2024/09/21 09:16:39 set grid charge limit: 0.2
Sep 21 09:16:39 evcc evcc[372]: [influx] TRACE 2024/09/21 09:16:39 write batteryGridChargeLimit=map[value:0.2] (map[])
Sep 21 09:16:40 evcc evcc[372]: [site  ] DEBUG 2024/09/21 09:16:40 ----
Sep 21 09:16:40 evcc evcc[372]: [openwb-2.0] TRACE 2024/09/21 09:16:40 modbus: send 00 0e 00 00 00 06 01 04 27 74 00 02
[... 93 lines cut ...]
Sep 21 09:16:40 evcc evcc[372]: [openwb-2.0] TRACE 2024/09/21 09:16:40 modbus: recv 00 11 00 00 00 05 01 04 02 00 00
Sep 21 09:16:40 evcc evcc[372]: [openwb-2.0] TRACE 2024/09/21 09:16:40 modbus: send 00 12 00 00 00 06 01 04 27 82 00 01
Sep 21 09:16:40 evcc evcc[372]: [openwb-2.0] TRACE 2024/09/21 09:16:40 modbus: recv 00 12 00 00 00 05 01 04 02 00 00
Sep 21 09:16:40 evcc evcc[372]: [lp-1  ] DEBUG 2024/09/21 09:16:40 charger status: A
Sep 21 09:16:40 evcc evcc[372]: [openwb-2.0] TRACE 2024/09/21 09:16:40 modbus: send 00 13 00 00 00 06 01 04 27 83 00 01
Sep 21 09:16:40 evcc evcc[372]: [openwb-2.0] TRACE 2024/09/21 09:16:40 modbus: recv 00 13 00 00 00 05 01 04 02 00 00
Sep 21 09:16:40 evcc evcc[372]: [openwb-2.0] TRACE 2024/09/21 09:16:40 modbus: send 00 14 00 00 00 06 01 04 27 82 00 01
Sep 21 09:16:40 evcc evcc[372]: [openwb-2.0] TRACE 2024/09/21 09:16:40 modbus: recv 00 14 00 00 00 05 01 04 02 00 00
Sep 21 09:16:40 evcc evcc[372]: [influx] TRACE 2024/09/21 09:16:40 write greenShareHome=map[value:1] (map[])
Sep 21 09:16:40 evcc evcc[372]: [influx] TRACE 2024/09/21 09:16:40 write greenShareLoadpoints=map[value:0] (map[])
Sep 21 09:16:40 evcc evcc[372]: [influx] TRACE 2024/09/21 09:16:40 write tariffPriceHome=map[value:0.0784] (map[])
Sep 21 09:16:40 evcc evcc[372]: [influx] TRACE 2024/09/21 09:16:40 write tariffPriceLoadpoints=map[value:0.1888] (map[])
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.11

andig, 2024-09-21:
>  I have verified that the issue is reproducible with the latest nightly build

Wirklich?

closed 2024-09-21 by andig

neupertd, 2024-09-21:
Ich muss mich hiermit entschuldigen. Hatte dieser Tage schon mal mit dem Night-Build versucht und da ging es noch nicht. Jetzt klappt es aber.
Bitte schließt den Issue. Damit sollte es ja in einem der nächsten Releases produktiv sein.

andig, 2024-09-21:
Genau- kommt asap. Danke fürs probieren!