evcc-io/evcc #16275: Passwort Reset not working @ RaspberryPi, 0.130.12

g33kt00l, 2024-09-22:
### Describe the bug

I cannot change my Password via command line like in the documentation mentioned:
evcc password set - Asked me for a new Password, after reboot of Raspberry Pi,  dit not work.
evcc password reset - tried reboot of raspberry pi, no popup to set new password, did not work

### Steps to reproduce

evcc password set - Asked me for a new Password, after reboot of Raspberry Pi,  dit not work.
evcc password reset - tried reboot of raspberry pi, no popup to set new password, did not work

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 1715a6f68106b6f0af5d3986924f1a8b6726e39bc51948847ef76c3ff28b4696

interval: 10s # control cycle interval

sponsortoken: 

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: solarlog 
  usage: grid  
  host: 10.10.33.222  
  port: 502  
  name: grid1
- type: template
  template: solarlog 
  usage: pv  
  host: 10.10.33.222  
  port: 502  
  name: pv2

chargers:
- type: template
  template: easee 
  user: ottofischer@me.com  
  password:
  charger: EHPBAQ2B  
  name: wallbox4

vehicles:
- type: template
  template: smart 
  title: SAD-RF 65 E  
  user: ottofischer@me.com  
  password:
  vin: W1A4533911K441921  
  capacity: 17.6  
  name: ev3

loadpoints:
- title: Unterstand & Hofeinfahrt
  charger: wallbox4
  vehicle: ev3
  mode: minpv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
#  resetOnDisconnect: true

site:
  title: Gabelsbergerstr. 33
  meters:
    grid: grid1
    pv:
    - pv2

# tariffs are the fixed or variable tariffs
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price (or price zones)
    type: fixed
    price: 0.2920 # EUR/kWh
    #zones:
      #- days: Mo-Fr
        #hours: 2-5
        #price: 0.2 # EUR/kWh
      #- days: Sa,So
        #price: 0.15 # EUR/kWh

    # or variable tariffs
    # type: tibber
    # token: "476c477d8a039529478ebd690d35ddd80e3308ffc49b59c65b142321aee963a4" # access token
    # homeid: "cc83e83e-8cbf-4595-9bf7-c3cf192f7d9c" # optional if multiple homes associated to account

    # type: awattar
    # region: de # optional, choose at for Austria
    # charges: # optional, additional charges per kWh
    # tax: # optional, additional tax (0.1 for 10%)

    # type: elering # Nordpool
    # region: ee # or lt, lv, fi
    # charges: # optional, additional charges per kWh
    # tax: # optional, additional tax (0.1 for 10%)
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.12 # EUR/kWh
  co2:
    # co2 tariff provides co2 intensity forecast and is for co2-optimized target charging if no variable grid tariff is specified
    # type: grünstromindex # GrünStromIndex (Germany only)
    # zip: <zip>

    # type: electricitymaps # https://app.electricitymaps.com/map
    # uri: <uri>
    # token: <token>
    # zone: DE
```

### Log details

```text
cannot export log because of missing password.
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.12

StefanSchoof, 2024-09-22:
See https://github.com/evcc-io/evcc/discussions/14862#discussioncomment-10398502

closed 2024-09-22 by andig

g33kt00l, 2024-09-23:
Danke und sorry‼️