evcc-io/evcc #16281: Improvement request: Tesla asleep should not lead to an error

endurance1968, 2024-09-23:
If a tesla is in sleep (power saving) a loadpoint creates an error (default car tesla). Since this is a normal status this should be only an info not an error. 
Not sure if we can distinghuis between sleep and fully offline/car not available at all.

The error is annoying as you always check the logs.... for some serious things.

evcc 1.130.12

```
Sep 23 04:46:03 lx-evcc evcc[321710]: [tesla ] TRACE 2024/09/23 04:46:03 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/xxx/vehicle_data
Sep 23 04:46:03 lx-evcc evcc[321710]: [tesla ] TRACE 2024/09/23 04:46:03 {"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":"","txid":"88c13680d24d53ddd7afda381411e390"}
Sep 23 04:46:03 lx-evcc evcc[321710]: [lp-2  ] ERROR 2024/09/23 04:46:03 vehicle odometer: asleep
```

andig, 2024-09-23:
Is Odometer the only error?

closed 2024-09-23 by andig

endurance1968, 2024-09-23:
Hi saw only the odometer yet.