evcc-io/evcc #16284: Load management should not generate alarm notifications

RikMe, 2024-09-23:
**Is your feature request related to a problem? Please describe.**
Even though it's still experimental, i very much like the load management feature! It does however generate alarm notifications when it's activated. This could raise that the impression that something is 'wrong' while the system is just reacting 'as intended'

**Describe the solution you'd like**
No alarm notification when load management is working as intended 

[image]

andig, 2024-09-23:
Understand. It would be great to collect some opinions here. It's a warning, not an error.

RikMe, 2024-09-23:
Jup, but it gave me the 'impression' as if something was wrong when i first saw it. While actually there was nothing wrong and EVCC worked exactly as intended.

Roeland54, 2024-09-23:
As user of the load management feature I agree on this. The warning gives the impression something is not optimal or needs changing but that is not the case.  Already mentioned this[ here ](https://github.com/evcc-io/evcc/issues/14261#issuecomment-2245807478)in the past.

RenatusRo, 2024-09-23:
for the time being leave it as it is now, 
later move as visualization to site and LP UX
as planned in epic: https://github.com/evcc-io/evcc/issues/14261

andig, 2024-09-28:
No consensus, declined.

closed 2024-09-28 by andig