evcc-io/evcc #16296: go-e v2 (API v1) geladene Energie feiner aufgelöst anzeigen

tm-107, 2024-09-24:
Hallo,

beim go-e v2 wird die geladene Energie in evcc immer nur in vollen kWh angezeigt. Somit kann ich nie genau sagen, ob es nun zum Beispiel 2,0 oder 2,9 kWh waren ...
Bei unserem PHEV ohne Möglichkeit der Online-Ladestandsabfrage (Mitsubishi Outlander) wäre das aber schon ein Unterschied von knapp 10% SOC.

Ich nehme an, hier wird in evcc die Differenz der geladenen Gesamt-Energiemenge zur Anzeige genutzt (API-Wert "eto"). Dieser Wert wird entgegen der Dokumentation nicht in 0,1 kWh-Schritten erhöht, sondern immer nur volle kWh (also immer um 10).

Wäre es möglich stattdessen die Energiemenge des aktuellen Ladevorgangs (API-Wert "dws") zur Anzeige der geladenen Energie zu nutzen? Dieser wird in Deka-Watt-Sekunden aufgelöst und erlaubt somit eine wesentlich genauere Anzeige.
Evcc ermittelt diesen Wert auch, aber nutzt ihn scheinbar nicht zur Anzeige der geladenen Energiemenge.

thx

mfuchs1984, 2024-09-24:
Das würde bei der go-e aufgrund von Problemen mal geändert
https://github.com/evcc-io/evcc/issues/13726
https://github.com/evcc-io/evcc/issues/13390
https://github.com/evcc-io/evcc/pull/14730

closed 2024-09-24 by andig

tm-107, 2024-09-24:
Okay, schade!
Das eigentliche Problem liegt natürlich bei go-e, aber da brauche ich wohl kein Update mehr erwarten.

Ich habe vorhin übrigens 1,081 kWh geladen (gemessen über ein extra angeschlossenen Shelly, dessen Werte sich bis auf die go-e v2 übliche Abweichung von ~6% mit dem dws-Wert von der go-e deckt).
Der Gesamtzähler (eto) hat sich dabei leider nicht verändert, wodurch evcc mir 0,0 kWh anzeigt ... und das nach knapp 50 Minuten mit 1,3kW.

Zur genauen Auswertung werde ich mir jetzt auf dem RaspberryPi ein kleines Script schreiben, welches den dws-Wert ermittelt und nach jedem Ladevorgang (also wenn dws-neu > dws-alt) speichert. Bin gespannt wie die Gesamtabweichung nach einigen Wochen ausfällt.