evcc-io/evcc #16300: Kostal Plenticore ext. Batteriesteuerung blockiert BYD Batterie Stromabgabe

WernerRH21, 2024-09-24:
### Describe the bug

Der Ladevorgang via evcc funktioniert einwandfrei und im Schnelllademodus wird der Akku blockiert und visuell in der Oberfläche angezeigt. Nach Beendigung des Ladevorgangs verschwindet auch die Anzeige des gesperrten Akkus. Seitdem gibt der Akku aber seinen Strom nicht mehr ab.
Was ist zu tun, um dies wieder zu ermöglichen?
[image]
[image]
[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  schema: http
  host: ...
  port: 7070

interval: 30s 

log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

meters:
  - name: grid
    type: template
    template: kostal-ksem
    usage: grid
    modbus: tcpip
    id: 71
    host: 10.21.10.204 
    port: 502  
  - name: pv1
    type: template
    template: kostal-plenticore
    usage: pv
    endianness: little
    capacity: 5.6
    minsoc: 5
    watchdog: 60s
    modbus: tcpip
    id: 71
    host: 10.21.100.31 
    port: 1502
  - name: pv2
    type: template
    template: kostal-plenticore
    usage: pv
    endianness: little
    capacity: 2.625
    modbus: tcpip
    id: 71
    host: 10.21.100.32 
    port: 1502 
  - name: byd
    type: template
    template: kostal-plenticore
    usage: battery
    modbus: tcpip
    id: 71
[... 355 lines cut ...]
Sep 07 03:27:57 evcc evcc[1552]: [lp-1  ] DEBUG 2024/09/07 03:27:57 detected connected phases: 3p
Sep 07 03:27:57 evcc evcc[1552]: [lp-1  ] DEBUG 2024/09/07 03:27:57 charge total import: 2958.000kWh
Sep 07 03:27:57 evcc evcc[1552]: [lp-1  ] DEBUG 2024/09/07 03:27:57 charger status: B
Sep 07 03:27:57 evcc evcc[1552]: [lp-1  ] DEBUG 2024/09/07 03:27:57 charge start energy: 2958.000kWh
Sep 07 03:27:57 evcc evcc[1552]: [lp-1  ] INFO 2024/09/07 03:27:57 car connected
Sep 07 03:27:57 evcc evcc[1552]: [lp-1  ] DEBUG 2024/09/07 03:27:57 vehicle api refresh
Sep 07 03:27:57 evcc evcc[1552]: [lp-1  ] DEBUG 2024/09/07 03:27:57 pv timer elapse
Sep 07 03:27:57 evcc evcc[1552]: [lp-1  ] DEBUG 2024/09/07 03:27:57 pv timer inactive
Sep 07 03:27:57 evcc evcc[1552]: [lp-1  ] DEBUG 2024/09/07 03:27:57 charge total import: 2958.000kWh
Sep 07 03:28:25 evcc evcc[1552]: [site  ] DEBUG 2024/09/07 03:28:25 ----
Sep 07 03:28:25 evcc evcc[1552]: [lp-1  ] DEBUG 2024/09/07 03:28:25 charge power: 0W
Sep 07 03:28:25 evcc evcc[1552]: [lp-1  ] DEBUG 2024/09/07 03:28:25 charge currents: [0 0 0]A
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.12

frankheit, 2024-09-24:
Selbes Problem hier (Kostal Plenticore 10, KSEM, BYD Speicher).
Speicher mit Netzstrom laden funktioniert, allerdings scheint es, als würde der Plenticore danach nicht in seine eigene Steuerung "zurückfallen" und nutzt nach Deaktivierung des Netzladens seinen Speicher nicht mehr.
Einzige Möglichkeit die ich gefunden habe damit das System wieder normal funktioniert war die manuelle Umstellung des Plenticore auf "interne" Batteriesteuerung.

andig, 2024-09-24:
@iseeberg79 in https://github.com/evcc-io/evcc/pull/15709/files Sieht „normal“ so aus wie „hold“. Das dürfte falsch sein?

WernerRH21, 2024-09-24:
Oh man, das hört sich gar nicht gut an. Für die interne Umstellung brauche ich ja wieder meinen Solar-Installateur, der dafür Geld nimmt. Das Problem sollten dann doch einige haben.

premultiply, 2024-09-24:
Der WR resettet sich in "normal" über das _Ausbleiben_ der Registerzugriffe nach Ablauf des Timeouts.

Bitte drauf achten, dass die Werte für Timeout und watchdog in der evcc-Konfig übereinstimmen.
Je kleiner die Werte desto schneller sollte der WR wieder in den Normalbetrieb zurückwechseln.

andig, 2024-09-24:
@WernerRH21 wie lange besteht das Problem denn? Löst sich das auch nach 1min nicht auf?

frankheit, 2024-09-24:
Ich hab ca. 5h gewartet bis ich dann eingegriffen habe :-(

WernerRH21, 2024-09-24:
@andig Seit knapp einer Woche

WernerRH21, 2024-09-24:
@premultiply watchdog steht auf 60s. Wo und wird das timeout eingetragen? Oder meinst du mit timeout den watchdog Eintrag?

premultiply, 2024-09-24:
```
    minsoc: 5
```
Ist _deprecated_. Soll raus.

```
    watchdog: 60s
```
Muss zur Timeout-Konfiguration des WR passen.

Ich mutmaße mal, dass ein Update durchgeführt wurde und der WR ggf. noch den alten MinSoC gesetzt hat?

WernerRH21, 2024-09-24:
@premultiply 
minSOC Einträge wurden entfernt und das Timeout im WR auf 60s geändert.
Bin gespannt.

premultiply, 2024-09-24:
Falls `minsoc` im WR noch gesetzt ist nutzt das auch nix.
Da müssen erst die Werte in der WR-Konfig auf ihre Normalwerte korrigiert werden.

WernerRH21, 2024-09-24:
Was sind denn die Normalwerte?
 
[image]

andig, 2024-09-24:
Ein minsoc von 5%- so wie im Screenshot dargestellt- kann kein Problem darstellen. Stellt sich also die Frage, aufgrund welchen Parameters Dein WR (und anscheinend auch andere) nicht mehr entladen wollen?

Zwei Ideen:
- mal evcc stoppen- ändert das (nach 1min) was?
- mal Steuerung von ext. Modbus auf etwas andere umstellen- ändert das (nach 1min) was?

Dazu Frage an Hersteller: warum entlädt der WR nicht obwohl er keine Sperre der Batterie anzeigt?

WernerRH21, 2024-09-24:
evcc gestoppt ... Verhalten bleibt unverändert und es wird weiterhin ext. Batteriesteuerung aktiv im WR angezeigt. So werden momentan 12.8 kWh Speicher fast nicht genutzt. 
[image]

WernerRH21, 2024-09-24:
Die Steuerung auf modbus umzustellen hat mich knapp 100€ gekostet.

andig, 2024-09-24:
Ich sehe "Batteriestatus normal". Die externe Steuerung läuft nicht. Wo ist also das Problem? Was sagt der Hersteller?

WernerRH21, 2024-09-24:
Das Problem ist, dass der Akku bei über 90% stehen bleibt und die ganze Nacht kostenpflichtiger Strom bezogen wird, obwohl fast 13 kWh zur Verfügung stehen. Auch lädt er tagsüber bei PV Überschuss nicht mehr auf 100% voll. Es passiert einfach nichts mehr mit dem Akku.
Ich schließe mal gerade ein Auto an und lade im Schnelllademodus. Hierbei greift ja die Batteriesteuerung und die Entladung des Akkus wird gesperrt. Mal schauen, wie die Anzeige im WR dann aussieht.
Wenn alles nichts hilft, wende ich mich natürlich an Kostal.

andig, 2024-09-24:
> Das Problem ist, dass der Akku bei über 90% stehen bleibt 

Gleiche Frage wie oben. Warum sollte er es unter den Bedingungen tun? Was sagt der Hersteller?

frankheit, 2024-09-24:
Ich vermute hier leider auch eher ein Kostal als ein evcc Problem.
Hab wie gesagt den gleichen Fehler, ich vermute der Plenticore fällt einfach nicht mehr in seinen "normalen" Modus zurück wenn kein Befehl per Modus mehr kommt sondern bleibt irgendwo hängen.
Habe das vor ner Weile auch schon mal im PV-Forum thematisiert, leider gab es auch da keine Lösung die ich nachvollziehen konnte (ab Seite 5):
https://www.photovoltaikforum.com/thread/179492-kostal-plenticore-bi-externe-batteriesteuerung-freischalten/?pageNo=5

andig, 2024-09-24:
Dann bitte an den Herstellersupport wenden. Wir können da leider nicht weiter helfen- aus Sicht Anwendung sieht es korrekt aus.

WernerRH21, 2024-09-24:
Vielen Dank für deine Hilfe @andig. Ich sehe es mittlerweile auch so, dass das Problem bei Kostal liegt.

WernerRH21, 2024-09-24:
Es scheint ein Herstellerproblem zu sein.

StevieC121176, 2024-09-24:
Also bei mir funktioniert das. Welche Software hast du auf dem Plenticore?

WernerRH21, 2024-09-24:
Hi @StevieC121176
Das macht Hoffnung.
[image]

StevieC121176, 2024-09-24:
Ich hab noch nicht upgedatet und hab noch die Version zuvor auf dem WR.

[image]

WernerRH21, 2024-09-24:
Dann mache ich mich mal auf die Suche, um downzugraden. Danke dir.

deadrabbit87, 2024-09-24:
Ich habe die `01.30.12092` und bei mir funktioniert es trotzdem - also an der FW scheint es mir nicht zu liegen. 

Hast du sicher nichts anderes am laufen, was irgendwie die Modbus Register des WR beschreibt?

WernerRH21, 2024-09-24:
@deadrabbit87 Bei dir sehe ich die 200er HW Version. Ich habe eine 100er Hardware.
Welche Version hast du @StevieC121176 ?

WernerRH21, 2024-09-24:
@deadrabbit87 Ich habe sogar meinen unraid Server komplett abgeschaltet, damit wirklich nichts per via modbus läuft.
Vielleicht schalte ich mal den WR und die Batterie komplett aus und starte beides neu.

deadrabbit87, 2024-09-24:
Wenn übrigens der WR selbst regelt, dann sieht das bei mir so aus: 
[image]

Ich denke daher immer noch, dass irgendwas noch deinen WR beschreibt. Eventuell eine zweite evcc Instanz? 

Was ist im Kostal Energy Meter eingestellt? Ist nur eine Vermutung, aber ich meine das kann da auch was schreiben...