evcc-io/evcc #16309: Modus "Schnell" ohne Funktion & Ladeabbrüche - Bender

CKlapper, 2024-09-24:
### Describe the bug

Hallo, ich habe akut 2 große Probleme festgestellt:

Cupra Formentor, 1phasig, 
Bender (Mennekes)
EEBus

Mit v.0.130.6 - v.0.130.12 (evtl aber auch mit niedriger Verionsnummern)
1. Modus "Schnell" funktioniert nicht mehr. Laden bricht ab wenn man "Schnell" aktiviert. Bzw. startet gar nicht erst. Mindestens seit Version 130.6
Ebenfalls in 130.12 hat der Ladeplan das Auto Nachts nicht geladen. Denke ist das gleiche Problem. (Log1)

[log1 - evcc-20240924-224600-trace.log](https://github.com/user-attachments/files/17121259/log1.-.evcc-20240924-224600-trace.log)

2. Mit v.0.130.12
Im "PV" Modus während die Dämmerung einsetzte, brach das Laden ab, startete wieder, brach ab usw.
Der Hausakku war dabei noch >95% und durfte bis <90% genutzt werden. Die PV Leistung schwankte enorm. Ich meine beobachtet zu haben, dass das Laden abrupt stoppte sobald die PV Leistung kleiner der Ladeleistung des Autos war (3,6kw). Und wieder startete sobald genug PV Leistung zu Verfügung stand. (Log2) 
Als dann die PV Leistung dauerhaft unterhalb der Ladeleistung lag hat er brav den Akku mitbenutzt und normal bis zur Akku Grenze geladen.

[log2 - evcc-20240924-180303-trace.log.txt](https://github.com/user-attachments/files/17121277/log2.-.evcc-20240924-180303-trace.log.txt)

Hatte ja bereits letztens schonmal ein EEbus Problem https://github.com/evcc-io/evcc/issues/15246
vielleicht hängt das wieder damit zusammen ?

### Steps to reproduce

siehe Beschreibung

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db

log: debug

# unique installation id
plant: 6a9108443f558ede6279a487f79bde8d52cf63de0139ce3554c59e10b17a0bd7

interval: 15s # control cycle interval 10s Standart

sponsortoken: e

meters:
- name: my_grid
  type: template
  template: goodwe-hybrid 
  id: 247  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: grid
  modbus: rs485serial  
- name: my_pv 
  type: template
  template: goodwe-hybrid 
  id: 247  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: pv  
  modbus: rs485serial  
- name: my_battery 
  type: template
  template: goodwe-hybrid 
  id: 247  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: battery  
  modbus: rs485serial  
  
chargers:
- name: my_wallbox
  type: template
  template: eebus 
  ski: 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3
  host: 192.168.0.71 
  
vehicles:
- type: template
  template: peugeot 
  title: e208 
  user: christian-klapper@gmx.de  
  password: Myp3ug07()  
  vin: VR3UHZKXZNT032746
  accesstoken: 4b07e6d7
  refreshtoken: 4b02f44d
  capacity: 46.1
  phases: 3
  mode: PV
  minCurrent: 6
  maxCurrent: 16
  name: my_peugot
- type: template
  template: seat 
  title: Cupra  
  user: esther-klapper@web.de
  password: F0rm3n70r#
  vin: VSSZZZKMZNR066489
  capacity: 10.4
  phases: 1
  mode: PV
  minCurrent: 14
  maxCurrent: 16
  name: my_cupra 
  
  

loadpoints:
- title: Garage
  charger: my_wallbox
  mode: pv
  guardduration: 5m  

site:
  title: Zuhause
  meters:
     grid: my_grid
     pv:
     - my_pv
     battery:
     - my_battery
  residualPower: 100  
  maxGridSupplyWhileBatteryCharging: 100
  
eebus:
  shipid: EVCC
  interfaces:
  - eth0
  certificate:
    public: |
      -----BEGIN CERTIFICATE-----
      

      -----END CERTIFICATE-----
      
    private: |
      -----BEGIN EC PRIVATE KEY-----
      
      -----END EC PRIVATE KEY-----
```

### Log details

```text
siehe Log Files
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.12

andig, 2024-09-25:
> Laden bricht ab wenn man "Schnell" aktiviert.

Wo sehe ich das im Log?

andig, 2024-09-25:
Ich sehe- völlig unabhängig von eebus- im Log:

    [lp-1  ] DEBUG 2024/09/25 14:06:26 charger disable

CKlapper, 2024-09-25:
Ich bin direkt von "Min+PV" auf "Schnell" umgeschaltet. Nicht auf "Aus". 
Schalte ich dann wieder zurück auf "Min+PV" startet das laden auch wieder. Ich kann das Verhalten jederzeit so wiederholen.

andig, 2024-09-25:
Spannend. Vorher sehe ich:

```
[eebus ] TRACE 2024/09/25 14:05:02 Recv: 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.2.0"},{"addressSource":[{"device":"d:_i:4711_CC612_2S0R_CC"},{"entity":[1,1]},{"feature":10}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-EVCC"},{"entity":[1]},{"feature":7}]},{"msgCounter":2172525},{"msgCounterReference":32},{"cmdClassifier":"reply"}]},{"payload":[{"cmd":[[{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0}],[{"limitId":1}],[{"limitId":2}]]}]}]]}]}]}}]}
```

Da kommen alle Limits als "inaktiv" zurück, mithin wäre der Ladepunkt "enabled".

@DerAndereAndi Im Log ist nicht so einfach zu erkennen, welcher UC das ist (vmtl. über das Feature?). Falls möglich wäre es cool, das in den Request/Reply Zeilen mit auszugeben.

Für mich sieht es aus als würde `Enabled() == false` raus kommen und evcc daher den internen State korrigieren. Ich sehe allerdings nicht, warum das der Fall sein sollte.

mfuchs1984, 2024-09-25:
Das?
https://github.com/enbility/eebus-go/pull/110

andig, 2024-09-25:
Bitte nicht raten. Wir haben ja seitdem enbility geupdated.

andig, 2024-09-25:
@CKlapper wäre schön beim nächsten Log modbus, peugeut, db und anderen unrelated Kram raus zu lassen- machts viel leichter zu lesen...

mfuchs1984, 2024-09-25:
> Bitte nicht raten. Wir haben ja seitdem enbility geupdated.

Hast recht, ist ja schon drin.

DerAndereAndi, 2024-09-25:
@andig die Events sagen dir für welchen Usecase Daten übermittelt wurden.

```
[eebus ] DEBUG 2024/09/25 14:06:27 Recv: d:_i:4711_CC612_2S0R_CC:LoadControl to LoadControl notify 2172949 LoadControlLimitListData
[eebus ] DEBUG 2024/09/25 14:06:27 ski 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 event cem-opev-DataUpdateLimit
```

Heisst hier OPEV -> Emobility.

> Für mich sieht es aus als würde Enabled() == false raus kommen und evcc daher den internen State korrigieren. Ich sehe allerdings nicht, warum das der Fall sein sollte.

Wenn Enabled false wäre, würde er nicht das Laden stoppen wollen sondern starten. Oder nicht? Aber es wird das Laden ja aktiv gestoppt. Er schreibt ein aktives Limit mit 0A. Das passiert nur wenn `Enable(false)` aufgerufen wird.

mfuchs1984, 2024-09-25:
So, dieses mal kein Raten, sondern code angeschaut, log angeschaut und nachgestellt:
- du hast beim Seat minCurrent: 14 A
- du hast beim loadpoint in der GUI "maximaler Ladestrom": 13 A

Dieser Konflikt führt dazu, dass im schnell Modus nicht geladen werden kann. 

Aus der Funktion pvMaxCurrent, die im PV und minPV Modus verwendet wird, kommt, wenn man eine Batterie hat, die minCurrent raus, das ist 14A aufgrund des Fahrzeugs. Die Funktion setLimit akzeptiert diesen Wert weil er größer gleich lp.effectiveMinCurrent() (14 A) ist und die Ladung startet. 

Im Fall "schnell" wird die Funktion fastCharging() verwendet, diese gibt lp.effectiveMaxCurrent zurück (13 A). Das ist kleiner lp.effectiveMinCurrent() und die Ladung wird in setLimit nicht gestartet bzw. beendet.

andig, 2024-09-25:
Stark 👍🏻. Hätte ich NIE gefunden. Jetzt müssen wir mal überlegen wie sich so ein Blödsinn besser abfangen/ debuggen liessen.

mfuchs1984, 2024-09-25:
Man könnte ohne tiefer über die Logik nachzudenken zumindest schonmal eine warning einbauen, wenn effectiveMaxCurrent < effectiveMinCurrent, ggf. in der loadpoint Update Funktion.

CKlapper, 2024-09-26:
Ich danke euch, da habe ich ja einen bösen Fehler eingebaut... Asche auf mein Haupt.

Grund des ganzen ist tatsächlich, dass mein Goodwe Wechselrichter ein Problem hat und bei einer Gesamtleitung >10kw die Erzeugungsleistung auf 0 schaltet und alles aus der Hausbatterie nimmt. Goodwe selbst sucht hier noch den Fehler....

Darum habe ich vor ein paar Tagen den Ladestrom begrenzt damit der Peugot nicht über 8 -9 kw geht. Das sich das dann mit den Werten vom Cupra beißt hatte ich nicht bedacht.

closed 2024-10-03 by andig