evcc-io/evcc #16326: Active Battery Control FoxESS H3 MinSoC not changed back after charging is finished

Gics78, 2024-09-25:
### Describe the bug

Active Battery Control to protect the battery from discharge protection seem to work in setting the minimal SoC (MinSoc) to the current or prespecified value. But after charging is completed it is not reset to the original value. So the battery is essentially crippled because it will not discharge any further until I change the MinSoc manually.
[site ] ERROR  battery mode: read tcp 172.23.0.2:45672->192.168.178.8:502: i/o timeout

### Steps to reproduce

1. Beginn Charging
2.End charging
3. Check MinSoc, set to current SoC and not the typical MinSoc
...

### Configuration details

```yaml
H3 10.0.E Firmware 2.01., Slave 1.03.; BMS 1.017
```

### Log details

```text
[ocpp ] TRACE 2024/09/25 19:58:59 handling incoming CALL [63200794-c46d-4772-ae13-dac67cbf4b13, MeterValues] from 60AH73B037AQ016
[ocpp ] TRACE 2024/09/25 19:58:59 sent CALL RESULT [63200794-c46d-4772-ae13-dac67cbf4b13] for 60AH73B037AQ016
[ocpp ] TRACE 2024/09/25 19:58:59 sent JSON message to 60AH73B037AQ016: [3,"63200794-c46d-4772-ae13-dac67cbf4b13",{}]
[lp-1 ] DEBUG 2024/09/25 19:59:02 set charge mode: now
[site ] DEBUG 2024/09/25 19:59:02 ----
[lp-1 ] DEBUG 2024/09/25 19:59:02 charge power: 0W
[lp-1 ] DEBUG 2024/09/25 19:59:02 charge currents: [0 0 0]A
[modbus] TRACE 2024/09/25 19:59:02 modbus: send 39 4e 00 00 00 06 01 03 79 1a 00 01
[modbus] TRACE 2024/09/25 19:59:02 modbus: recv 39 4e 00 00 00 05 01 03 02 00 00
[modbus] TRACE 2024/09/25 19:59:02 modbus: send 39 4f 00 00 00 06 01 03 79 1d 00 01
[modbus] TRACE 2024/09/25 19:59:02 modbus: recv 39 4f 00 00 00 05 01 03 02 00 00
[site ] DEBUG 2024/09/25 19:59:02 pv power: 0W
[modbus] TRACE 2024/09/25 19:59:02 modbus: send 39 50 00 00 00 06 01 03 79 3c 00 01
[modbus] TRACE 2024/09/25 19:59:02 modbus: recv 39 50 00 00 00 05 01 03 02 02 a4
[modbus] TRACE 2024/09/25 19:59:02 modbus: send 39 51 00 00 00 06 01 03 79 3e 00 01
[modbus] TRACE 2024/09/25 19:59:02 modbus: recv 39 51 00 00 00 05 01 03 02 00 5c
[site ] DEBUG 2024/09/25 19:59:02 battery soc: 92%
[site ] DEBUG 2024/09/25 19:59:02 battery power: 676W
[modbus] TRACE 2024/09/25 19:59:02 modbus: send 39 52 00 00 00 06 01 03 79 32 00 01
[modbus] TRACE 2024/09/25 19:59:02 modbus: recv 39 52 00 00 00 05 01 03 02 ff 3f
[modbus] TRACE 2024/09/25 19:59:02 modbus: send 39 53 00 00 00 06 01 03 79 33 00 01
[modbus] TRACE 2024/09/25 19:59:02 modbus: recv 39 53 00 00 00 05 01 03 02 00 31
[modbus] TRACE 2024/09/25 19:59:02 modbus: send 39 54 00 00 00 06 01 03 79 34 00 01
[modbus] TRACE 2024/09/25 19:59:02 modbus: recv 39 54 00 00 00 05 01 03 02 00 91
[site ] DEBUG 2024/09/25 19:59:02 grid meter: -1W
[modbus] TRACE 2024/09/25 19:59:02 modbus: send 39 55 00 00 00 06 01 03 7d 0c 00 02
[modbus] TRACE 2024/09/25 19:59:02 modbus: recv 39 55 00 00 00 07 01 03 04 00 00 e4 e8
[site ] DEBUG 2024/09/25 19:59:02 site power: 675W
[lp-1 ] DEBUG 2024/09/25 19:59:02 charge voltages: [232 233 234]V
[lp-1 ] DEBUG 2024/09/25 19:59:02 detected connected phases: 3p
[lp-1 ] DEBUG 2024/09/25 19:59:02 charge total import: 133.150kWh
[lp-1 ] DEBUG 2024/09/25 19:59:02 charger status: B
[ocpp ] TRACE 2024/09/25 19:59:02 enqueued CALL [3997912919, SetChargingProfile] for 60AH73B037AQ016
[ocpp ] TRACE 2024/09/25 19:59:02 dispatched request 3997912919 for 60AH73B037AQ016
[ocpp ] TRACE 2024/09/25 19:59:02 sent JSON message to 60AH73B037AQ016: [2,"3997912919","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":64,"stackLevel":9,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-09-25T17:59:02Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":16}]}}}]
[ocpp ] TRACE 2024/09/25 19:59:02 started timeout timer for 60AH73B037AQ016
[ocpp ] TRACE 2024/09/25 19:59:03 received JSON message from 60AH73B037AQ016: [3,"3997912919",{"status":"Accepted"}]
[ocpp ] TRACE 2024/09/25 19:59:03 handling incoming CALL RESULT [3997912919] from 60AH73B037AQ016
[ocpp ] TRACE 2024/09/25 19:59:03 completed request 3997912919 for 60AH73B037AQ016
[lp-1 ] DEBUG 2024/09/25 19:59:03 max charge current: 16A
[site ] DEBUG 2024/09/25 19:59:03 ----
[lp-1 ] DEBUG 2024/09/25 19:59:03 charge power: 0W
[ocpp ] TRACE 2024/09/25 19:59:03 60AH73B037AQ016 ready to transmit again
[lp-1 ] DEBUG 2024/09/25 19:59:03 charge currents: [0 0 0]A
[modbus] TRACE 2024/09/25 19:59:03 modbus: send 39 56 00 00 00 06 01 03 79 1a 00 01
[ocpp ] TRACE 2024/09/25 19:59:03 timeout canceled for 60AH73B037AQ016
[modbus] TRACE 2024/09/25 19:59:03 modbus: recv 39 56 00 00 00 05 01 03 02 00 00
[modbus] TRACE 2024/09/25 19:59:03 modbus: send 39 57 00 00 00 06 01 03 79 1d 00 01
[modbus] TRACE 2024/09/25 19:59:03 modbus: recv 39 57 00 00 00 05 01 03 02 00 00
[... 478 lines cut ...]
[modbus] TRACE 2024/09/25 20:00:53 modbus: send 39 c7 00 00 00 06 01 03 79 33 00 01
[modbus] TRACE 2024/09/25 20:00:53 modbus: recv 39 c7 00 00 00 05 01 03 02 ff 16
[modbus] TRACE 2024/09/25 20:00:53 modbus: send 39 c8 00 00 00 06 01 03 79 34 00 01
[modbus] TRACE 2024/09/25 20:00:53 modbus: recv 39 c8 00 00 00 05 01 03 02 ff 69
[site ] DEBUG 2024/09/25 20:00:53 grid meter: 863W
[modbus] TRACE 2024/09/25 20:00:53 modbus: send 39 c9 00 00 00 06 01 03 7d 0c 00 02
[modbus] TRACE 2024/09/25 20:00:53 modbus: recv 39 c9 00 00 00 07 01 03 04 00 00 e4 eb
[site ] DEBUG 2024/09/25 20:00:53 site power: 863W
[lp-1 ] DEBUG 2024/09/25 20:00:53 charge voltages: [232 233 234]V
[lp-1 ] DEBUG 2024/09/25 20:00:53 detected connected phases: 3p
[lp-1 ] DEBUG 2024/09/25 20:00:53 charge total import: 133.360kWh
[lp-1 ] DEBUG 2024/09/25 20:00:53 charger status: B
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.12

thse22, 2024-09-25:
Hi,
can you check your evcc.yaml configuration. You need a minsoc value. It does not remember your old minsoc...
I use the following:

- type: template
  template: fox-ess-h3 
  id: 247  
  host: 192.168.1.7  
  port: 502  
  usage: battery  
  modbus: tcpip  
  minsoc: 10
  maxsoc: 95  
  name: battery1

andig, 2024-09-25:
> You need a minsoc value

No. Evcc has safe defaults of 20/95% for that.

Gics78, 2024-09-26:
Unfortunately andig is correct. Minsoc in .yaml doesn't change anything, still the same problem and the same battery mode timeout errors.

andig, 2024-09-26:
Die Frage hier ist doch, warum setzen des Battery Mode zu einem Fehler führt:

    [site ] ERROR 2024/09/25 19:59:33 battery mode: read tcp 172.23.0.2:34104->192.168.178.8:502: i/o timeout

Lässt sich das auch mit

    evcc charger

nachvollziehen?

/cc @thse22

Gics78, 2024-09-26:
Teste ich nachher!
Und die Anschlussfrage: das initiale Setzen des MinSoc durch EVCC scheint ja trotz Fehlermeldung zu funktionieren, das am Ende des Ladevorgangs aber nicht. Warum?

Gics78, 2024-09-26:
> Lässt sich das auch mit
> `evcc charger`
>nachvollziehen? 

Ja, selber Modbus Timeout Fehler.
Ich habe mal interessehalber die [HA Modbus-Integration](https://github.com/nathanmarlor/foxess_modbus) installiert. Mit der kann ich den MinSoc bzw. auch den Workmode (self-use -> Backup) erfolgreich steuern, allerdings kommt bei Workmode  (nicht beim Einstellen des MinSoc) auch eine Fehlermeldung zurück ` Start: 41000; values: [2]; slave: 1 from tcp://192.168.178.8:502: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)`

Vielleicht ein ähnliches Problem?

andig, 2024-09-26:
Tja. Jetzt ist die Frage warum? Was sagt der Hersteller?

Gics78, 2024-09-28:
FoxEss sagt dass es eine offene [API](https://www.foxesscloud.com/public/i18n/en/OpenApiDocument.html) gibt. 
Nachdem ich aber stark annehme dass EVCC das lokal über den Modbus umsetzt ist doch vielmehr die Frage wie das umgesetzt wurde? 
Möglicherweise bestehen die Probleme weil FoxEss mit neueren Firmware die Register verändert hat? Einen Changelog dazu gibt es leider nie.
Ich versuche nochmal die aktuellen Register bei Foxess nachzufragen (die mann dann aber nie veröffentlichen darf).

andig, 2024-09-28:
Wozu fehleranfällige Cloud APIs wenn man das Problem lokal lösen kann? Ihr seid die Kunden- macht Druck beim Hersteller! Ansonsten müsste jemand die Cloud API implementieren. Seitens Core Team gibts keine Kapazität schlechtere Lösungen als notwendig zu bauen.

Gics78, 2024-09-28:
@ andigDa bin ich voll bei Dir!

CPU-6502, 2024-10-07:
Hallo, ich kann dieses Verhalten bestätigen, der MinSoC bleibt nach dem beenden des Schnellladevorgangs auf dem aktuellen SoC der Batterie und wird nicht auf den ursprünglichen Wert zurückgesetzt. (siehe log-file)

[evcc-20241007-141246-trace.log](https://github.com/user-attachments/files/17278329/evcc-20241007-141246-trace.log)

Ich habe auch versucht, das entsprechende Holding-Register (41009) mit Hilfe des Modbus-Adapters im iobroker zu modifizieren und bekomme dort ebenfalls einen Timeout-Fehler beim Beschreiben; Auslesen funktioniert einwandfrei. Trotzdem gelingt es evcc offenbar diesen Wert zumindest *einmal* beim Beginn des Schnellademodus zu modifizieren. 

Wenn ich euch hier helfen kann dieses Problem in meiner Umgebung zu debuggen, lasst es mich gerne wissen. Wir kommen der Sache schon auf den Grund ;-)

andig, 2024-10-07:
Siehe https://github.com/evcc-io/evcc/issues/16326#issuecomment-2377791957. Ich wüsste nicht was wir da debuggen sollten. :(

CPU-6502, 2024-10-07:
... um der Frage nachzugehen, warum evcc zunächst den MinSoC erfolgreich setzen kann, dann aber später beim Zurücksetzten scheitert. Oder ist das euch allen schon klar, und nur ich stehe hier auf dem Schlauch? Auch der HA scheint ja kein Problem damit zu haben...

andig, 2024-10-07:
Und- ist er durch evcc auch gesetzt? Was sagt der Hersteller zu den Fehlern beim setzen?

CPU-6502, 2024-10-08:
Ja, wie gesagt, auch evcc schafft es den Wert zu setzen, aber nur beim Start des Schnellladens - und das genau ist mein Punkt. Wieso ist das so und was genau geht dann beim Zurücksetzen am Ende des Ladevorgangs genau schief? Im Logfile sehe ich dazu nichts, daher mein Vorschlag hier zu debuggen.
Auf den Hersteller zuzugehen macht noch keinen Sinn, da wir ja alle hier ein ModBus-TCP-Gateway zwischengeschaltet haben und der Fehler auch dort liegen könnte.

andig, 2024-10-08:
Was soll evcc da tun? Modbus ist kein Hexenwerk. Der Ball liegt beim Hersteller. ich mache hier zu- es gibt einfach nichts, was wir da tun könnten.

closed 2024-10-08 by andig

hasselmm, 2025-05-20:
> @ andigDa bin ich voll bei Dir! Am 28.09.2024 um 11:30 schrieb andig ***@***.***>:﻿
> Wozu fehleranfällige Cloud APIs wenn man das Problem lokal lösen kann? Ihr seid die Kunden- macht Druck beim [...]

Weil die fehleranfällige Cloud API vollkommen ausreicht, um einfache Anwendungsfälle abzudecken, jedoch offiziell unterstützt und dokumentiert ist. Vor allem aber weil die Cloud API genutzt werden kann, ohne dass man weitere Stromverbraucher installieren und die Verkabelung der PV-Anlage manipulieren muss. Ist halt für eine Endverbraucher, die nicht zugleich Elektroinstallateur sind, eine unschöne, aber gangbare Lösung.

andig, 2025-05-20:
Das teile ich, ändert aber nichts an

> Ansonsten müsste jemand die Cloud API implementieren.