evcc-io/evcc #16361: Netzladen: Nur einmalig ausführen

maatinh, 2024-09-27:
**Is your feature request related to a problem? Please describe.**
Vielen Dank für die Netzladen-Funktion! Das funktioniert bei mir sehr gut. Gestern war wenig Sonne, aber der Strom am Nachmittag günstig. Also habe ich eingestellt, dass evcc in den beiden günstigsten Stunden den Akku nachlädt. Somit war der Akku am Nachmittag voll und konnte am Abend genutzt werden. Gegen Mitternacht war der Akku noch >50% und hätte locker gereicht für den Rest der Nacht bis zum Morgen um wieder durch PV aufgeladen zu werden.
Leider war aber der Plan noch aktiv, so dass evcc um Mitternacht den Akku wieder auf 100% aufgeladen hat. Ich hatte vergessen am Nachmittag vorher, nachdem der Ladevorgang abgeschlossen war, das Preislevel wieder zu ändern.

**Describe the solution you'd like**
Ich fände es gut, wenn es ein Häckchen gäbe, um das Netzladen nur einmalig auszuführen

**Describe alternatives you've considered**
Relative Preislevel wären eventuell noch eine Alternative um das Problem zu vermeiden.

andig, 2024-09-27:
Aktuell nicht vorgesehen. Du kannst das natürlich über die APIs realisieren.

/cc @naltatis

naltatis, 2024-09-27:
> Aktuell nicht vorgesehen.

Hi @maatinh, ja die Netzladenfunktion ist aktuell bewusst noch sehr simpel gehalten. Um die Ideen und Wünsche zu sammeln, hab ich hier mal einen Diskussionsthread aufgemacht. Der ist dann eine gute Grundlage für unsere nächsten Schritte bei dem Thema.

https://github.com/evcc-io/evcc/discussions/16364

Diesen Issue mache ich daher erst einmal zu.

closed 2024-09-27 by naltatis