evcc-io/evcc #16363: feature-request: data-thrift-mode for cloud-requests (mainly cars, specific Hyundai-Bluelink and maybe Kia)

TJ23456, 2024-09-27:
**Is your feature request related to a problem? Please describe.**
Currently evcc requests data from the Cloud (Cars - in my case Hyundai-Bluelink) very often in short time intervals to determine the car-SoC.
Unfortunately Hyundai is currently allowing only a low amount of requests for each account/car. After a "short" time his results in error 509 in evcc and of course the Bluelink-app doesn´t allow login anymore - at least until midnight. With start of the new day the counter of Bluelink-access-requests is resetted to 0 and Bluelink is reachable again.
Even this is not an issue of evcc it should be considered to handle this "behavior".

**Describe the solution you'd like**
I would suggest to introduce an optional data-thrift-mode as a choice compared to the current data-polling-mode:

1. With activated "data-thrift-mode" evcc only request data from the car-cloud (in my case Bluelink) _**one time**_ (when evcc see that a car is just connected to one of the controlled wallboxes) to get current SoC and other data.
2. Then the SoC of the car during the charging-process is only calculated from the energy-amount determined by the Wallbox.
3. Maybe at the end of the calculated-charging-process evcc could request the current SoC one more time from the cloud.

Of course the data would be less precise in this "data-thrift-mode", but it would be much more benefitial than just have no access to the car remotely for the rest of the day.
And of course the battery-size of the car must be known by evcc for calculation.

**Describe alternatives you've considered**
Looking for some kind of automation with Homeassistant to start evcc on the linux only once the wallbox detects a connected car and bring down (stop the evcc-daemon) evcc after the charging-process is finished.
Currently I have to do this anyway because evcc doesn´t try to reach out to the Bluelink-cloud again once a new day started.

Any other ideas for workaround?

andig, 2024-09-27:
> Currently evcc requests data from the Cloud (Cars - in my case Hyundai-Bluelink) very often in short time intervals to determine the car-SoC.

That would be a bug. Needs config/logfile.

TJ23456, 2024-09-27:
> > Currently evcc requests data from the Cloud (Cars - in my case Hyundai-Bluelink) very often in short time intervals to determine the car-SoC.
> 
> That would be a bug. Needs config/logfile.

One example of the logfile is below. I will enhance the log-levels for this evening and upload again as file.
However, what exactly would be the bug here? This behavior of the Bluelink-Cloud is reproduceable even without evcc but with homeassistant and the Bluelink-App. But in homeassistant the Bluelink-integration can be modified regarding "Scan-intervall", "force-refresh-interval" and config of clock-times where "force-refreshes" are prevented.

```
> Sep 25 09:06:28 wj-pv-laptop systemd[1]: Started evcc.
> Sep 25 09:06:28 wj-pv-laptop evcc[614145]: [main  ] INFO 2024/09/25 09:06:28 evcc 0.130.12
> Sep 25 09:06:28 wj-pv-laptop evcc[614145]: [main  ] INFO 2024/09/25 09:06:28 using config file: /etc/evcc.yaml
> Sep 25 09:06:28 wj-pv-laptop evcc[614145]: [db    ] INFO 2024/09/25 09:06:28 using sqlite database: /var/lib/evcc/evcc.db
> Sep 25 09:06:29 wj-pv-laptop evcc[614145]: [main  ] INFO 2024/09/25 09:06:29 listening at :7070
> Sep 25 09:06:31 wj-pv-laptop evcc[614145]: [lp-1  ] WARN 2024/09/25 09:06:31 deprecated: mincurrent setting is ignored, please remove
> Sep 25 09:06:31 wj-pv-laptop evcc[614145]: [lp-1  ] WARN 2024/09/25 09:06:31 deprecated: maxcurrent setting is ignored, please remove
> Sep 25 09:06:32 wj-pv-laptop evcc[614145]: [semp  ] WARN 2024/09/25 09:06:32 SEMP_BASE_URL unspecified, using http://192.168.100.75:7070 instead
> Sep 25 09:06:32 wj-pv-laptop evcc[614145]: [site  ] INFO 2024/09/25 09:06:32 site config:
> Sep 25 09:06:32 wj-pv-laptop evcc[614145]: [site  ] INFO 2024/09/25 09:06:32   meters:      grid ✓ pv ✓ battery ✓
> Sep 25 09:06:32 wj-pv-laptop evcc[614145]: [site  ] INFO 2024/09/25 09:06:32     grid:      power ✓ energy ✓ currents ✓
> Sep 25 09:06:32 wj-pv-laptop evcc[614145]: [site  ] INFO 2024/09/25 09:06:32     pv 1:      power ✓ energy ✓ currents ✗
> Sep 25 09:06:32 wj-pv-laptop evcc[614145]: [site  ] INFO 2024/09/25 09:06:32     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
> Sep 25 09:06:32 wj-pv-laptop evcc[614145]: [site  ] INFO 2024/09/25 09:06:32   vehicles:
> Sep 25 09:06:32 wj-pv-laptop evcc[614145]: [site  ] INFO 2024/09/25 09:06:32     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
> Sep 25 09:06:32 wj-pv-laptop evcc[614145]: [lp-1  ] INFO 2024/09/25 09:06:32 loadpoint 1:
> Sep 25 09:06:32 wj-pv-laptop evcc[614145]: [lp-1  ] INFO 2024/09/25 09:06:32   mode:        pv
> Sep 25 09:06:32 wj-pv-laptop evcc[614145]: [lp-1  ] INFO 2024/09/25 09:06:32   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
> Sep 25 09:06:32 wj-pv-laptop evcc[614145]: [lp-1  ] INFO 2024/09/25 09:06:32   meters:      charge ✓
> Sep 25 09:06:32 wj-pv-laptop evcc[614145]: [lp-1  ] INFO 2024/09/25 09:06:32     charge:    power ✓ energy ✓ currents ✓
> Sep 25 09:06:32 wj-pv-laptop evcc[614145]: [site  ] WARN 2024/09/25 09:06:32 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
> Sep 25 09:06:32 wj-pv-laptop evcc[614145]: [lp-1  ] INFO 2024/09/25 09:06:32 car disconnected
> Sep 25 09:06:49 wj-pv-laptop evcc[614145]: [lp-1  ] INFO 2024/09/25 09:06:49 vehicle updated: unknown -> Kona
> Sep 25 09:07:29 wj-pv-laptop evcc[614145]: [lp-1  ] INFO 2024/09/25 09:07:29 vehicle updated: Kona -> unknown
> Sep 25 09:07:36 wj-pv-laptop evcc[614145]: [lp-1  ] INFO 2024/09/25 09:07:36 vehicle updated: unknown -> Kona
> Sep 25 16:04:42 wj-pv-laptop evcc[614145]: [lp-1  ] INFO 2024/09/25 16:04:42 car connected
> Sep 25 16:05:02 wj-pv-laptop evcc[614145]: [lp-1  ] INFO 2024/09/25 16:05:02 start charging ->
> Sep 25 16:15:42 wj-pv-laptop evcc[614145]: [lp-1  ] ERROR 2024/09/25 16:15:42 vehicle soc limit: unexpected status: 509 ()
> Sep 25 16:15:42 wj-pv-laptop evcc[614145]: [lp-1  ] ERROR 2024/09/25 16:15:42 vehicle range: must retry
> Sep 25 16:17:52 wj-pv-laptop evcc[614145]: [lp-1  ] WARN 2024/09/25 16:17:52 vehicle soc: unexpected status: 509 () (ignored by estimator)
> Sep 25 16:17:52 wj-pv-laptop evcc[614145]: [lp-1  ] ERROR 2024/09/25 16:17:52 vehicle soc limit: unexpected status: 509 ()
> Sep 25 16:17:52 wj-pv-laptop evcc[614145]: [lp-1  ] ERROR 2024/09/25 16:17:52 vehicle range: must retry
> Sep 25 16:19:53 wj-pv-laptop evcc[614145]: [lp-1  ] WARN 2024/09/25 16:19:53 vehicle soc: unexpected status: 509 () (ignored by estimator)
> Sep 25 16:19:53 wj-pv-laptop evcc[614145]: [lp-1  ] ERROR 2024/09/25 16:19:53 vehicle soc limit: unexpected status: 509 ()
> Sep 25 16:19:53 wj-pv-laptop evcc[614145]: [lp-1  ] ERROR 2024/09/25 16:19:53 vehicle range: must retry
> Sep 25 16:22:02 wj-pv-laptop evcc[614145]: [lp-1  ] WARN 2024/09/25 16:22:02 vehicle soc: unexpected status: 509 () (ignored by estimator)
> Sep 25 16:22:02 wj-pv-laptop evcc[614145]: [lp-1  ] ERROR 2024/09/25 16:22:02 vehicle soc limit: unexpected status: 509 ()
> Sep 25 16:22:02 wj-pv-laptop evcc[614145]: [lp-1  ] ERROR 2024/09/25 16:22:02 vehicle range: must retry
> Sep 25 16:24:03 wj-pv-laptop evcc[614145]: [lp-1  ] WARN 2024/09/25 16:24:03 vehicle soc: unexpected status: 509 () (ignored by estimator)
> Sep 25 16:24:04 wj-pv-laptop evcc[614145]: [lp-1  ] ERROR 2024/09/25 16:24:04 vehicle soc limit: unexpected status: 509 ()
> Sep 25 16:24:04 wj-pv-laptop evcc[614145]: [lp-1  ] ERROR 2024/09/25 16:24:04 vehicle range: must retry
[... cut ...]
> Sep 25 20:47:32 wj-pv-laptop evcc[614145]: [lp-1  ] WARN 2024/09/25 20:47:32 vehicle soc: unexpected status: 509 () (ignored by estimator)
> Sep 25 20:47:32 wj-pv-laptop evcc[614145]: [lp-1  ] ERROR 2024/09/25 20:47:32 vehicle soc limit: unexpected status: 509 ()
> Sep 25 20:47:32 wj-pv-laptop evcc[614145]: [lp-1  ] ERROR 2024/09/25 20:47:32 vehicle range: must retry
> Sep 25 20:49:06 wj-pv-laptop evcc[614145]: [main  ] INFO 2024/09/25 20:49:06 evcc was stopped by user. OS should restart the service. Or restart manually.
> Sep 25 20:49:07 wj-pv-laptop systemd[1]: evcc.service: Deactivated successfully.
```

andig, 2024-09-27:
We need trace log for http, other components debug. If it errors right from start the problem is before. Make sure that's not the case.

TJ23456, 2024-09-27:
> I am using evcc with bluelink (Ioniq5) since nearly 3 years (but without HA or ioBroker or any other tool) and I didn´t have this error at any time.

Womöglich liegt das Problem bei mir auch an einer Kombination aus verschiedenen Ursachen - muss ich noch testen:
- Am Ort der Ladepunkte besteht eine schlechte Netzabdeckung - womöglich führt das zu häufigen Cloud-Requests
- oder die Hyundai hat die Bluelink-Cloud für den Kona anders konfiguriert als für den Ioniq5
- oder evcc und Homeassistant produzieren zusammen zuviele Cloud-Requests (jedenfalls kam der Homeassistant bislang allein halbwegs klar).
Eines ist jedoch Fakt: die Bluelink-Cloud sperrt aber einer (mir unbekannten) Anzahl Cloud-Requests den Zugang bis zum Ende des Tages. Daher war mein erster Gedanke die Implementierung des "Daten-Sparmodus", weil mit den vorliegenden Daten (Start-Fahrzeug-SoC, Batteriegröße (aus der Config) und Lademenge (aus der Wallbox) der SoC während des Ladevorgangs halbwegs brauchbar berechnet werden könnte, ohne die Daten aus der Cloud abzufragen (zumindest bis zum (nahen) Ende des Ladevorgangs)...

andig, 2024-09-27:
> oder evcc und Homeassistant produzieren zusammen

Wenn Du das Api mit HA zuballerst leider natürlich evcc. Wir fragen nur max. all 15min an.

closed 2024-09-27 by andig

TJ23456, 2024-09-27:
> > oder evcc und Homeassistant produzieren zusammen
> 
> Wenn Du das Api mit HA zuballerst leider natürlich evcc. Wir fragen nur max. all 15min an.

HA fragt bei mir eigentlich nur alle 60 min ab. Aber ich schalte es für den Test ab. Logfiles kommen dann morgen.

andig, 2024-09-27:
I still like the idea of the thrift mode for those cases where the charger provides an initial soc only (simulated DC) and vehicle API is down or not configured:
[image]

TJ23456, 2024-09-27:
> I still like the idea of the thrift mode for those cases where the charger provides an initial soc only (simulated DC) and vehicle API is down or not configured: <img alt="Screenshot 2024-09-27 at 11 30 50" width="215" src="https://private-user-images.githubusercontent.com/184815/371476432-beffc2fd-d02f-4e1e-b728-79b19ac1e4bd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc0Mjk5NTEsIm5iZiI6MTcyNzQyOTY1MSwicGF0aCI6Ii8xODQ4MTUvMzcxNDc2NDMyLWJlZmZjMmZkLWQwMmYtNGUxZS1iNzI4LTc5YjE5YWMxZTRiZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyN1QwOTM0MTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kZjZiYjdkYTg1ZjhhMmI2OGZiYWQ2Y2ViZGE5MTZkZjljMDczMDU0OWRmYTgxNDBkY2Q4YTEyOTE5Y2Y2NzU2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.acjweD1plTlAILeAHyS8LDpQ_cJ6tK1EYPhq8IHvf9U">

... and maybe where a mobile-data-connection for Internet with limited amount of data is in place. Ok, I know that the API-requests doesn´t consume that much data - but maybe it helps

TJ23456, 2024-09-27:
> We need trace log for http, other components debug. If it errors right from start the problem is before. Make sure that's not the case.

So, now it was faster than expected until the error-message was dropped - log file attached.
(please ignore the message: "charge power: Get "http://192.168.100.74/api/status?filter=alw,car,eto,nrg,wh,trx,cards": context deadline exceeded (Client.Timeout exceeded while awaiting headers)" - the Go-e charger has only a weak WiFi-connection for my tests)
[evcc.log](https://github.com/user-attachments/files/17166380/evcc.log)

TJ23456, 2024-09-27:
> > Interpretiere ich die Fehler-Logs richtig, dass evcc alle 2,5 Minuten bei der Cloud versucht anzufragen?
> 
> Das ist nicht die Regel, sondern nur situationsbedingt, weil die Anfragen abgelehnt werden. Dann gibt es immer wieder einen retry.
> 
> Das ist auch nicht normal. (Oder hast du da händisch eingegriffen?)
> 
> ```
> > Sep 25 09:06:49 wj-pv-laptop evcc[614145]: [lp-1  ] INFO 2024/09/25 09:06:49 vehicle updated: unknown -> Kona
> > Sep 25 09:07:29 wj-pv-laptop evcc[614145]: [lp-1  ] INFO 2024/09/25 09:07:29 vehicle updated: Kona -> unknown
> > Sep 25 09:07:36 wj-pv-laptop evcc[614145]: [lp-1  ] INFO 2024/09/25 09:07:36 vehicle updated: unknown -> Kona
> ```

Das sind alte Log-Meldungen - da habe ich händisch eingegriffen, weil der Kona irgendwann am nächsten Morgen nicht mehr von evcc abgefragt wird - vielleicht gibt evcc nach einer gewissen Menge an "retry"´s auf - jedenfalls muss ich das Fahrzeug jeden morgen erneut speichern, damit die Cloud-Abfrage wieder funktioniert.

Ich versuche halt gerade, evcc für mich zum Laufen zu bringen und teste dazu einiges. An sich läuft alles, aber dass die Bluelink-Cloud irgendwann die API-calls ablehnt und dieser Thread geschlossen wurde (Homeassistant hatte ich natürlich deaktiviert - hätte ich mal schreiben sollen, vielleicht hätte andig dann nicht den Status auf "Closed" gesetzt, weil der Feature-Request vielleicht doch sinnvoll sein könnte), ist für mich schwierig, weil ich das Fahrzeug jeden Morgen neu anlegen/speichern muss...

VolkerK62, 2024-09-27:
Versteh ich nicht. Läuft bei mir ohne Probleme.
Hast du in der yaml beim Auto oder Loadpoint irgendetwas besonderes konfiguriert? (soc polling oder cache)

Bin mir auch nicht sicher, ob `http: trace` richtig ist.
Ich denke eher `hyundai: trace`

TJ23456, 2024-09-27:
> hyundai: trace

Erstmal: vielen Dank für deine Unterstützung!
Deinen Vorschlag mit dem Trace probiere mal aus.

Das Auto ist über die GUI konfiguriert - die einzige Besonderheit ist vielleicht, dass ich den Cache auf "60" gesetzt hatte (in der Hoffnung, dass damit vielleicht das Problem behoben würde.)

Der Loadpoint sieht so aus:
```
loadpoints:
  - title: Go-e-Charger # display name for UI
    charger: Werner_GoE # charger
    mode: "pv" # set default charge mode, use "off" to disable by default if charger is publicly available, charge mode (off, now, minpv, pv)
    phases: 0 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    #minCurrent: 6 # minimum charge current (default 6A)
    #maxCurrent: 16 # maximum charge current (default 16A)
    enable:
      # einschalten, wenn 1 Minute lang mindestens 3500 W Überschuss vorhanden ist
      delay: 1m
      threshold: -3500
    disable:
      # ausschalten, wenn 5 Minuten lang mehr als 2000 W aus dem Netz bezogen werden
      delay: 5m
      threshold: 2000
```

andig, 2024-09-27:
60 wären dann Nanosekunden. Warum nicht einfach mal den Standard lassen am Anfang?!

TJ23456, 2024-09-27:
> 60 wären dann Nanosekunden. Warum nicht einfach mal den Standard lassen am Anfang?!

Das hatte ich erst nach einigen Versuchen gemacht. Die Eingabe war einer meiner Folgeversuche, die Cloud-API-Calls hinauszuzögern (also die Intervalle zu verlängern). Aber 60ns machen die Sache wohl nicht besser - habe ich wieder etwas gelernt.

Eigentlich war mein Ziel, den Cache auf 60 minuten zu erhöhen (allerdings nimmt die GUI die Eingabe "60m" nicht - daher die 60.

Was ist denn mit meinem Feature-Request - da dieser Thread "closed" ist, würde ich mal ein "wird nicht umgesetzt" vermuten? Ist auch ok - Ihr macht das ja freiwillig.

andig, 2024-09-27:
/cc @naltatis wie ist das mit Durations im UI?

naltatis, 2024-09-27:
Ja, Duration-Felder in Config UI scheinen noch eine Baustelle zu sein. Die Eingabe erfolgt immer in Sekunden (also nicht als Go Duration Strings), aber wir kommunizieren das nicht und zeigen zudem noch ein falsches Beispiel das für die evcc.yaml gedacht ist (5m). Ist ein separates issue.

[image]