evcc-io/evcc #16368: Wallbox Dadapower Fehler Phasenwechsel PV Mode

TheShunxxx, 2024-09-27:
### Describe the bug

Hello there,

I use the EVCC addon in Home Assistant with a Dadapower wallbox. I've been getting the following error message for a few days now.

Garage: charger logic error: current
mismatch (got 16A measured,
expected 6A)

Nothing has actually changed, except the wallbox manufacturer updated the wallbox a few weeks ago, after which it worked again, although I needed a sponsor token shortly afterwards, which wasn't the case before. It does not recognize the excess correctly and always charges with 11kW when there is an excess, even if there is only a 2kW excess. Somehow he can't handle the phase change.
Does anyone have an idea why this is or how I can fix it?

### Steps to reproduce

1. Start EVCC starten
2. Choose PV Mode and wait for PV surplus
3. Then the error occurs

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.xxx.xx:8123
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: xxx

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
[... 358 lines cut ...]
[lp-1  ] DEBUG 2024/09/27 12:17:23 charge total import: 3176.631kWh
[lp-1  ] DEBUG 2024/09/27 12:17:23 charger status: B
[lp-1  ] DEBUG 2024/09/27 12:17:23 pv charge current: 4.79A = 0A + 4.79A (-1101W @ 1p)
[site  ] DEBUG 2024/09/27 12:17:33 ----
[lp-1  ] DEBUG 2024/09/27 12:17:33 charge power: 0W
[lp-1  ] DEBUG 2024/09/27 12:17:33 charge currents: [0 0 0]A
[site  ] DEBUG 2024/09/27 12:17:33 pv power: 1472W
[site  ] DEBUG 2024/09/27 12:17:33 grid meter: -1072W
[site  ] DEBUG 2024/09/27 12:17:33 site power: -1072W
[lp-1  ] DEBUG 2024/09/27 12:17:33 charge total import: 3176.631kWh
[lp-1  ] DEBUG 2024/09/27 12:17:33 charger status: B
[lp-1  ] DEBUG 2024/09/27 12:17:33 pv charge current: 4.66A = 0A + 4.66A (-1072W @ 1p)
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.130.12

andig, 2024-09-28:
> Nothing has actually changed, except the wallbox manufacturer updated the wallbox a few weeks ago, after which it worked again, 

So it's not an evcc issue if that wasn't updated!

> although I needed a sponsor token shortly afterwards, which wasn't the case before.

Vendor has stopped sponsoring us, sorry for the inconvenience.

> It does not recognize the excess correctly and always charges with 11kW when there is an excess, even if there is only a 2kW excess. Somehow he can't handle the phase change.

You can test that standalone (without evcc running) using

    evcc meter

An-Wi, 2024-09-28:
> Nothing has actually changed, except the wallbox manufacturer updated the wallbox a few weeks ago, after which it worked again

Which firmware version is now working at your dadapower?
Do you have a Single- or Dual-Wallbox?

TheShunxxx, 2024-09-29:
It is the single wallbox with software version 1.2.22.
Is it maybe necessary to adapt the configuration for example the power limits?

DerArtem, 2024-09-30:
Hello,

there is a bug in version 1.2.22. Please install version [1.2.23](https://ftp.dadacon.de/downloads/firmware/1.2.23/buildroot.swu).

andig, 2024-09-30:
Great support, thank you 👍🏻

closed 2024-09-30 by andig

TheShunxxx, 2024-09-30:
Thank you I will check it and if the error happen again I will repost it.