evcc-io/evcc #16390: UX LP: Harmonize  unit and resolution of planned energy with the other numbers

RenatusRo, 2024-09-28:
**Is your feature request related to a problem? Please describe.**
Derzeit erscheinen da stets volle kWh. 
Werden via API gebrochene kWh angefragt, wird auf volle kWh gerundet.
Das aergert den Zahlenfetischisten.

Use cases (bei externer Steuerung der Ladeplaene):

- Fuer den Boiler werden am Ende des Sonnentages die noetigen kWh berechnet, die bis zum Erreichen der Zieltemperatur noch fehlen (2.5 ...  3.5 kWh) und als Plan gepusht.
- Der e-bike Akku muss ueber den Tag 0.65 kWh erhalten - ist aber der letzte in der Nahrungskette und bekommt dann nach 16:00 noch die noetige Restmenge per Ladeplan.

**Describe the solution you'd like**
Die gleiche Aufloesungs-/Einheitenlogik verwenden wie Power/Charge/Charged auf dem LP.

**Additional context**
[image]

andig, 2024-09-28:
> Derzeit erscheinen da stets volle kWh.

Das ist erstmal ne mutige Behauptung. Bei mir nicht.

RenatusRo, 2024-09-28:
sende :
`{"time": "%2024-%09-%28T%22:00:00Z", "value" : "0.65" }`
log sagt: 
`[lp-4  ] DEBUG 2024/09/28 12:02:44 set plan energy: 0.65kWh @ 2024-09-29 00:00:00 +0200 CEST`
GUI sagt: 1kWh
CHARGED ist in Wh, PLAN in kWh gerundet

[image]

naltatis, 2024-09-28:
Hast du für das Fahrzeug eine Capacity hinterlegt? Wir verwenden an dieser Stelle die gleiche Formatierung (Einheit und Genauigkeit) wie im Dropdown in dem du den Wert auswählst. Zumindest soll das so sein.

RenatusRo, 2024-09-28:
Ah, nee.
Dann haut es hin - case closed. 

Thx & Sorry!

(Randbemerkung: 
Fuer den Boiler klappt das nicht: Wenn man dort Capacity hinterlegt, steht der Plan in %. 
Aber offiziell is der Plan fuer heating ja sowieso unsichtbar. Dann hacke ich das Komma selber da rein ...)

[image]

closed 2024-09-28 by RenatusRo

naltatis, 2024-09-28:
> Fuer den Boiler klappt das nicht: Wenn man dort Capacity hinterlegt, steht der Plan in %.

Ja, Plan für Heating ist ein anderes Thema: https://github.com/evcc-io/evcc/issues/10220