evcc-io/evcc #16401: Unable to use Elli Charger

faxioman, 2024-09-28:
### Describe the bug

Hello. I'm trying to use an Elli charger without success.
I managed to pair it with EVCC, and there are no errors in the EVCC logs.
In fact, I can see that communication is happening:

```
[eebus ] DEBUG 2024/09/28 20:07:36 Send: d:_i:47859_Elli-Wallbox-2045A11JLS notify 185 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/09/28 20:07:38 Send: d:_i:47859_Elli-Wallbox-2045A11JLS notify 186 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/09/28 20:07:40 Send: d:_i:47859_Elli-Wallbox-2045A11JLS notify 187 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/09/28 20:07:42 Send: d:_i:47859_Elli-Wallbox-2045A11JLS notify 188 DeviceDiagnosisHeartbeatData
```

When I connect the car, charging starts always at 6A, but all the commands I send via EVCC seem to have no effect.
For example, if I try to stop the charging from EVCC while it’s charging, I see:

```
[lp-1  ] DEBUG 2024/09/28 20:10:13 charger disable
[lp-1  ] DEBUG 2024/09/28 20:10:13 wake-up timer: stop
[eebus ] DEBUG 2024/09/28 20:10:13 Recv: d:_i:47859_Elli-Wallbox-2045A11JLS:LoadControl to LoadControl notify 355 LoadControlLimitListData
[eebus ] DEBUG 2024/09/28 20:10:13 ski d882aa00b3dacb31acd0f4be199ebdca71506496 event cem-opev-DataUpdateLimit
[eebus ] DEBUG 2024/09/28 20:10:13 Recv: d:_i:47859_Elli-Wallbox-2045A11JLS:LoadControl to LoadControl result 356 266 ResultData 0
```

But nothing happens at the wallbox, and it keeps charging.
I have configured an external meter.
I’ve tried restarting multiple times, with the car disconnected, with the car connected… no success.
I’m probably making a simple mistake… but I just can’t figure out what it is.
I’m aware that Elli has several bugs regarding EEBUS… but it seems like nothing is working for me with this charger.

Thanks!!

### Steps to reproduce

I'm simply trying to use an Elli charger connect.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: myinstallationid

interval: 30s # control cycle interval

eebus:
  certificate:
    public: |
      -----BEGIN CERTIFICATE-----
      MIIBvDC...ErZkCw==
      -----END CERTIFICATE-----

    private: |
      -----BEGIN EC PRIVATE KEY-----
      MHcCAQE...+SUjw==
      -----END EC PRIVATE KEY-----

chargers:
- name: elli
  type: template
  template: elliconnect
  ski: D882AA00B3DAC.....4BE199EBDCA71506496
  ip: 192.168.118.45

vehicles:
  - name: my_car
    type: template
    template: offline

meters:
- name: wallbox
  type: template
  template: shelly-1pm
  usage: charge
  channel: 0
  host: 192.168.4.40
- name: contatore
  type: template
  template: shelly-1pm
  usage: grid
  channel: 1
  host: 192.168.4.40

loadpoints:
- title: Garage
  charger: elli
  meter: wallbox
  mode: now
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  vehicle: my_car

site:
  title: Casa
  meters:
    grid: contatore
```

### Log details

```text
Log inline with ticket
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.12

andig, 2024-09-28:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-09-28 by andig

faxioman, 2024-09-28:
Purtroppo non parlo il tedesco 😅
Grazie comunque.