evcc-io/evcc #16417: Preis bzgl. Effizienz des Batteriespeichersystems

Tombra1889, 2024-09-29:
**Is your feature request related to a problem? Please describe.**
Bei einem Batteriespeicher gibt es einen gewissen Speicherverlust, aktuell wird der Preis auf Basis der Einspeisevergütung angegeben.

**Describe the solution you'd like**
Es soll ein separates Feld zur Erfassung der Effizienz des Batteriespeichersystems in die Systemkonfiguration aufgenommen werden. Dieses Feld ermöglicht die Angabe der tatsächlichen Effizienz des Speichers (in %), um den Verbrauchspreis unter Berücksichtigung der Speicherverluste präzise zu berechnen. Bei einem SMA Hybridwechselrichter in Kombination mit einem BYD HVS Speicher liegt beispielsweise die Effizienz bei ca. 64 %. 
**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.
Angabe eines Batteriespeicher-Verbrauchspreises

andig, 2024-09-30:
Zu welchem Zweck soll das dienen?

Tombra1889, 2024-09-30:
Im Endeffekt für eine korrektere Darstellung des jeweiligen Preises für die Ladesession. Dort wird aktuell die Photovoltaik-Einspeisevergütung herangezogen. Durch die Ladeverluste des Batteriespeichers, müsste dieser aber mit einem Korrekturfaktor in die Statistik einfließen.

nilseckert, 2024-09-30:
Aber die Effizient ist doch auch in gewissem Maße abhängig von der Last, oder?

Wenn der WR nachts nur 300W liefert ist der Eigenverbrauch von dem Wechselrichter (sagen wir mal 100W) so groß, dass neben Verlusten für die Elektronik im eigentlichen Vorgang noch 100W Grundrauschen anfallen.

Der Wirkungsgrad im Batteriebetrieb wird umgekehrt umso besser, je mehr Leistung man aus der Batterie zieht.

Was ich damit sagen will, der Wirkungsgrad ist vermutlich keine einfache Prozentzahl.

andig, 2024-09-30:
> Im Endeffekt für eine korrektere Darstellung des jeweiligen Preises für die Ladesession

Das ist aktuell kein Fokus. Letztlich spielts auch keine Rolle, ob die Effizienz des Speichers beim Hausstromverbraucht oder beim EV laden zuschlägt.

closed 2024-09-30 by andig

naltatis, 2024-09-30:
Ja, aktuell ist das eine bewusste Vereinfachung und wir berücksichtigen Wandlungsverluste in der Batterie nicht. Siehe auch Erklärung unter "Was ist Sonnenstrom" hier https://docs.evcc.io/docs/faq#ersparnisberechnung

Mit dem Thema Netzladen des Hausakkus ändert sich an der Annahme ein bisschen was und die Preise mit Dynamischen Stromtarifen stimmen dann auch nicht mehr. Das werden wir früher oder später auch abbilden wollen. In https://github.com/evcc-io/evcc/discussions/16364 gibts auch Ideen in die Richtung, dass wir den "Wert des Speicherinhaltes" abbilden wollen/müssen um streng preisoptimiert laden zu können. In dem Zuge kann ich mir vorstellen, dass wir auch Verluste irgendwie pauschal mit berücksichtigen. Aber bislang alles Diskussion.