evcc-io/evcc #16432: SofarSolar lädt nicht mehr

PeterPablo, 2024-09-30:
### Describe the bug

Seit heute (?!) lädt der am SofarSolar HYD 5-20KTL-3PH angeschlossene Batteriespeicher nicht mehr. Das hängt möglicherweise mit dem kürzlichen Release von v0.130.12 zusammen. In den Release Notes wird auf https://github.com/evcc-io/evcc/pull/16113 verwiesen, was ein starkes Indiz für eine Regression ist.
Evcc läuft als Erweiterung von HomeAssistant auf einem Raspberry Pi. Das Setup lief unverändert monatelang problemlos.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
in Teilen abgetippt (bin eigentlich nur Nutzer des Setups):

- name: grid-sofar-v2
  type: template
  template: sofarsolar-g3
  usage: grid

  # Modbus TCP
  modbus: tcpip
  id: 1
  host: 192.168.1.100
  port: 8899

- name: battery-sofar-v2
  type: template
  template: sofarsolar-g3
  usage: battery

  # Modbus TCP
  modbus: tcpip
  id: 1
  host: 192.168.1.100
  port: 8899
```
```

### Log details

```text
Ich kann mit mehr Aufwand ein log bereitstellen. Wegen wireguard und co würde ich das nur im tatsächlichen Bedarfsfall nachliefern. Ich habe keinen direkten Zugriff auf den Raspberry Pi.
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.130.12

PeterPablo, 2024-09-30:
Auf https://docs.evcc.io/docs/devices/meters#hyd-520k-3ph findet sich (noch) kein Hinweis, wie die neue `capability` `battery-control` zu konfigurieren ist. Ich kann das gerne testen.

closed 2024-09-30 by andig

PeterPablo, 2024-09-30:
bene. 
`pv 1 power: read failed: modbus: response data size '68' does not match request quantity '1'` ist alles was ich aus der GUI extrahieren kann. An das Log komme ich mangels Kenntnis des Passworts über die Oberfläche nicht.
Downgrade wird versucht. Hiermit gibts einen Anlaufpunkt für weitere Betroffene, die gerne ein echtes Issue anlegen, das von vornherein mit Log versehen ist.

andig, 2024-09-30:
Mir wäre nicht bekannt, dass wir an PV etwas geändert hätten.

PeterPablo, 2024-09-30:
Downgrade auf v0.130.11 gemäß https://github.com/evcc-io/evcc/discussions/5209 ist erfolgt.
Auch damit wird aktuell der leere Akku nicht mehr geladen, jedoch noch mit 250W ins Netz eingespeist. Ich habe die Diskussionen in den diversen Issues lesend verfolgt und es könnte wohl sein, dass die Register im Akku das erneute Laden trotz Downgrade nicht mehr zulassen.
Das ist als reine Spekulatin zu werten...

Ich habe nun das PW, um an die Logs zu kommen. Reiche ich nach, vermutilch sobald ich wieder auf v0.130.12 bin.
Bei v0.130.11 erscheint rechts oben keine "Fehlermeldung" (also kein `pv 1 power`...). Log wirkt auf Handy sauber -- ich muss das mitm Laptop machen. Update folgt..

andig, 2024-09-30:
/cc @cschlipf

cschlipf, 2024-09-30:
@PeterPablo 

Checke mal ob Dein Wechselrichter noch im Passive Mode ist oder ob er auf Self-Use zurückgesetzt wurde.

PeterPablo, 2024-09-30:
Muss ich das am Display des Wechselrichters machen oder irgendwie direkt per Modbus? Falls letzteres muss ich erstmal per ssh auf den pi und die passende Magie anwenden... Danke für einen kleinen Tipp.

cschlipf, 2024-09-30:
Gerade meinen Kommentar oben angepasst: Display des Wechselrichters.

Ich vermute, dass aus irgendeinem Grund nicht mehr korrekt auf Self-Use zurückgeschaltet wurde nachdem die Batterie wieder aktiviert worden ist.

PeterPablo, 2024-09-30:
In der Tat, im deutsprachigen menü heßt es unter "Wechselrichter-Infos(4)" bei "Energiespeicher-Modus" "Passiver-Modus" (zeichengenau, orthographisch nicht ganz richtig 😄).
Wie kann der wieder zurück auf "self use" gebracht werden?

PeterPablo, 2024-09-30:
oha (!)
Mein Vermieter und Betreiber des Systems hat sich offenbar "gestern" an https://www.photovoltaikforum.com/thread/221806-sofar-hyd-mit-evcc-akkuentladung-verhindern/ versucht. Ich versuche zu klären, ob das jetzt eine zeitlich ungünstige Überschneidung ist. Ich weiß noch nichts genaues, entschuldige mich aber bereits für den verursachten Aufwand.

Für mich ist noch nicht klar welche der beiden Einflüsse hier nun ausschlaggebend sind.

😞

cschlipf, 2024-09-30:
Einfach unter 1/4 den ersten Punkt aktivieren. Das ist 'Self-Use'.

Jetzt wäre die Frage warum EVCC nicht korrekt zurück auf Self-Use geschaltet hat, wenn der Speicher wieder aktiviert werden soll. Ich habe leider diese neuen Features bei mir noch nicht aktiv. Dazu muss ich mein gesamtes YAML umstellen wozu mir derzeit leider die Zeit fehlt.

cschlipf, 2024-09-30:
> oha (!) Mein Vermieter und Betreiber des Systems hat sich offenbar "gestern" an https://www.photovoltaikforum.com/thread/221806-sofar-hyd-mit-evcc-akkuentladung-verhindern/ versucht. Ich versuche zu [...]

Ja, das haben wir mal gestartet als EVCC das noch nicht unterstützt hat. Sollte nun obsolete sein, wenn EVCC das nativ kann. Außerdem passt einiges aus dem Thread mittlerweile nicht mehr zur aktuellen Version der Integration. Deswegen haben wir ja das GitHub angefangen - aber auch hier müssen wir schon wieder nachziehen.

Lange Rede, kurzer Sinn: Wenn es nur um das Abschalten des Speichers geht, empfehle ich eher auf die neuen Funktionen innerhalb von EVCC zu setzen.

PeterPablo, 2024-09-30:
Danke @cschlipf , ich habe auf "Selbstnutzungs-Modus" zurück gestellt. evcc läuft wieder mit dem jüngsten v0.130.12 mit obigen Einstellungen für das Template. Bezüglich Batteriesteuerung haben wir in der Konfiguration noch nichts eingestellt, weil ich einerseits hier noch genügend positive Rückmeldungen gelesen habe und auch nicht genau weiß, wie ich die "capability" scharf schalte.

Habe Deine Nachricht gelesen und halte den Weg über evcc ebenfalls erstrebenswert. Ich kann auch gerne Dinge testen, falls es hilft.

PeterPablo, 2024-09-30:
Heute ist es nicht mehr hell genug. Ich beobachte das Verhalten, tendiere allerdings zum ersten schnellen Feedback von @andig  -- kein Fehler in evcc.

PeterPablo, 2024-09-30:
Gemäß https://github.com/evcc-io/docs/blob/d28662643b553fa41456885560d4aaa95094116f/i18n/en/docusaurus-plugin-content-blog/2024-02-01/v0.124-new-tesla-api.mdx#active-battery-control müsste in den Einstellungen der Hausbatterie für unterstütze Akkus ein Schieberegler erscheinen. Ich meine, dass genau diese Funktionalität über besagte `battery-control`-Fähigkeit des sofarsolar-g3-Template gesteuert wird?!
Bei unserem Setup wird  dieses Toggle nicht angezeigt. Soll ich hierfür ein neues Issue anlegen oder ist das in einer Diskussion besser aufgehoben?

PeterPablo, 2024-09-30:
Kann es sein, dass es an der fehlenden Angabe von `capacity` liegt?

sahomm, 2024-09-30:
Ich habe ähnliche wenn nicht gar gleicher Beobachtungen mit der 130.12 gemacht. Wenn "Verhindere Entladen im Schnell-Modus und bei geplantem Laden" aktiv ist funktioniert die Umschaltung des HYD auf PassiveMode aber zurück auf SelfUse leider nicht.
Genauer habe ich das hier beschrieben [klick](https://github.com/evcc-io/evcc/issues/12220#issuecomment-2374993010)

Auch wurde dort ja auch das vermutete Problem geschildert.

Grüße

Saho

puzich, 2024-10-06:
Schließe mich dem Thema an. Habe das gleiche Problem. 
konnte es erst einmal wie oben beschrieben wieder herstellen