evcc-io/evcc #16435: "feinere" current Auswahl in Loadpoint Settings

VolkerK62, 2024-09-30:
Aus Slack Diskussion:
Gewünscht wird eine feinere Current Auswahl in den Loadpoint Settings, damit (vor allem bei Switchsockets) die Einschaltschwelle etwas genauer festgelegt werden kann.
Wenn man es etwas genauer handhaben möchte, sind 230W Schritte zu grob.
Vorschlag wäre eine Unterteilung in 0,5A Schritte im Bereich 1A bis 10A (ich denke das sollte als Obergrenze genügen).

andig, 2024-09-30:
0…6A falls es auf Steckdosen abzielen soll?

VolkerK62, 2024-09-30:
auch ok, wenn es das einfacher macht.
Bei den 10A hab ich an so manchen Heizstab gedacht.

astrakid, 2024-12-20:
würde mich auch über feinere abstufungen freuen, da mein auto eine automatische umschaltung von 2 auf 3 phasen macht ab 6,8A, daher würde ich gerne 6,7a über die GUI setzen wollen.

andig, 2025-03-26:
@naltatis gibts hier noch was zu tun?

naltatis, 2026-09-17:
Ist inzwischen erledigt. In Config UI sind Min/Max-Current numerische Eingabefelder. Die Dropdowns in den Loadpoint-Settings sind weiterhin feste Stufen. Können wir ggf. überarbeiten (konkreter an benötigte Range anpassen) wenn https://github.com/evcc-io/evcc/issues/13355 umgesetzt ist.

closed 2026-09-17 by naltatis