evcc-io/evcc #16439: Adding Fox ESS OCPP Charger as tested device

lucode, 2024-10-01:
I'm not quite sure if I placed this correctly as an Feature Request?  A separate Issue Template for new Chargers, would be nice, to get a more structured input.

### FOX ESS OCPP Charger tested
I installed a Fox ESS OCPP Charger last week and with the newest firmware on it, it runs without issues. So an entry in the Template section would be useful.

**Context**
- Tested Firmware: V 1.15,  former version doesn't support OCPP 1.6 
-  Tested evccc version:  0.130.12 
- Tested Model: 7,3 kW Fox ESS AC EV Charger A7300S1-E-2
- All Fox Chargers A Series with OCPP should  work, except model  A7300S1-E-1 A7300p1-E-1 (have no OCPP)
- all support Plug&Play/RFID card/APP start

**Supported models**

_Supported models (ignoring the P, because the S models are technically the same just with a Socket)_ 

- A7300P1-E-2
- A011KP1-E-2
- A011KP1-E-A
- A022KP1-E-A
- A7300P1-E1-E-A
- A022KP1-E-2

**My OCPP config section**
```
chargers:
  - name: Carport
    type: template
    template: ocpp
#    stationid: 60AE73B039PQ007  # Station ID of the charging point. Optional
#    connector: 1 # Connector number for charging stations with multiple connectors. Counting starts at 1 for the first connector. (optional)
    remotestart: true # false/true Only enable this option if there is no way to initiate transactions from the charger side! This is only the case if e.g. no RFID reader is available and charging processes would have to be released individually via app. Normally, the charger should always be configured at the device so that either an RFID card is used for activation or the charger is set to "Autostart", "Free Charging" or similar. First check the documentation and configuration possibilities of the charger, ask the manufacturer if necessary! (optional)
    connecttimeout: 5m # optional
    meterinterval: 10s # optional
#    metervalues: Power.Active.Import,SoC,Current.Offered,Power.Offered,Current.Import,Voltage # optional
```

**Additional context**
 Screenshot Elite / Fox app OCPP config

[image]

premultiply, 2024-10-02:
Ok, should be added to https://github.com/evcc-io/evcc/pull/15293

andig, 2024-10-06:
@lucode autostart ist nicht möglich?

lucode, 2024-10-06:
> @lucode autostart ist nicht möglich?

Nein es gibt nur entweder den Mode Controlled oder PlugAndPlay. D.h. im Mode Controlled  wartet der Charger auf ein Signal, RFID, App oder über OCPP.

closed 2024-10-18 by github-actions[bot]

Planckus, 2025-11-04:
Any having success with integrating the L-series Foxess chargers? I can connect it to EVCC, but for some reason it won’t start charging when I connect it to the car.

lucode, 2025-11-05:
> Any having success with integrating the L-series Foxess chargers? I can connect it to EVCC, but for some reason it won’t start charging when I connect it to the car.

With this information, it is very unlikely that someone can help you. At least we need the logs of the OCPP protocol. And your config. The log you should take when you try to initiate a charge with fast charge.

lucode, 2025-11-05:
> After much trial and error, I actually got it to work - Can only be initiated with the RFID-cards.

You need to set the charger into  "Controlled mode", otherwise it doesn't is under OCPP control. You can only have controlled or by App/RFID.

[image]

lucode, 2026-03-10:
> Hi [@lucode](https://github.com/lucode) I'm glad I found this post I am considering buying the fox A7300P1-E-B charger because I will have a the fox inverter
> 
> That model is not listed by you but perhaps because this thread is a bit old and that model is newer? But it does support ocpp 1.6
> 
> I'm just wondering though it looks like the ocpp might go through the cloud not local because if it's cloud I heard there is a 5 minute lag with fox servers?
> 
> I'm actually considering using modbus to connect to the charger because then it will be local but I can't confirm if the modbus on the charger will accept commands or if it's read only?
> 
> I want to be able to adjust amps dynamically so it doesn't max out the inverter and use grid power only battery power etc
> 
> If you are able to help in anyway that would be amazing cheers

As far what I can see in the specs this série has the same abilities as the older series. It has modbus and OCPP built in.
https://www.fox-ess.com/Public/Uploads/uploadfile/files/Download/EN-A-V2.0-EV-Charger-AEB-Datasheet-V1.1-20250619.pdf

Planckus, 2026-03-11:
Where do you set the auto charge feature ? It doesn’t work when I set I up in the app, and the switch to auto start. It just says mode not supported.

lucode, 2026-03-11:
It is just the FOX ESS AC charger.

[image]

In the screen that follows this one, (After the connections is established) you need to click on the Advanced Options where you find the Remote Start (Transaction).