evcc-io/evcc #16450: Better support for Loadpoints (PV mode) in offgrid and no feedin scenarios

lucode, 2024-10-02:
**Is your feature request related to a problem? Please describe.**
I know there is already some discussions under way how to improve the PV mode in different ways.
(see here https://github.com/evcc-io/evcc/issues/14578 and here https://github.com/evcc-io/evcc/discussions/3530)
But as the discussion in 3530 seems to me to be a bit bloated and the issue 14578 is locked I start a new one.
And my problem is, I suppose, quite specific and and I have a possible solution, which doesn't require a lot of redesign.

The issue that I have, is that in Solar (pv) mode I can't be sure that the charging process does always start.
```
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: -10 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 500 # maximum import power (W)
```
The problem is that with sufficient solar power, my grid meter basically fluctuates somewhere around the value 0 and is rarely positive or negative for more than 1 minute. Actually, the enable parameter via grid power is completely irrelevant for me. What would be relevant, however, would be a way to tell evcc that if the system feeds more than 2000W into the battery for longer than 5 minutes, the surplus is sufficient to start the charging process. 

There are as well some discussions going on but so far I didn't find a solution, except doing some automation via Home Assistant as Startin in MINPV mode and after the charging process has started switching over to PC mode. 

https://github.com/evcc-io/evcc/discussions/7807
https://github.com/evcc-io/evcc/discussions/16434

**Describe the solution you'd like**
I don't understand why the grid power was used for the loadpoint enable, in principle this value is only relevant if there is no battery or with feedin. Otherwise, it should be relatively easy to derive the potential PV surplus. (PV power - Consumption (without Battery and Grid feedin) ) or simply Battery (feedin) + Grid Power (feedin).
I did make a small spreadsheet and added every possible usage scenario of a site. From this you can see that it would work and the config yaml of existing installation aren't necessary to change.

The calculation I used is quite simple: **Grid Power - Grid Power - Grid Power + Battery Power**
(The grid power feedin needs to be inverted)

Assumptions are :

> Threshold power | 2000 W
> -- | --
> Excess Power | 2010 W

Scenario | has Grid | has PV | has Battery | Grid Feedin | Grid Power | Battery Power | Excess Power | Start Charge
-- | -- | -- | -- | -- | -- | -- | -- | --
Grid, Solar, Bat, Feedin | true | true | true | true | -1005 W | 1005 W | 2010 W | true
Grid, Solar, Feedin | true | true |   | true | -2010 W |   | 2010 W | true
Grid, Solar, Bat | true | true | true |   | 0 W | 2010 W | 2010 W | true
Grid, Solar | true | true |   |   | -2010 W |   | 2010 W | true
Solar, Bat (offgrid) |   | true | true |   |   | 2010 W | 2010 W | true
Grid, Bat | true |   | true |   | 3010 W | 2010 W | -1000 W |  
Grid, Bat, feedin | true |   | true | true | 3010 W | 1005 W | -2005 W |  

</body>

</html>

**Describe alternatives you've considered**
See the links to the discussions.

**Additional context**
In contrast to enable via Grid Power, disable Grid Power is more useful for me. Because if my inverter gets too warm, it reduces the maximum power and the system then falls back on grid power. I would then prefer to interrupt charging, but evcc may eventually also regulate this automatically. 
But the disable parameter does have another problem in my scenario. I don't want to charge my EV with the house battery when not necessary. But it does charge now if the house battery SOC is still above the set threshold because again the disable parameter gets involved one in the case of powering something from the grid. But if the battery is above the minimal level this is not the case. Principally this is a second issue and a bit more complex because I think it would need to have an additional parameter. Something as disable-bat. This parameter would work as the "disable-grid" parameter, but is revealed to the battery power. The existing wouldn't be touched the system just would have a second way to disable charging. It could get triggered from both.

the config would look then as something as this:
 ```
    disable: # pv mode disable behavior from grid power
      delay: 3m # threshold must be exceeded for this long
      threshold: 500 # maximum import power (W)
    disable-bat: # pv mode disable behavior from battery power
      delay: 5m # threshold must be exceeded for this long
      threshold: 700 # maximum import power (W)
```

In contrast to enable via Grid Power, disable via Grid Power is still more useful for me. Because if my inverter gets too warm, it reduces the maximum power and the system then falls back on grid power. I would then prefer to interrupt charging, but evcc may eventually also regulate this automatically.

VolkerK62, 2024-10-02:
When the battery is charged, the batterypower is negative.

andig, 2024-10-02:
Not sure I really understand. Enabling considers site power and that includes battery charging. That should be sufficient to start charging. Are you sure your (evcc) battery settings support that?

/cc @VolkerK62

VolkerK62, 2024-10-02:
>Die Thresholds schauen auf site power nicht auf grid power.
Wenn du die Priorität deiner Batterie entsprechend niedrig einstellst, wird deren Ladeleistung als verfügbarer Überschuss angesehen und spiegelt sich in der site power wider.

https://github.com/evcc-io/evcc/discussions/16434#discussioncomment-10800531

closed 2024-10-02 by andig

lucode, 2024-10-02:
Wir können die Diskussion natürlich hier wiederholen, aber fact ist, es wird immer der Grid Meter als Referenz verwendet. 
Das gilt auch für priority. Doch wenn das Grid Meter immer O misst (plus-minus) funktioniert es nicht. 
> Wichtig für die Funktionalität ist ein  (grid-meter). Dieser ermittelt die aktuelle Überschussleistung.

> threshold
> Defines the power threshold at the grid connection point in watts (W)

VolkerK62, 2024-10-02:
Probier es doch einfach mal aus, indem du die Batterie Priorität auf 0% einstellst.
Und ein Log File wäre nicht schlecht.

lucode, 2024-10-02:
Du meinst loadpoints priority?

VolkerK62, 2024-10-02:
Nein, battery priority
https://docs.evcc.io/docs/features/battery#wohin-geht-die-sonnenenergie-zuerst

andig, 2024-10-02:
Ich mache hier mal zugunsten der Discussion zu. Macht keinen Sinn, das doppelt zu besprechen.