evcc-io/evcc #16456: Ginglong Solis RHI charge discharge problem

fastsonic, 2024-10-02:
### Describe the bug

Hi@all,
i have a Problem with an Ginglong Solis RHI5k that is connected over rs485tcpip
When i configure it with the experimental GUI, it only show loading Battery. Also when the Battery is discharge, it shows charge.
evcc 0.130.13
Solis RHI5K with Battery
[image]
In the Log always say -xxxW
not sure why he not switch to discharge !?

### Steps to reproduce

1. configure Solar over GUI connect RHI5K over rs485TCPIP
2.
3.
...
[image]

### Configuration details

```yaml
not needet configured with GUI
```

### Log details

```text
[site ] DEBUG 2024/10/02 15:36:29 ----
[lp-1 ] DEBUG 2024/10/02 15:36:29 charge power: 0W
[lp-1 ] DEBUG 2024/10/02 15:36:29 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:36:29 pv power: 932W
[site ] DEBUG 2024/10/02 15:36:30 battery soc: 24%
[site ] DEBUG 2024/10/02 15:36:30 battery power: -1165W
[site ] DEBUG 2024/10/02 15:36:30 grid meter: 31W
[site ] DEBUG 2024/10/02 15:36:30 grid currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:36:30 battery has priority at soc 24% (< 40%)
[site ] DEBUG 2024/10/02 15:36:30 site power: 31W
[lp-1 ] DEBUG 2024/10/02 15:36:30 charger status: A
[site ] DEBUG 2024/10/02 15:36:39 ----
[lp-1 ] DEBUG 2024/10/02 15:36:39 charge power: 0W
[lp-1 ] DEBUG 2024/10/02 15:36:39 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:36:39 pv power: 921W
[site ] DEBUG 2024/10/02 15:36:40 battery soc: 24%
[site ] DEBUG 2024/10/02 15:36:40 battery power: -45W
[site ] DEBUG 2024/10/02 15:36:40 grid meter: -227W
[site ] DEBUG 2024/10/02 15:36:40 grid currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:36:40 battery has priority at soc 24% (< 40%)
[site ] DEBUG 2024/10/02 15:36:40 site power: -227W
[lp-1 ] DEBUG 2024/10/02 15:36:40 charger status: A
[site ] DEBUG 2024/10/02 15:36:49 ----
[lp-1 ] DEBUG 2024/10/02 15:36:49 charge power: 0W
[lp-1 ] DEBUG 2024/10/02 15:36:49 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:36:49 pv power: 942W
[site ] DEBUG 2024/10/02 15:36:50 battery soc: 24%
[site ] DEBUG 2024/10/02 15:36:50 battery power: -380W
[site ] DEBUG 2024/10/02 15:36:50 grid meter: -49W
[site ] DEBUG 2024/10/02 15:36:50 grid currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:36:50 battery has priority at soc 24% (< 40%)
[site ] DEBUG 2024/10/02 15:36:50 site power: -49W
[lp-1 ] DEBUG 2024/10/02 15:36:50 charger status: A
[site ] DEBUG 2024/10/02 15:36:59 ----
[lp-1 ] DEBUG 2024/10/02 15:36:59 charge power: 0W
[lp-1 ] DEBUG 2024/10/02 15:36:59 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:36:59 pv power: 1032W
[site ] DEBUG 2024/10/02 15:37:00 battery soc: 24%
[site ] DEBUG 2024/10/02 15:37:00 battery power: -430W
[site ] DEBUG 2024/10/02 15:37:00 grid meter: 6W
[site ] DEBUG 2024/10/02 15:37:00 grid currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:37:00 battery has priority at soc 24% (< 40%)
[site ] DEBUG 2024/10/02 15:37:00 site power: 6W
[lp-1 ] DEBUG 2024/10/02 15:37:00 charger status: A
[site ] DEBUG 2024/10/02 15:37:09 ----
[lp-1 ] DEBUG 2024/10/02 15:37:09 charge power: 0W
[lp-1 ] DEBUG 2024/10/02 15:37:09 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:37:11 pv power: 1176W
[site ] DEBUG 2024/10/02 15:37:11 battery soc: 24%
[site ] DEBUG 2024/10/02 15:37:11 battery power: -1087W
[site ] DEBUG 2024/10/02 15:37:11 grid meter: 110W
[site ] DEBUG 2024/10/02 15:37:11 grid currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:37:11 battery has priority at soc 24% (< 40%)
[site ] DEBUG 2024/10/02 15:37:11 site power: 110W
[lp-1 ] DEBUG 2024/10/02 15:37:12 charger status: A
[site ] DEBUG 2024/10/02 15:37:19 ----
[lp-1 ] DEBUG 2024/10/02 15:37:19 charge power: 0W
[lp-1 ] DEBUG 2024/10/02 15:37:19 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:37:19 pv power: 2223W
[site ] DEBUG 2024/10/02 15:37:20 battery soc: 24%
[site ] DEBUG 2024/10/02 15:37:20 battery power: -570W
[site ] DEBUG 2024/10/02 15:37:20 grid meter: -502W
[site ] DEBUG 2024/10/02 15:37:20 grid currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:37:20 battery has priority at soc 24% (< 40%)
[site ] DEBUG 2024/10/02 15:37:20 site power: -502W
[lp-1 ] DEBUG 2024/10/02 15:37:20 charger status: A
[site ] DEBUG 2024/10/02 15:37:29 ----
[lp-1 ] DEBUG 2024/10/02 15:37:29 charge power: 0W
[lp-1 ] DEBUG 2024/10/02 15:37:29 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:37:29 pv power: 4260W
[site ] DEBUG 2024/10/02 15:37:30 battery soc: 24%
[site ] DEBUG 2024/10/02 15:37:30 battery power: -2489W
[site ] DEBUG 2024/10/02 15:37:30 grid meter: -1750W
[site ] DEBUG 2024/10/02 15:37:30 grid currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:37:30 battery has priority at soc 24% (< 40%)
[site ] DEBUG 2024/10/02 15:37:30 site power: -1750W
[lp-1 ] DEBUG 2024/10/02 15:37:30 charger status: A
[site ] DEBUG 2024/10/02 15:37:39 ----
[lp-1 ] DEBUG 2024/10/02 15:37:39 charge power: 0W
[lp-1 ] DEBUG 2024/10/02 15:37:39 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:37:39 pv power: 4359W
[site ] DEBUG 2024/10/02 15:37:40 battery soc: 25%
[site ] DEBUG 2024/10/02 15:37:40 battery power: -3248W
[site ] DEBUG 2024/10/02 15:37:40 grid meter: -56W
[site ] DEBUG 2024/10/02 15:37:40 grid currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:37:40 battery has priority at soc 25% (< 40%)
[site ] DEBUG 2024/10/02 15:37:40 site power: -56W
[lp-1 ] DEBUG 2024/10/02 15:37:40 charger status: A
[site ] DEBUG 2024/10/02 15:37:49 ----
[lp-1 ] DEBUG 2024/10/02 15:37:49 charge power: 0W
[lp-1 ] DEBUG 2024/10/02 15:37:49 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:37:50 pv power: 3255W
[site ] DEBUG 2024/10/02 15:37:51 battery soc: 25%
[site ] DEBUG 2024/10/02 15:37:51 battery power: -3094W
[site ] DEBUG 2024/10/02 15:37:51 grid meter: -56W
[site ] DEBUG 2024/10/02 15:37:51 grid currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:37:51 battery has priority at soc 25% (< 40%)
[site ] DEBUG 2024/10/02 15:37:51 site power: -56W
[lp-1 ] DEBUG 2024/10/02 15:37:51 charger status: A
[site ] DEBUG 2024/10/02 15:37:59 ----
[lp-1 ] DEBUG 2024/10/02 15:37:59 charge power: 0W
[lp-1 ] DEBUG 2024/10/02 15:37:59 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:37:59 pv power: 2810W
[site ] DEBUG 2024/10/02 15:38:00 battery soc: 25%
[site ] DEBUG 2024/10/02 15:38:00 battery power: -3707W
[site ] DEBUG 2024/10/02 15:38:00 grid meter: -56W
[site ] DEBUG 2024/10/02 15:38:00 grid currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:38:00 battery has priority at soc 25% (< 40%)
[site ] DEBUG 2024/10/02 15:38:00 site power: -56W
[lp-1 ] DEBUG 2024/10/02 15:38:00 charger status: A
[site ] DEBUG 2024/10/02 15:38:09 ----
[lp-1 ] DEBUG 2024/10/02 15:38:09 charge power: 0W
[lp-1 ] DEBUG 2024/10/02 15:38:09 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:38:09 pv power: 2114W
[site ] DEBUG 2024/10/02 15:38:10 battery soc: 25%
[site ] DEBUG 2024/10/02 15:38:10 battery power: -3023W
[site ] DEBUG 2024/10/02 15:38:10 grid meter: -56W
[site ] DEBUG 2024/10/02 15:38:10 grid currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:38:10 battery has priority at soc 25% (< 40%)
[site ] DEBUG 2024/10/02 15:38:10 site power: -56W
[lp-1 ] DEBUG 2024/10/02 15:38:10 charger status: A
[site ] DEBUG 2024/10/02 15:38:19 ----
[lp-1 ] DEBUG 2024/10/02 15:38:19 charge power: 0W
[lp-1 ] DEBUG 2024/10/02 15:38:19 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:38:20 pv power: 1400W
[site ] DEBUG 2024/10/02 15:38:21 battery soc: 25%
[site ] DEBUG 2024/10/02 15:38:21 battery power: -1033W
[site ] DEBUG 2024/10/02 15:38:21 grid meter: -56W
[site ] DEBUG 2024/10/02 15:38:21 grid currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:38:21 battery has priority at soc 25% (< 40%)
[site ] DEBUG 2024/10/02 15:38:21 site power: -56W
[lp-1 ] DEBUG 2024/10/02 15:38:21 charger status: A
[site ] DEBUG 2024/10/02 15:38:29 ----
[lp-1 ] DEBUG 2024/10/02 15:38:29 charge power: 0W
[lp-1 ] DEBUG 2024/10/02 15:38:29 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:38:29 pv power: 1116W
[site ] DEBUG 2024/10/02 15:38:35 battery soc: 25%
[site ] DEBUG 2024/10/02 15:38:35 battery power: -1030W
[site ] DEBUG 2024/10/02 15:38:35 grid meter: -56W
[site ] DEBUG 2024/10/02 15:38:35 grid currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:38:35 battery has priority at soc 25% (< 40%)
[site ] DEBUG 2024/10/02 15:38:35 site power: -56W
[lp-1 ] DEBUG 2024/10/02 15:38:35 charger status: A
[site ] DEBUG 2024/10/02 15:38:39 ----
[lp-1 ] DEBUG 2024/10/02 15:38:39 charge power: 0W
[lp-1 ] DEBUG 2024/10/02 15:38:39 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/02 15:38:39 pv power: 1063W
[site ] DEBUG 2024/10/02 15:38:40 battery soc: 25%
[site ] DEBUG 2024/10/02 15:38:40 battery power: -1487W
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.13

andig, 2024-10-03:
Es braucht bitte mal ein

    evcc meter

jeweils beim Laden und Entladen

fastsonic, 2024-10-04:
Hi,
evcc läuft als addon im Homeassistant.
ich komme in den Container, aber nach dem absenden von evcc Meter bekomme ich folgende Meldung:
➜  ~ docker exec -it addon_49686a9f_evcc sh       
/app # evcc meter
[main  ] INFO 2024/10/04 09:07:18 evcc 0.130.13
[main  ] FATAL 2024/10/04 09:07:18 Config File "evcc" Not Found in "[/root /app /etc]"

Oben im Log sieht man jeweils ein paar Lademomente aber auch Entlademomente. Der Wert vom Wechselrichter ist aber immer negativ. verarbeitet ihr "current direction" 
Gruß

andig, 2024-10-04:
Es braucht bitte ein Log. Dafür braucht es kein HA. Welche Register verwendet werden ist im Template ersichtlich.

fastsonic, 2024-10-04:
Gibt es noch eine andere Möglichkeit wie ich da ran komme ? Wenn ich das richtig verstehe geht das nicht!? 
"""""""""""""""""""""
hm, problem.
evcc läuft hier als homeassistant addon.
zugriff auf evcc über das HA docker funktioniert nicht so richtig:

/app # evcc meter
[main  ] INFO 2024/06/07 14:31:34 evcc 0.126.6
[main  ] FATAL 2024/06/07 14:31:34 Config File "evcc" Not Found in "[/root /app /etc]"
/app # 
ich bin jetzt kein linux experte, als das ich auf die richtige config.yaml verweisen kann, welche ja außerhalb vom evcc docker im ha config verzeichniss liegt....

@[VolkerK62](https://github.com/VolkerK62)
VolkerK62
[on Jun 7](https://github.com/evcc-io/evcc/discussions/14232#discussioncomment-9701904)
**ne, unter HA geht das nicht.**
"""""""""""""""""""""""""""""""""""""""

Wenn es nicht ander möglich ist, muss ich wenn Zeit ist erst EVCC wo anderst laufen lassen.
Gruß

andig, 2024-10-08:
Das Batterievorzeichen wurde erst in https://github.com/evcc-io/evcc/pull/14844 geändert aufgrund https://github.com/evcc-io/evcc/issues/14711. 

@dapetra @cs33lm hat ihr unterschiedliches Verhalten bzgl. Vorzeichen der Batterieleistung?

fastsonic, 2024-10-08:
Soweit ich weiß ändert sich nicht das Vorzeichen des Current Registers.
Es gibt nur ein zusätzliches Register der die Richtung mit 1 oder 0 anzeigt. 
Weiterhin einen Register für Ladeleistung und ein Register für Entladeleistung.
Ich schau mal das ich die Register Für die getrennte Lade und Entladeleistung bekomme..

fastsonic, 2024-10-08:
Wenn ich das richtig aus der Doku ziehe von Solax Plugin:
register 33134 ist battery Current oder power 33149
dann gibt es Register 33135 das die Richtung mit 1/0 bestimmt.
Ab Zeile 1441
Aber ich weiß nicht wie hier  Battery Input Energy und Output entsteht 
Zeile 1512 
[https://github.com/wills106/homeassistant-solax-modbus/blob/main/custom_components/solax_modbus/plugin_solis.py](url)

andig, 2024-10-08:
Super Input, danke. Da wir das jetzt aber schon 2x geändert haben fände ich es hilfereich wenns dazu Einigung gibt ;)

fastsonic, 2024-10-08:
Das ist garnicht so einfach.. 
Bei meinem zweiten Solis:
S6-GR1P4K habe ich die Infos das die Register alle -1 sind, also bei rhi5k hybrid adresse 3001 ist es bei dem S6 inverter 3000.
Ist es aber wieder der S6 Hybrid passen die alten Adressen vom rhi5k. Das aber noch ohne Gewähr da ich mich da erst durchwurschtel.
Im Hinterkopf behalten wenn einer mit einem S6 kommt der nicht will...
Hat Zeit... Bin selber gespannt ob es bei den anderen wieder anderst ist...

andig, 2024-10-19:
> S6-GR1P4K habe ich die Infos das die Register alle -1 sind

Das kannst Du ignorieren. ich mache hier mal zu. Gerne kommentieren wenn es verbindliche, getestete Information zu notwendigen Registern gibt.

fastsonic, 2024-10-22:
für einen RHI5k-48 funktioniert es definitiv nicht.

Hier nochmal alle Infos die ich auftreiben konnte:
[https://github.com/wills106/homeassistant-solax-modbus/blob/main/custom_components/solax_modbus/plugin_solis.py](url)

hier aus der definition, wie Das SolaxPlugin in HA schon die Werte zur Verfügung stellt:
Ab Zeile 1512
SolisModbusSensorEntityDescription(
        name = "Battery Input Energy",
        key = "battery_input_energy",
        native_unit_of_measurement = UnitOfPower.WATT,
        device_class = SensorDeviceClass.POWER,
        state_class = SensorStateClass.MEASUREMENT,
        value_function = value_function_battery_input_solis,
        allowedtypes = HYBRID,
        icon = "mdi:battery-arrow-up",
    ),
    SolisModbusSensorEntityDescription(
        name = "Battery Output Energy",
        key = "battery_output_energy",
        native_unit_of_measurement = UnitOfPower.WATT,
        device_class = SensorDeviceClass.POWER,
        state_class = SensorStateClass.MEASUREMENT,
        value_function = value_function_battery_output_solis,
        allowedtypes = HYBRID,
        icon = "mdi:battery-arrow-down",

hier wird die Berechnung schon im Adapter gemacht mit  :

def value_function_battery_output_solis(initval, descr, datadict):
    inout = datadict.get('battery_charge_direction', 0)
    val = datadict.get('battery_power', 0)
    if inout == 1: return abs(val)
    else: return 0

def value_function_battery_input_solis(initval, descr, datadict):
    inout = datadict.get('battery_charge_direction', 0)
    val = datadict.get('battery_power', 0)
    if inout == 0: return val
    else: return 0

Wie oben schon beschrieben ist der Powerwert immer positiv und wird nur mit dem zusätzlichen Register für die Direction für ECC brauchbar. 
Das gilt für den Solis RHI5K-48

das Register für Battery Power ist:
Zeile 1500   
Register steht in 1505   register = 33149

Das Register für Batter Direction ist:
Zeile 1399
Register steht in 1401  register = 33135

Wenn das nicht umgesetzt wird, bitte kurz info. 
Dann hol ich mir die Werte wieder aus HA mit mqtt und bastel mir da das ganz wieder um.

Gruß
Sascha

premultiply, 2024-10-22:
Die entscheidende Funktion ist hier `abs()` in Verbindung mit dem getrennten Richtungs-Register.

Offensichtlich mischen hier wie bei SunGrow unterschiedliche Firmware-Versionen die Register-Werte.
Mal mit Vorzeichen, mal ohne.

premultiply, 2024-10-22:
> Weder RHI noch S funktioniert hier. Da sich das Vorzeichen nicht ändert !

Kann ich nicht nachvollziehen. `solis-hybrid` implementiert die Variante mit signed integer. `solis-hybrid-s` die Variante mit dem separaten Vorzeichenregister.

closed 2024-10-22 by premultiply

andig, 2024-10-22:
Bitte Remotezugang zum WR an info@evcc.io.

fastsonic, 2024-10-22:
> > Weder RHI noch S funktioniert hier. Da sich das Vorzeichen nicht ändert !
> 
> Kann ich nicht nachvollziehen. `solis-hybrid` implementiert die Variante mit signed integer. `solis-hybrid-s` die Variante mit dem separaten Vorzeichenregister.

Wenn du sie über Web aufnimmst ? ( nicht in der evcc.yaml ) !?

fastsonic, 2024-10-22:
> Bitte Remotezugang zum WR an [info@evcc.io](mailto:info@evcc.io).

ich melde mich für einen Remotezugang..

andig, 2024-10-23:
> When i configure it with the experimental GUI, it only show loading Battery. Also when the Battery is discharge, it shows charge.

Mit remote access getestet. Ergebnis:

```
rhi5k-1
-------
[rhi5k-1] TRACE 2024/10/23 15:13:39 modbus: send 00 01 00 00 00 06 01 04 81 7d 00 02
[rhi5k-1] TRACE 2024/10/23 15:13:39 modbus: recv 00 01 00 00 00 07 01 04 04 00 00 02 6e
[rhi5k-1] TRACE 2024/10/23 15:13:39 modbus: send 00 02 00 00 00 06 01 04 81 8d 00 02
[rhi5k-1] TRACE 2024/10/23 15:13:39 modbus: recv 00 02 00 00 00 07 01 04 04 00 00 10 9d
[rhi5k-1] TRACE 2024/10/23 15:13:39 modbus: send 00 03 00 00 00 06 01 04 81 73 00 01
[rhi5k-1] TRACE 2024/10/23 15:13:40 modbus: recv 00 03 00 00 00 05 01 04 02 00 32
Power:  -622W
Energy: 4253.0kWh
Soc:    50%

rhi5k-2
-------
[rhi5k-2] TRACE 2024/10/23 15:13:40 modbus: send 00 04 00 00 00 06 01 04 81 7d 00 02
[rhi5k-2] TRACE 2024/10/23 15:13:40 modbus: recv 00 04 00 00 00 07 01 04 04 00 00 02 69
[rhi5k-2] TRACE 2024/10/23 15:13:40 modbus: send 00 05 00 00 00 06 01 04 81 6f 00 01
[rhi5k-2] TRACE 2024/10/23 15:13:40 modbus: recv 00 05 00 00 00 05 01 04 02 00 01
[rhi5k-2] TRACE 2024/10/23 15:13:40 modbus: send 00 06 00 00 00 06 01 04 81 8d 00 02
[rhi5k-2] TRACE 2024/10/23 15:13:41 modbus: recv 00 06 00 00 00 07 01 04 04 00 00 10 9d
[rhi5k-2] TRACE 2024/10/23 15:13:41 modbus: send 00 07 00 00 00 06 01 04 81 73 00 01
[rhi5k-2] TRACE 2024/10/23 15:13:41 modbus: recv 00 07 00 00 00 05 01 04 02 00 32
Power:  617W
Energy: 4253.0kWh
Soc:    50%
```

Bei Entladung zeigt `solis-hybrid-s` das richtige Ergebnis. Allerdings ist die Bezeichnung der Templates unglücklich, da es auch den RHI anscheinend in 2 Varianten gibt.