evcc-io/evcc #16460: SMA ST SE Aktive Batteriesteuerung

MarSlz, 2024-10-02:
### Describe the bug

Beim schnellen Laden wird die Batterie entladen, obwohl diese als gesperrt angezeigt wird.
[image]

Zudem scheint 

Hardware:
SMA Hybrid Wechselrichter
SUNNY TRIPOWER 10.0 SE
Firmware-Version | 3.4.21.R
Batterie-Hersteller | SMA

### Steps to reproduce

1. SMA Hybridwechselrichter einrichten
2. Entladung beim Schnelladen verhindern aktivieren
3. Auto laden

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: xxxxx

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
telemetry: true
plant: xxxxx

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  ID.4: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
[... 386 lines cut ...]
[sma   ] TRACE 2024/10/02 17:27:37 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/02 17:27:37 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/02 17:27:37 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/02 17:27:37 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/02 17:27:37 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/02 17:27:38 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/02 17:27:38 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/10/02 17:27:38 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[site  ] DEBUG 2024/10/02 17:27:38 ----
[lp-1  ] DEBUG 2024/10/02 17:27:38 charge power: 4127W
[lp-1  ] DEBUG 2024/10/02 17:27:38 charge currents: [5.91 5.99 5.99]A
[site  ] WARN 2024/10/02 17:27:38 planner: no matching rate
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.13

andig, 2024-10-04:
Zu Diagnosezwecken mal Batteriesteuerung in evcc deaktivieren und dann auf der Kommandozeile testen ob das funktionert/ den gewünschten Effekt hat:

    evcc meter --log debug --battery-mode hold --battery-wait 5m

wittmannchris, 2024-10-04:
Sunny home manager Fw 2.15.x ?

mit der 2.14.3 funktioniert das nicht.

andig, 2024-10-05:
@MarSlz welche Version hat Dein SHM?

MarSlz, 2024-10-06:
Hallo 
Ich habe noch 2.14.13.R.
Automatische Updates sind aber aktiviert.

closed 2024-10-06 by andig