evcc-io/evcc #16464: Request to add Charger

MrB-AU, 2024-10-02:
My charger appears not to be supported and wondered if it could be added.  It goes by a few different names.  Mine is an EVSwitch [http://evswitchstore.com.au/](url) however it is also known as the EV+ Caro [https://www.en-plustech.com/product/caro-pro-series/](url)

Also, do you plan to add the BYD Atto 3 to your vehicle list?

thanks
Chris

premultiply, 2024-10-02:
It says OCPP is supported.
Should work. Have you tried it?

jeffborg, 2024-10-02:
@MrB-AU I have another branded en+ wallbox and it's working just fine. The firmware and setup screenshots all look identical.

Not sure about byd cars

MrB-AU, 2024-10-03:
Ok. So when setting up EVCC do I just choose the Generic OCCP charger as my option?  I have done this but get a load point error. 

Do I need to manually edit anything in the yaml files? 

Fully declaring my limited technical knowledge here. On a steep learning curve of entering/setting all this up through terminal on my Mac.