evcc-io/evcc #16466: Expansion of the “Threshold” setting to include stepping power up and down

pssturges, 2024-10-02:
In solar mode, a user may want to prioritise using all available PV even at the expense of using a little from the grid.

In my case, I am using 3 phase power at 240v. The charger power can be stepped up and down in 1A increments. That means the power increments are approx 720w. Currently, while charging, if excess PV isn’t greater than 720w, that excess won’t be used to charge and instead exported.  It would be nice if the user could specify a small amount (say 200w) they are happy to draw from the grid in order to be able to step up to the next power increment and therefore use all available excess. The same should apply in reverse as excess drops.

Not sure if this should be configurable separately to the current Threshold setting, or just a seperate opt in setting to the current one, or perhaps just change the current behaviour. 

Thanks for your consideration

VolkerK62, 2024-10-03:
I guess, this should work with "negative residualpower".

pssturges, 2024-10-03:
Ah OK. I wasn’t aware of that setting. I’ve added it now and initial testing looks good. I will continue to watch it over the next few days. Curious how that setting interacts with the threshold setting. Seems like they might clash at times? Perhaps I might need to disable the threshold setting?

thanks!

VolkerK62, 2024-10-03:
The thresholds are looking on `site power` (= available charging power) not on `grid power`.
https://docs.evcc.io/en/docs/reference/configuration#manipulation-options

jeffborg, 2024-10-03:
@pssturges it setting the residual power to say -500 will effectively lower your pv thresholds by 500w as well. You can simply increase them by 500 if you want the old settings

andig, 2024-10-03:
Das ist https://github.com/evcc-io/evcc/issues/14578

closed 2024-10-03 by andig