evcc-io/evcc #16475: Error when adding slimmemeter

denniswham, 2024-10-03:
### Describe the bug

When adding a slimmemeter as a meter, I get a 404

### Steps to reproduce

I've added the following node in my config
`meters:
  - name: slimmemeter
    type: template
    template: slimmelezer
    usage: grid
    host: 10.0.20.12 # IP-Adresse oder Hostname`

When starting EVCC I get ERROR 2024/10/03 14:42:36 grid meter: add[0]: unexpected status: 404 (Not Found)

When go to either of the following URLs in a browser I get correct data
http://10.0.20.12/sensor/power_consumed
http://10.0.20.12/sensor/power_consumed_phase_3

### Configuration details

```yaml
meters:
  - name: slimmemeter
    type: template
    template: slimmelezer
    usage: grid
    host: 10.0.20.12 # IP-Adresse oder Hostname
```

### Log details

```text
ERROR 2024/10/03 14:42:36 grid meter: add[0]: unexpected status: 404 (Not Found)
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

andig, 2024-10-03:
See template- the URLs are different:

    /sensor/power_returned

andig, 2024-10-03:
It is: https://docs.evcc.io/docs/devices/meters#slimmelezer-v2

closed 2024-10-03 by andig