evcc-io/evcc #16487: Resend charge current setpoint after Modbus connection is restored

busyb-now, 2024-10-03:
**Is your feature request related to a problem? Please describe.**
I use a Vestel EVC04 wallbox together with evcc via Modbus communication. Because the network connection to the garage is a bit unstable, there are i/o timeout messages in evcc from time to time. During and after a connection interruption, the wallbox continues to run with the “fail-safe current” configured in the wallbox. I have currently set this to the minimum current of 6 A. My problem: Even after the connection is restored, the box continues to run at this fail-safe current. This is particularly annoying in the "Fast" and "PV" modes until the configured minimum charge level is reached. In these modes, evcc presumably only sends the charging current setpoint to the box once and expects the box to continue running with this current until a new setpoint  is sent. 

**Describe the solution you'd like**
Is it possible to extend evcc so that the last setpoint sent to the wallbox is sent again after a lost connection is restored? With a stable connection, this should never happen and therefore not cause any restrictions. With my unstable connection, however, this would prevent the box from getting stuck on the wrong current setpoint. 

**Describe alternatives you've considered**
I have tried other settings for the “fail-safe power” setting in the wallbox. At 0 A, the car aborts the charging process completely after a few connection interruptions and thus charging interruptions. At 16 A, the proportion of energy erroneously not charged from PV increases because, for example, charging is carried out at full power from the grid for 3 minutes at night when the connection is interrupted. 

**Additional context**
Furthermore I would be happy about every hint on how I can make evcc more tolerant to connection interruptions.
[evcc-20241003-215727-debug.log](https://github.com/user-attachments/files/17249780/evcc-20241003-215727-debug.log)
[evcc.yaml.txt](https://github.com/user-attachments/files/17249782/evcc.yaml.txt)

andig, 2024-10-03:
Evcc does not „see“ this connection los. If setpoint current could be read that could be used to verify. In general, you‘ll need to fix the root cause- unstable network.

andig, 2024-10-04:
Probier mal bitte, ob der PR Dir hilft. Müsstest Du allerdings selbst compilieren.

busyb-now, 2024-10-04:
Sorry, habe leider keine Erfahrung damit, evcc selbst zu kompilieren. Gibt es dazu ein "Getting started"? Wüsste auch nicht, wie ich mein selbst kompiliertes evcc wie bisher als Add-on in Home Assistant nutzen und testen kann.

andig, 2024-10-04:
Gar nicht, aber Du kannst das ohne HA testen. Compilieren ist in der README dokumentiert: https://github.com/evcc-io/evcc/blob/master/CONTRIBUTING.md.

andig, 2024-10-04:
Kannst du morgen im nightly testen.

closed 2024-10-04 by andig

andig, 2024-10-05:
Bin gespannt auf Ergebnis 👍🏻

busyb-now, 2024-10-05:
Wäre ich auch. Aber wie kann ich das nightly unter Windows oder in HA nutzen? Habe in der Doku nur gefunden, wie ich das unter Linux installieren kann. Habe aber keinen Zugriff auf eine Linux-Maschine.