evcc-io/evcc #16490: Struktur für Sunspec Templates

runtologist, 2024-10-04:
Ich habe einen Vorschlag zu Sunspec Templates.

Hintergrund: Ich habe seit kurzem einen Kontron Solbrid Hybridwechselrichter. Das Ding ist absoluter Sunspec Standard, keinerlei Quirks: Gridzähler über Model 203, Model 160 für insgesamt 4 MPPTs, Model 802 für die Batterie. Im Gegensatz zu anderen Modellen taucht die Batterie nicht zusätzlich als MMPT auf, was AFAIU eigentlich auch nicht im Standard vorgesehen ist. Ich habe also erwartet, dass der WR super einfach einzurichten ist. Das war aber nicht der Fall. Es gibt kein Template, welches diese Kombination unterstützt. Wenn man die Config aus verschiedenen Templates zusammen stellt (`grid` aus `sunspec-hybrid`, `battery` aus `sunspec-battery-control`, etc.) es gibt kein Template, welches Model 160 mit 4 MPPTs unterstützt, und `battery` ohne die Annahme, dass es einen fake MPPT dafür gibt. 

Mögliche Lösung wäre nun, dass ich für genau meinen Wechselrichter ein Template anlege, und das habe ich lokal auch erstmal so gemacht. Allerdings skaliert das eher schlecht im Sinne von Wartungs- und Entwicklungsaufwand. Idealer Weise legt sich ein evcc Nutzer einen standardkonformen Wechselrichter zu, und der funktioniert dann einfach. Daher folgender Vorschlag:

* Sunspec Templates werden nach Funktion aufgeteilt. Anstatt `grid`, `pv`, und `battery` zu unterstützen, unterstützen sie genau eines, z.B. 
  * `sunspec-grid`
  * `sunspec-pv`
  * `sunspec-battery`
* Wo es mehrere kompatible Sunspec Modelle gibt, landet alles in einem Template. z.B. sind aus evcc Sicht 203 und 213 kompatibel, daher gibt es nur ein `sunspec-grid` Template, welches beide unterstützt.
* Wo es mehrere inkompatible Sunspec Modelle gibt, gibt es mehrere Templates. z.B. gibt es ein `sunspec-pv-160` und ein `sunspec-pv-103` Template, und ein `sunspec-battery-802` und ein `sunspec-battery-160` Template.
* Varianten oder quirks werden separate Templates. z.B. gibt es ein `sunspec-pv-160-2mppt` Template, welches 2 MPPTs unterstützt, und ein `sunspec-pv-160-4mppt` Template, welches 4 MPPTs unterstützt.

Vorteile:
* Templates sind einfacher zu warten, da sie nur eine Funktion haben.
* Für einen neuen WR schaue ich ins Datenblatt, greife das passende Template, und es funktioniert.
* Auf Sicht weniger Templates, da nicht jede Kombination von Modellen ein eigenes Template braucht.
* Für einen neuen WR, der nicht unterstützt wird, oder aus Komfortgründen, kann ich nach wie vor ein neues Template anlegen.

Nachteile:
* Mehr `sunspec-*` Templates
* Braucht ggf etwas besser Doku gegenüber einem Template pro WR.

andig, 2024-10-04:
Die Idee finde ich prinzipiell gut. Wie gut die dann in der Praxis ist müsste man probieren. V.a. sollte der Anwender eine Chance haben zu verstehen, was zu verwenden ist. Z.B kann ein `sunspec-battery-802` m.E. nur für AC-gekoppelte WR funktionieren _oder_ muss doch wieder die 160 benutzen.
Zu 4-MPPT gab es ebenfalls schon mal einen PR (den ich grad nicht wieder finde).

Magst Du PR machen? Vielleicht wäre es auch hilfreich, hier vorher überhaupt mal die möglichen Konstellationen aufzulisten.

/cc @premultiply

runtologist, 2024-10-04:
> Z.B kann ein `sunspec-battery-802` m.E. nur für AC-gekoppelte WR funktionieren _oder_ muss doch wieder die 160 benutzen.

Meiner ist DC gekoppelt, das funktioniert. :)

runtologist, 2024-10-04:
Ich liste gerne die Kombinationen und mache nen PR auf.

andig, 2024-10-04:
...und woher bekommst Du dann die Batterieleistung? Aus `W` jedenfalls nicht?

runtologist, 2024-10-04:
Aus `W`. Den Wert liefert ja der WR, nicht die Batterie.

andig, 2024-10-04:
Ist eine Einzeldiskussion- aber bei einem Hybrid-WR kann das ja allenfalls die AC Entladeleistung sein, aber nie die DC Ladung. Du siehst also nicht, welche Ladeleistung eigentlich evcc zur Verfügung hätte die stattdessen in den Akku geht. Das können wir aber gerne einzeln anschauen wenns eine Liste der potenziellen Templates gibt. Vllt. habe ich auch einen Denkfehler...

runtologist, 2024-10-05:
[image]

andig, 2024-10-05:
Spannend. Warum zeigt Deine Batterie DC Ladung als AC an? Das hätte ich so nicht erwartet und ist m.W. auch bei anderen nicht der Fall.

/cc @premultiply

runtologist, 2024-10-05:
@andig Der Hybridwechselrichter liefert in Model 802 die Leistung, die an die Batterie geht. Der Wert kommt nicht von der Batterie selbst.

andig, 2024-10-05:
Und dennoch ist `W` in Sunspec ein AC Wert, denn sonst müsste er `DCW` heissen.

runtologist, 2024-10-05:
Sicher, dass das nicht einfach inkonsistent ist? Die Doku zu dem Wert ist "Total power flowing to/from the battery bank." ohne Einschränkuing auf AC/DC. Current ist ausdrücklich DC.

andig, 2024-10-05:
Das weiss man leider nie... Können wir ja mal so als Hypothese nehmen. War hier aber eh der Einzelfall und wie wollten ja auf das Gesamtbild schauen.

runtologist, 2024-10-07:
Hier eine Gesamtliste, auf Basis dessen, was wir heute haben zzgl. dem,  was bei mir fehlte:

## All combinations

### grid meter
// 3 phase grid meter implementing the full model
`sunspec-grid-3phase-wph` (213 | 203) [power, energy, currents, voltages, powers]
// only if optional `powers` are not supported, not sure we need this
`sunspec-grid-3phase` (213 | 203) [power, energy, currents, voltages]
// any 1 to 3 phase grid meter. `currents`, `voltages` and `powers` are not supported.
`sunspec-grid` (211 | 212 | 213 | 201 | 202 | 203) [power, energy]

### pv
// any inverter, 1 or 3 phase, using total power and energy
`sunspec-pv-inverter` (101 | 103 | 111 | 113 | 102 | 112) [power, energy]
// values derived from sum of DC strings, use only if not all MPPTs should be counted, e.g. there are 4 MPPTs where 1,2 are PV and 3,4 are battery => use sunspec-pv-mppt-2. Use only if sunspec-pv-inverter is not applicable.
`sunspec-pv-mppt-2` (160) [power, energy]
`sunspec-pv-mppt-3` (160) [power, energy]
`sunspec-pv-mppt-4` (160) [power, energy]
...

### battery
// 124 basic storage control, 802 basic battery.
// Prefer 124 for battery control. 124 allows for actually putting the battery on hold, 802 only allows for setting the min, around which it might oscillate. Prefer to read power from 802, fall back to suitable inverter. `inverter`, if the battery has a dedicated inverter, `mppt-n` to complement the corresponding pv inverter, e.g. `mppt-2` if mppt 1,2 are pv and 3,4 are battery
`sunspec-battery-124+802` (124 && 802) [power, soc, batterymode, capacity]
// MinRsvPct is optional, if not supported, use sunspec-battery-124+802
`sunspec-battery-124+802-limitsoc` (124 && 108) [power, soc, batterymode, limitsoc, capacity]
`sunspec-battery-802` (802) [power, soc, capacity]
// MinRsvPct is optional, if not supported, use sunspec-battery-802
`sunspec-battery-802-limitsoc` (802) [power, soc, limitsoc, capacity]
`sunspec-battery-124+inverter` (124 && (101 | 103 | 111 | 113 | 102 | 112)) [power, energy, soc, batterymode, capacity]
`sunspec-battery-124+mppt-2` (124 && 160) [power, energy, soc, batterymode, capacity]
`sunspec-battery-124+mppt-3` (124 && 160) [power, energy, soc, batterymode, capacity]
`sunspec-battery-124+mppt-4` (124 && 160) [power, energy, soc, batterymode, capacity]

Ich bin nicht sicher, ob wir die optionalen Fälle berücksichtigen wollen, oder ob wir erstmal annehmen, dass die vollständigen Modelle implementiert werden. Ich bin auch nicht sicher, ob 802 ohne 124 existiert, und ob wir die mppt battery varianten brauchen, denn `W` ist mandatory in 802. Wenn wir diese alle raus lassen, haben wir:

## minimal

### grid 
`sunspec-grid-3phase-wph` (213 | 203) [power, energy, currents, voltages, powers]
`sunspec-grid` (211 | 212 | 213 | 201 | 202 | 203) [power, energy]

### pv
`sunspec-pv-inverter` (101 | 103 | 111 | 113 | 102 | 112) [power, energy]

### battery
`sunspec-battery-124+802-limitsoc` (124 && 108) [power, soc, batterymode, limitsoc, capacity]

runtologist, 2024-10-07:
Bei den ganzen optionalen Registern explodiert die Anzahl Templates recht schnell, hauptsächlich für battery control. Evtl. braucht es noch eine Trennung von meter und battery control.

premultiply, 2024-10-08:
Das "MPPT-Problem" tritt ausschließlich bei Hybrid-WR auf, da hierbei DC-seitig zwischen Batterie und PV unterschieden werden muss und SunSpec leider keine DC-Summenleistung für PV liefert. Hier muss man daher über die Summe der (PV-)MPPT-Leistungen gehen.

Bei normalen (=nicht Hybrid-)WR ist die PV-Summe identisch mit der Ausgangsleistung. Dabei haben wir nie ein Problem.

Ich denke es sollte daher reichen hier ein zusätzliches SunSpec-Template `sunspec-hybrid-4mppt` hinzuzufügen welches dann für die Batterie `802:W` verwendet.

Für das bestehende `sunspec-hybrid` Template ist das schwierig, da die meisten dieser Hybrid-WR noch kein "neues" Model 802 kennen sondern dies noch über den Umweg 124+160 abbilden.

andig, 2024-10-09:
Batterie: würde ich mich @premultiply  anschließen. Die Grid Templates mit/ohne Phasen sind mir unklar. Die Datenpunkte sind ja vorhanden und nicht optional- warum brauchen wir da 2 Templates?

runtologist, 2024-10-09:
@andig 201/211 haben mandatory nur Werte für eine Phase, 202/212 für zwei, Wph* ist immer optional, auch für 203/213. Übersehe ich da was?

andig, 2024-10-09:
Für die 2er Templates kenne ich keinen Bedarf. Die 1er Tempaltes haben alle Phasen die es braucht. Und Wph braucht es m.E. gar nicht weil wir das auschließlich für Phasenströme verwenden, die sollten bei Sunspec aber schon "signed" sein?

runtologist, 2024-10-09:
Das heißt Du meinst ein 1er, ein 3er Template, und `powers` brauchen wir gar nicht?

andig, 2024-10-10:
> Das heißt Du meinst ein 1er, ein 3er Template, und powers brauchen wir gar nicht?

M.E. nicht.

> Für das bestehende sunspec-hybrid Template ist das schwierig, da die meisten dieser Hybrid-WR noch kein "neues" Model 802 kennen sondern dies noch über den Umweg 124+160 abbilden.

https://github.com/evcc-io/evcc/discussions/16253#discussioncomment-10895247 ist interessant. Battery control scheint bei Fronius nur über 124 zu funktionieren. Ob sich das verallgemeinern lässt???

runtologist, 2024-10-14:
> [#16253 (comment)](https://github.com/evcc-io/evcc/discussions/16253#discussioncomment-10895247) ist interessant. Battery control scheint bei Fronius nur über 124 zu funktionieren. Ob sich das verallgemeinern lässt???

Mein Plan ist, das nur über die Kombination der nicht-optionalen register in 124 und 802 umzusetzen. Bin noch nicht dazu gekommen, es zu testen.

andig, 2024-10-20:
> Ich denke es sollte daher reichen hier ein zusätzliches SunSpec-Template sunspec-hybrid-4mppt hinzuzufügen welches dann für die Batterie 802:W verwendet.

Siehe https://github.com/evcc-io/evcc/pull/16750

andig, 2024-10-20:
Siehe https://github.com/evcc-io/evcc/pull/16751 für Batterieleistung im "Inverter" Template

premultiply, 2024-10-20:
Sofern ein Hybrid-WR Model 802 unterstützt könnte man sich die PV-Erzeugung theoretisch auch aus
1xx:DCW - 802:W berechnen statt über die Summe der MPPTs zu gehen.
Nur mit 124 fällt das aber auf die Füße, da kann man nur über die (gesplitten) MPPTs gehen

closed 2024-11-01 by github-actions[bot]