evcc-io/evcc #16497: Seit evcc Update ist der Webserver nicht mehr erreichbar

mseele, 2024-10-04:
### Describe the bug

Ich habe heute morgen evcc auf meinem PI geupdated und seitdem ist der Webserver nicht mehr verfügbar. Davor ging alles problemlos.
Es wurden mit dem Update keine anderen Packages geupdated. Auf andere Webserver (lighttpd etc.) komme ich problemlos drauf.
Leider liefert das Log auch in Trace nicht mehr als das der Webserver auf dem eingestellten Port läuft.

### Steps to reproduce

1.curl localhost:9091
curl: (7) Failed to connect to localhost port 9091 after 1 ms: Couldn't connect to server

2. mseele@pi:~ $ sudo netstat -tuln | grep 9091
mseele@pi:~ $
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 9091

log: trace
levels:
  cache: error
```

### Log details

```text
Oct 04 13:36:10 pi systemd[1]: Started evcc.service - evcc.
Oct 04 13:36:10 pi evcc[1487]: [main  ] INFO 2024/10/04 13:36:10 evcc 0.130.13
Oct 04 13:36:10 pi evcc[1487]: [main  ] INFO 2024/10/04 13:36:10 using config file: /etc/evcc.yaml
Oct 04 13:36:10 pi evcc[1487]: [db    ] INFO 2024/10/04 13:36:10 using sqlite database: /var/lib/evcc/evcc.db
Oct 04 13:36:10 pi evcc[1487]: [db    ] TRACE 2024/10/04 13:36:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
Oct 04 13:36:10 pi evcc[1487]: [db    ] TRACE 2024/10/04 13:36:10 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc >
Oct 04 13:36:10 pi evcc[1487]: [db    ] TRACE 2024/10/04 13:36:10 SELECT * FROM `settings` LIMIT 1 -1 <nil>
Oct 04 13:36:10 pi evcc[1487]: [db    ] TRACE 2024/10/04 13:36:10 SELECT * FROM `settings` 33 <nil>
Oct 04 13:36:11 pi evcc[1487]: [db    ] TRACE 2024/10/04 13:36:11 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
Oct 04 13:36:11 pi evcc[1487]: [db    ] TRACE 2024/10/04 13:36:11 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
Oct 04 13:36:11 pi evcc[1487]: [db    ] TRACE 2024/10/04 13:36:11 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
Oct 04 13:36:11 pi evcc[1487]: [db    ] TRACE 2024/10/04 13:36:11 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1>
Oct 04 13:36:11 pi evcc[1487]: [db    ] TRACE 2024/10/04 13:36:11 SELECT * FROM `configs` LIMIT 1 -1 <nil>
Oct 04 13:36:11 pi evcc[1487]: [db    ] TRACE 2024/10/04 13:36:11 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
Oct 04 13:36:11 pi evcc[1487]: [db    ] TRACE 2024/10/04 13:36:11 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
Oct 04 13:36:11 pi evcc[1487]: [main  ] INFO 2024/10/04 13:36:11 listening at :9091
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.130.13

VolkerK62, 2024-10-04:
Deine Config ist recht dünn.
Tibber Pulse im Einsatz?

mseele, 2024-10-04:
Ja, tibber pulse ist im Einsatz. Ich kann gerne die ganze Config schicken
aber da es vor dem Update noch alles ging bin ich davon ausgegangen daß es
nicht an der Config lieg

VolkerK62, 2024-10-04:
Liegt nicht an der Config Aber da kann man halt die Hardware erkennen.
Check mal den Pulse.
Der Fehler tritt auf, wenn der Pulse nicht erreichbar ist.

mseele, 2024-10-04:
Okay mache ich, vielen Dank schon für die Hilfe

andig, 2024-10-04:
Im Zweifel mal ein Release zurück und schauen was passiert.

closed 2024-10-04 by andig