evcc-io/evcc #16515: Display of Energy in "Battery Storage" on configuration page shows wrong numbers

schnekrehuesli, 2024-10-05:
### Describe the bug

[image]

The numer for Energy is obviously wrong or it's unclear where it comes from

### Steps to reproduce

evcc installed on Raspberry Pi 4B
Fronius Symo Gen 24 8 Plus with BYD HVS battery storage 3x 10.2 kWh

### Configuration details

```yaml
meters:
- type: template
  template: fronius-gen24 
  usage: grid  
  host: 192.168.178.10  
  port: 502  
  name: grid1
- type: template
  template: fronius-gen24 
  usage: pv  
  host: 192.168.178.10  
  port: 502  
  name: pv2
- type: template
  template: fronius-gen24 
  usage: battery  
  host: 192.168.178.10  
  port: 502  
  name: battery3
  capacity: 30.6
```

### Log details

```text
[lp-1 ] DEBUG 2024/10/05 21:34:58 charger enable
[lp-1 ] DEBUG 2024/10/05 21:34:58 charger disable
[lp-1 ] DEBUG 2024/10/05 21:34:58 wake-up timer: stop
[site ] DEBUG 2024/10/05 21:35:27 ----
[lp-1 ] DEBUG 2024/10/05 21:35:27 charge power: 0W
[site ] DEBUG 2024/10/05 21:35:27 pv power: 1W
[site ] DEBUG 2024/10/05 21:35:27 battery soc: 94%
[site ] DEBUG 2024/10/05 21:35:27 battery power: 312W
[site ] DEBUG 2024/10/05 21:35:27 grid meter: 4W
[site ] DEBUG 2024/10/05 21:35:28 grid currents: [-0.339 0.975 -0.396]A
[site ] DEBUG 2024/10/05 21:35:28 site power: 566W
[lp-1 ] DEBUG 2024/10/05 21:35:28 charger status: A
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.13

andig, 2024-10-05:
Please show

    evcc meter

schnekrehuesli, 2024-10-05:
xxxx@raspberrypi:~ $ evcc meter
[main  ] INFO 2024/10/06 00:02:21 evcc 0.130.13
[main  ] INFO 2024/10/06 00:02:21 using config file: /etc/evcc.yaml
[db    ] INFO 2024/10/06 00:02:21 using sqlite database: /home/tuluk/.evcc/evcc.db
db:1
----
Power:          4W
Energy:         1311.7kWh
Current L1..L3: -0.567A 0.94A -0.593A

grid1
-----
Power:          4W
Energy:         1311.7kWh
Current L1..L3: -0.567A 0.94A -0.593A

db:2
----
Power: 2W

pv2
---
Power: 2W

battery3
--------
Power:        452W
Energy:       2994.5kWh
Soc:          92%
Capacity:     30.6kWh
Controllable: true

schnekrehuesli, 2024-10-05:
it seems to be the "Energy sourced from the battery" since lifetime, even the numer is not consistent with the number in Fronius' solarweb.com App. Right?
I thought, it is the actual energy stored in the battery

andig, 2024-10-06:
Its meter value/ lifetime.

closed 2024-10-06 by andig