evcc-io/evcc #16519: Entsoe error

djfanatix, 2024-10-06:
### Describe the bug

`site ] WARN 2024/10/06 08:26:28 planner: tariff not available: cannot create tariff type 'entsoe': no data for resolution: PT60M’

I get this error last night and the tariffs are blank in the planner.
I assume it is an entsoe api erro, as my Home Assistant entsoe is also not working, but in the afternoon it was working.

Question: Entsoe updates the tariffs around 14.00 LT, does EVCC save this values in the database? 
Is it possible to save the latest values and use them when the api is offline? (Instead of marking as unavailable)

### Steps to reproduce

1. Entsoe tariffs
2.
3.
...

### Configuration details

```yaml
/
```

### Log details

```text
[site ] WARN 2024/10/06 08:26:28 planner: tariff not available: cannot create tariff type 'entsoe': no data for resolution: PT60M
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.13

bsmout, 2024-10-06:
I can confirm I have exactly the same issue today.

andig, 2024-10-06:
Api issue, not evcc.

djfanatix, 2024-10-06:
Yes It is an API error, but why does EVCC not remember the values?

andig, 2024-10-06:
Because its usually not required.

bolisaf, 2024-10-06:
can we get more specific on the issue here so I can contact entsoe support?

djfanatix, 2024-10-07:
I like to reopen the issue, as entsoe has made changes for Belgian users.

See the issue here:  

[](https://github.com/JaccoR/hass-entso-e/issues/195)

andig, 2024-10-07:
Can you please share an API response for Belgium? Given evcc works in 60m slots right now the next question is how we should handle 15m slots: use the average? or the maximum?

djfanatix, 2024-10-07:
I will check tonight.

But there are no 15M slots in Belgium, and I have not read anywhere that this will implemented in the near future.

So we could use the first one (or average) for the time being.

rwegnerGermany, 2024-10-07:
I don't think it's limited to Belgium users. For Germany I also do have problems to get the prices, although with a different error message.
Please see [https://github.com/evcc-io/evcc/discussions/16539#discussioncomment-10865062](url)

To my understanding EVCC requires an update because of entsoe API changes within the last week.

As a workaround I just switched to Fraunhofer.

wonko, 2024-10-07:
This is an example request and response with some 15M data entries: 

```
➜ curl https://web-api.tp.entsoe.eu/api\?securityToken\=<REDACTED>\&documentType\=A44\&out_Domain\=10YBE----------2\&in_Domain\=10YBE----------2\&periodStart\=202410060000\&periodEnd\=202410072300

<?xml version="1.0" encoding="utf-8"?>
  <Publication_MarketDocument xmlns="urn:iec62325.351:tc57wg16:451-3:publicationdocument:7:3">
    <mRID>64e2af3a87c2404cbea80edc067a1b6f</mRID>
    <revisionNumber>1</revisionNumber>
    <type>A44</type>
    <sender_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</sender_MarketParticipant.mRID>
    <sender_MarketParticipant.marketRole.type>A32</sender_MarketParticipant.marketRole.type>
    <receiver_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</receiver_MarketParticipant.mRID>
    <receiver_MarketParticipant.marketRole.type>A33</receiver_MarketParticipant.marketRole.type>
    <createdDateTime>2024-10-07T14:36:40Z</createdDateTime>
    <period.timeInterval>
      <start>2024-10-05T22:00Z</start>
      <end>2024-10-06T22:00Z</end>
    </period.timeInterval>
      <TimeSeries>
        <mRID>1</mRID>
        <auction.type>A01</auction.type>
        <businessType>A62</businessType>
        <in_Domain.mRID codingScheme="A01">10YBE----------2</in_Domain.mRID>
        <out_Domain.mRID codingScheme="A01">10YBE----------2</out_Domain.mRID>
        <contract_MarketAgreement.type>A01</contract_MarketAgreement.type>
        <currency_Unit.name>EUR</currency_Unit.name>
        <price_Measure_Unit.name>MWH</price_Measure_Unit.name>
        <curveType>A03</curveType>
          <Period>
            <timeInterval>
              <start>2024-10-05T22:00Z</start>
              <end>2024-10-06T22:00Z</end>
            </timeInterval>
            <resolution>PT15M</resolution>
              <Point>
                <position>1</position>
                 <price.amount>55.35</price.amount>
              </Point>
              <Point>
                <position>5</position>
                 <price.amount>44.22</price.amount>
              </Point>
              <Point>
                <position>9</position>
                 <price.amount>40.32</price.amount>
              </Point>
              <Point>
                <position>13</position>
                 <price.amount>31.86</price.amount>
              </Point>
              <Point>
                <position>17</position>
                 <price.amount>28.37</price.amount>
              </Point>
              <Point>
                <position>21</position>
                 <price.amount>28.71</price.amount>
              </Point>
              <Point>
                <position>25</position>
                 <price.amount>31.75</price.amount>
              </Point>
              <Point>
                <position>29</position>
                 <price.amount>35.47</price.amount>
              </Point>
              <Point>
                <position>33</position>
                 <price.amount>37.8</price.amount>
[... 248 lines cut ...]
                <position>21</position>
                 <price.amount>86.85</price.amount>
              </Point>
              <Point>
                <position>22</position>
                 <price.amount>69.59</price.amount>
              </Point>
              <Point>
                <position>23</position>
                 <price.amount>57.42</price.amount>
              </Point>
              <Point>
                <position>24</position>
                 <price.amount>50</price.amount>
              </Point>
          </Period>
      </TimeSeries>
  </Publication_MarketDocument>
```

Notice that the 15M time intervals only give an update every 4 blocks, which actually results in "hourly" updates. The API states that if a certain point isn't supplied with CurveType A3, the previous block value is repeated for the missing block.

See page 10 of https://transparency.entsoe.eu/content/static_content/download?path=/Static%20content/knowledge%20base/entso-e-transparency-xml-schema-use-1-0.pdf&loggedUserIsPrivileged=false

Hope this helps.

Roeland54, 2024-10-07:
So the belgian data is a bug. But the api response format from entso-e has changed with an update last friday. So a fix to parse the new format is still needed to get it back up and running for other countries.

A few differences with the old format:

1.  There are now multiple 'period' tags that contain points.
2.  If the price is the same as the previous point. This point won't be part of the xml. In the example you can see for 2024-10-08 that position 13 is missing. More information on [the A3 curve type in this pdf](https://eepublicdownloads.entsoe.eu/clean-documents/EDI/Library/cim_based/Introduction_of_different_Timeseries_possibilities__curvetypes__with_ENTSO-E_electronic_document_v1.4.pdf) page 14.

This is a valid response for Portugal:
``` xml
<Publication_MarketDocument xmlns="urn:iec62325.351:tc57wg16:451-3:publicationdocument:7:3">
    <mRID>df5f07320f074c369d60428641fed537</mRID>
    <revisionNumber>1</revisionNumber>
    <type>A44</type>
    <sender_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</sender_MarketParticipant.mRID>
    <sender_MarketParticipant.marketRole.type>A32</sender_MarketParticipant.marketRole.type>
    <receiver_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</receiver_MarketParticipant.mRID>
    <receiver_MarketParticipant.marketRole.type>A33</receiver_MarketParticipant.marketRole.type>
    <createdDateTime>2024-10-07T16:18:12Z</createdDateTime>
    <period.timeInterval>
        <start>2024-10-06T22:00Z</start>
        <end>2024-10-07T22:00Z</end>
    </period.timeInterval>
    <TimeSeries>
        <mRID>1</mRID>
        <auction.type>A01</auction.type>
        <businessType>A62</businessType>
        <in_Domain.mRID codingScheme="A01">10YPT-REN------W</in_Domain.mRID>
        <out_Domain.mRID codingScheme="A01">10YPT-REN------W</out_Domain.mRID>
        <contract_MarketAgreement.type>A01</contract_MarketAgreement.type>
        <currency_Unit.name>EUR</currency_Unit.name>
        <price_Measure_Unit.name>MWH</price_Measure_Unit.name>
        <curveType>A03</curveType>
        <Period>
            <timeInterval>
                <start>2024-10-06T22:00Z</start>
                <end>2024-10-07T22:00Z</end>
            </timeInterval>
            <resolution>PT60M</resolution>
            <Point>
                <position>1</position>
                <price.amount>65.08</price.amount>
            </Point>
            <Point>
                <position>2</position>
                <price.amount>46.34</price.amount>
            </Point>
            <Point>
                <position>3</position>
                <price.amount>40</price.amount>
            </Point>
            <Point>
                <position>4</position>
                <price.amount>30.1</price.amount>
            </Point>
            <Point>
                <position>5</position>
                <price.amount>29.97</price.amount>
            </Point>
            <Point>
                <position>6</position>
                <price.amount>30.1</price.amount>
            </Point>
            <Point>
                <position>7</position>
                <price.amount>52.02</price.amount>
            </Point>
            <Point>
                <position>8</position>
                <price.amount>84.5</price.amount>
            </Point>
[... 142 lines cut ...]
            </Point>
            <Point>
                <position>20</position>
                <price.amount>79.9</price.amount>
            </Point>
            <Point>
                <position>21</position>
                <price.amount>90</price.amount>
            </Point>
            <Point>
                <position>22</position>
                <price.amount>83.19</price.amount>
            </Point>
            <Point>
                <position>23</position>
                <price.amount>46.08</price.amount>
            </Point>
            <Point>
                <position>24</position>
                <price.amount>33.1</price.amount>
            </Point>
        </Period>
    </TimeSeries>
</Publication_MarketDocument>
```

andig, 2024-10-07:
> But the api response format from entso-e has changed with an update last friday. 

Since the linked document s from 2022- is this change documented anywhere?

rwegnerGermany, 2024-10-07:
Maybe this helps: "All XML exports"....
[https://github.com/JaccoR/hass-entso-e/issues/184#issuecomment-2393716780](url)

andig, 2024-10-07:
Did a quick test. BE is broken as per Ensoe's docs. PT works fine afaikt. So... what's the issue?

rwegnerGermany, 2024-10-07:
I just switched back to entsoe, restarted the docker container and checked the GUI.
It is showing todays prices for tomorrow und no valid prices for now:
[image]

This is the entsoe response in trace log. Is maybe the splitting into 2 periods new?

```<?xml version="1.0" encoding="utf-8"?>
<Publication_MarketDocument xmlns="urn:iec62325.351:tc57wg16:451-3:publicationdocument:7:3">
    <mRID>938364a9beb64bcb9a7a689d16385b01</mRID>
    <revisionNumber>1</revisionNumber>
    <type>A44</type>
    <sender_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</sender_MarketParticipant.mRID>
    <sender_MarketParticipant.marketRole.type>A32</sender_MarketParticipant.marketRole.type>
    <receiver_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</receiver_MarketParticipant.mRID>
    <receiver_MarketParticipant.marketRole.type>A33</receiver_MarketParticipant.marketRole.type>
    <createdDateTime>2024-10-07T17:33:34Z</createdDateTime>
    <period.timeInterval>
        <start>2024-10-06T22:00Z</start>
        <end>2024-10-07T22:00Z</end>
    </period.timeInterval>
    <TimeSeries>
        <mRID>1</mRID>
        <auction.type>A01</auction.type>
        <businessType>A62</businessType>
        <in_Domain.mRID codingScheme="A01">10Y1001A1001A82H</in_Domain.mRID>
        <out_Domain.mRID codingScheme="A01">10Y1001A1001A82H</out_Domain.mRID>
        <contract_MarketAgreement.type>A01</contract_MarketAgreement.type>
        <currency_Unit.name>EUR</currency_Unit.name>
        <price_Measure_Unit.name>MWH</price_Measure_Unit.name>
        <classificationSequence_AttributeInstanceComponent.position>1</classificationSequence_AttributeInstanceComponent.position>
        <curveType>A03</curveType>
        <Period>
            <timeInterval>
                <start>2024-10-06T22:00Z</start>
                <end>2024-10-07T22:00Z</end>
            </timeInterval>
            <resolution>PT60M</resolution>
            <Point>
                <position>1</position>
                <price.amount>34.58</price.amount>
            </Point>
            <Point>
                <position>2</position>
                <price.amount>35.34</price.amount>
            </Point>
            <Point>
                <position>3</position>
                <price.amount>33.25</price.amount>
            </Point>
            <Point>
                <position>4</position>
                <price.amount>30.15</price.amount>
            </Point>
            <Point>
                <position>5</position>
                <price.amount>36.09</price.amount>
            </Point>
            <Point>
                <position>6</position>
                <price.amount>46.73</price.amount>
            </Point>
            <Point>
                <position>7</position>
                <price.amount>67.59</price.amount>
            </Point>
            <Point>
                <position>8</position>
                <price.amount>100.92</price.amount>
            </Point>
            <Point>
[... 133 lines cut ...]
            </Point>
            <Point>
                <position>17</position>
                <price.amount>101.19</price.amount>
            </Point>
            <Point>
                <position>18</position>
                <price.amount>126.34</price.amount>
            </Point>
            <Point>
                <position>19</position>
                <price.amount>156.77</price.amount>
            </Point>
            <Point>
                <position>20</position>
                <price.amount>168.2</price.amount>
            </Point>
            <Point>
                <position>21</position>
                <price.amount>114.41</price.amount>
            </Point>
            <Point>
                <position>22</position>
                <price.amount>94.9</price.amount>
            </Point>```

Roeland54, 2024-10-07:
This is the only[ official info ](https://transparency.entsoe.eu/news/widget?id=66f5203d792e84032cbb9b71)on the changes which is quite useless because it is lacking and explanation the xml format changes. 

But the response is definitely changed.

andig, 2024-10-08:
BZN|DE scheint nur für heute Daten zu haben, auch im XML. Für mich ist kein Fehler erkennbar.

closed 2024-10-08 by andig

andig, 2024-10-08:
> As this is not reflected in the [entsoe model](https://github.com/evcc-io/evcc/blob/master/tariff/entsoe/static.go#L107), only the last Period in a TimeSeries is read out by evcc.

I don‘t think so.

rwegnerGermany, 2024-10-08:
At least this somehow makes sense with a 24h data period shown/assigned to the wrong (later) day.

I also would expect `[]TimeSeriesPeriod` there.

And in the response it's weird that `period.timeInterval` start to end is only 24h, but 2 periods with 24h each are listed.

axel8viii, 2024-10-08:
> And in the response it's weird that `period.timeInterval` start to end is only 24h, but 2 periods with 24h each are listed.

As far as I remember, the interface was already behaving somewhat strangely in this respect before the change.
The difference, however, is that previously multiple TimeSeries with one period each were delivered and now one TimeSeries with multiple periods.

andig, 2024-10-09:
For Germany, A3 curve today has 24 data points. Nothing missing.

> At least this somehow makes sense with a 24h data period shown/assigned to the wrong (later) day.

I have not seen this error. Please open new issue.

Roeland54, 2024-10-09:
You need to validate this code against a entsoe response when the price for tomorrow is known. Before 13u00 the price of tomorrow will not be in the response.

andig, 2024-10-09:
```
From                   To                     Price/Cost
2024-10-10 00:00:00    2024-10-10 01:00:00    0.020
2024-10-10 01:00:00    2024-10-10 02:00:00    0.004
2024-10-10 02:00:00    2024-10-10 03:00:00    0.002
2024-10-10 03:00:00    2024-10-10 04:00:00    0.001
2024-10-10 04:00:00    2024-10-10 05:00:00    0.002
2024-10-10 05:00:00    2024-10-10 06:00:00    0.017
2024-10-10 06:00:00    2024-10-10 07:00:00    0.040
2024-10-10 07:00:00    2024-10-10 08:00:00    0.053
2024-10-10 08:00:00    2024-10-10 09:00:00    0.048
2024-10-10 09:00:00    2024-10-10 10:00:00    0.035
2024-10-10 10:00:00    2024-10-10 11:00:00    0.030
2024-10-10 11:00:00    2024-10-10 12:00:00    0.026
2024-10-10 12:00:00    2024-10-10 13:00:00    0.020
2024-10-10 13:00:00    2024-10-10 14:00:00    0.006
2024-10-10 14:00:00    2024-10-10 15:00:00    0.003
2024-10-10 15:00:00    2024-10-10 16:00:00    0.009
2024-10-10 16:00:00    2024-10-10 17:00:00    0.030
2024-10-10 17:00:00    2024-10-10 18:00:00    0.040
2024-10-10 18:00:00    2024-10-10 19:00:00    0.061
2024-10-10 19:00:00    2024-10-10 20:00:00    0.049
2024-10-10 20:00:00    2024-10-10 21:00:00    0.047
2024-10-10 21:00:00    2024-10-10 22:00:00    0.041
2024-10-10 22:00:00    2024-10-10 23:00:00    0.040
2024-10-10 23:00:00    2024-10-11 00:00:00    0.040
2024-10-11 00:00:00    2024-10-11 01:00:00    0.044
2024-10-11 01:00:00    2024-10-11 02:00:00    0.055
2024-10-11 02:00:00    2024-10-11 03:00:00    0.069
2024-10-11 03:00:00    2024-10-11 04:00:00    0.091
2024-10-11 04:00:00    2024-10-11 05:00:00    0.109
2024-10-11 05:00:00    2024-10-11 06:00:00    0.092
2024-10-11 06:00:00    2024-10-11 07:00:00    0.069
2024-10-11 07:00:00    2024-10-11 08:00:00    0.040
2024-10-11 08:00:00    2024-10-11 09:00:00    0.040
2024-10-11 09:00:00    2024-10-11 10:00:00    0.040
2024-10-11 10:00:00    2024-10-11 11:00:00    0.040
2024-10-11 11:00:00    2024-10-11 12:00:00    0.041
2024-10-11 12:00:00    2024-10-11 13:00:00    0.055
2024-10-11 13:00:00    2024-10-11 14:00:00    0.072
2024-10-11 14:00:00    2024-10-11 15:00:00    0.100
2024-10-11 15:00:00    2024-10-11 16:00:00    0.129
2024-10-11 16:00:00    2024-10-11 17:00:00    0.180
2024-10-11 17:00:00    2024-10-11 18:00:00    0.173
2024-10-11 18:00:00    2024-10-11 19:00:00    0.120
2024-10-11 19:00:00    2024-10-11 20:00:00    0.109
```

Works just fine

andig, 2024-10-09:
Although... it seems to be missing 4 hours. I'll double-check.