evcc-io/evcc #16533: Sofar Solar HYD 15K-3PH lädt sich nicht am Netz auf, wenn der minimale Preis erreicht ist

Frintrop, 2024-10-06:
### Describe the bug

Ich habe die experimentellen Features aktiviert und das Laden der Hausbatterie bei einem niedrigen Strompreis konfiguriert.
Sobald dieser erreicht wird, erscheint im Log

battery mode: modbus: response data size '18' does not match count '4'

und nichts weiter passiert.

### Steps to reproduce

1. Experimentelle Feature aktivieren
2. Strompreisgrenze hoch genug setzen, so dass evcc versucht die Batterie zu laden
3. GUI zeigt Versuch an
4. Fehler erscheint im Log
5. Batterie wird nicht geladen

### Configuration details

```yaml
Needed?
```

### Log details

```text
Log: ( site TRACE Modus) 
[circuit-house] DEBUG 2024/10/06 15:25:18 current: 29A
[site  ] ERROR 2024/10/06 15:25:22 battery mode: modbus: response data size '18' does not match count '4'
[site  ] DEBUG 2024/10/06 15:25:23 pv power: 5400W
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

andig, 2024-10-06:
Duplikat: https://github.com/evcc-io/evcc/issues/12220

andig, 2024-10-06:
Ist es eigentlich wirklich notwendig hier nochmal zu posten nachdem Du das andere Issue schon gefunden hattest?

closed 2024-10-06 by andig

Frintrop, 2024-10-07:
Nein ... der andere Stand auch schon auf Closed.

andig, 2024-10-07:
Das stimmt. Ohne Zugangsdaten wir es da leider auch nicht weiter gehen.