evcc-io/evcc #16540: Current always 0.0 under load management

denniswham, 2024-10-06:
### Describe the bug

Under Configuration - Integrations - Load Management, the current isn't calculated but always shows 0.0. For my grid meter I use the slimmemeter-v2, and reads current_phase_1, current_phase_2 and current_phase_3.
In the Load Management I would expect the sum of these three.

See the screenshot below. At the time of creating the screenshot, 
current_phase_1 = 1
current_phase_2 = 2
current_phase_3 = 0
[image]

 

### Steps to reproduce

1. Use Slimmetermeter-2 template for the meter
2. Goto Load Management under Configuration
3. Read the current for the main
4.
...

### Configuration details

```yaml
meters:
  - name: slimmelezer
    type: template
    template: slimmelezer-V2
    usage: grid
    host: 10.0.20.12 # IP address or hostname

circuits:
  - name: main # unique name, used as reference, e.g. as parent in other circuits
    title: Main Circuit # used in the UI
    maxcurrent: 40
    maxPower: 17000
    meter: slimmelezer # associated meter to monitor the power consumption (optional)
    parent: # no parent, this is the root circuit
```

### Log details

```text
N/a
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.130.13

andig, 2024-10-06:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-10-06 by andig

denniswham, 2024-10-06:
@andig I added the full log, can you re-open please?