evcc-io/evcc #16555: Plan Ladung nutzt gesperrte Batterie

Wakhangli, 2024-10-07:
### Describe the bug

Eine geplante Ladung nutzt die Hausbatterie, obwohl der Schalter 
"Verhindere Entladung im Schnell-Modus und bei geplantem Laden." aktiviert ist. 

### Steps to reproduce

1. Schalter ""Verhindere Entladung im Schnell-Modus und bei geplantem Laden." aktiv
2. Ladeplan eintragen und aktivieren
3. zusehen, wie nach Start der Ladung der Akku entladen wird obwohl "Battereie (gesperrt)" angezeigt wird.
4.
5.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 769ce22ae1afac961f01d6158032dde01da4e3f97f85d88dd7428d2a26e7f79f

interval: 30s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.100.86
  name: grid1
- type: template
  template: sma-hybrid
  id: 3
  host: 192.168.100.88
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: sma-hybrid
  id: 3
  host: 192.168.100.88
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3

chargers:
- type: template
  template: go-e-v3
  host: 192.168.100.85
  name: wallbox6

vehicles:
- type: template
  template: tesla-command
  title: I am a Teapot
  accessToken: xxx
  refreshToken: xxx
  vin: XXX
  capacity: 80
    #  minSoc: 0
  mode: pv
  phases: 3
    #  targetSoc: 90
  identifiers:
    - Y1
    - Y2
    - Y3
  icon: car
  name: ev4

- type: template
  template: nissan
  title: NCC1701E
  user: xxx
  password: XXX
  capacity: 62
    #  minSoc: 0
  mode: pv
    #  phases: 1
    #  targetSoc: 90
  identifiers:
    - X1
    - X2
  icon: car
  name: ev5

loadpoints:
- title: Ogog-e
  charger: wallbox6
  mode: pv
    #  phases: 1
    #  mincurrent: 6
    #  maxcurrent: 16
    #  resetOnDisconnect: true
  enable:
    delay: 1m
  disable:
    delay: 5m
site:
  title: PVBarm
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 100
  maxGridSupplyWhileBatteryCharging: 50

hems:
  type: sma
  AllowControl: false

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.3996
  feedin:
    type: fixed
    price: 0.082
```

### Log details

```text
Oct 07 15:44:43 hermes.local evcc[991039]: [main  ] INFO 2024/10/07 15:44:43 evcc 0.130.13
Oct 07 15:44:43 hermes.local evcc[991039]: [main  ] INFO 2024/10/07 15:44:43 using config file: /etc/evcc.yaml
Oct 07 15:44:43 hermes.local evcc[991039]: [db    ] INFO 2024/10/07 15:44:43 using sqlite database: /root/.evcc/evcc.db
Oct 07 15:44:43 hermes.local evcc[991039]: [main  ] INFO 2024/10/07 15:44:43 listening at :7070
Oct 07 15:44:45 hermes.local evcc[991039]: [semp  ] WARN 2024/10/07 15:44:45 SEMP_BASE_URL unspecified, using http://192.168.106.5:7070 instead
Oct 07 15:44:45 hermes.local evcc[991039]: [site  ] INFO 2024/10/07 15:44:45 site config:
Oct 07 15:44:45 hermes.local evcc[991039]: [site  ] INFO 2024/10/07 15:44:45   meters:      grid ✓ pv ✓ battery ✓
Oct 07 15:44:45 hermes.local evcc[991039]: [site  ] INFO 2024/10/07 15:44:45     grid:      power ✓ energy ✓ currents ✓
Oct 07 15:44:45 hermes.local evcc[991039]: [site  ] INFO 2024/10/07 15:44:45     pv 1:      power ✓ energy ✓ currents ✗
Oct 07 15:44:45 hermes.local evcc[991039]: [site  ] INFO 2024/10/07 15:44:45     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
Oct 07 15:44:45 hermes.local evcc[991039]: [site  ] INFO 2024/10/07 15:44:45   vehicles:
Oct 07 15:44:45 hermes.local evcc[991039]: [site  ] INFO 2024/10/07 15:44:45     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
Oct 07 15:44:45 hermes.local evcc[991039]: [site  ] INFO 2024/10/07 15:44:45     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
Oct 07 15:44:45 hermes.local evcc[991039]: [lp-1  ] INFO 2024/10/07 15:44:45 loadpoint 1:
Oct 07 15:44:45 hermes.local evcc[991039]: [lp-1  ] INFO 2024/10/07 15:44:45   mode:        pv
Oct 07 15:44:45 hermes.local evcc[991039]: [lp-1  ] INFO 2024/10/07 15:44:45   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
Oct 07 15:44:45 hermes.local evcc[991039]: [lp-1  ] INFO 2024/10/07 15:44:45   meters:      charge ✓
Oct 07 15:44:45 hermes.local evcc[991039]: [lp-1  ] INFO 2024/10/07 15:44:45     charge:    power ✓ energy ✓ currents ✓
Oct 07 15:44:46 hermes.local evcc[991039]: [lp-1  ] INFO 2024/10/07 15:44:46 car connected
Oct 07 15:44:46 hermes.local evcc[991039]: [lp-1  ] INFO 2024/10/07 15:44:46 start charging ->
Oct 07 15:44:46 hermes.local evcc[991039]: [lp-1  ] INFO 2024/10/07 15:44:46 vehicle updated: unknown -> I am a Teapot
Oct 07 15:45:16 hermes.local evcc[991039]: [lp-1  ] INFO 2024/10/07 15:45:16 stop charging <-
Oct 07 15:49:46 hermes.local evcc[991039]: [lp-1  ] INFO 2024/10/07 15:49:46 start charging ->
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.130.13

VolkerK62, 2024-10-07:
liegt an der Firmware vom SHM
https://github.com/evcc-io/evcc/discussions/15658

naltatis, 2024-10-07:
Ok, dann mach ich hier zu.

closed 2024-10-07 by naltatis