evcc-io/evcc #16559: Batterieentladesperre funktioniert nicht bei E3DC

Dirk77777, 2024-10-07:
### Describe the bug

Ich habe im Batteriedialog den Haken bei "Verhindere Entladung im Schnell-Modus und geplante Laden" gesetzt. Wenn das geplante Laden oder Laden im Modus Schnell startet, steht bei Batterie auch "Batterie (gesperrt) xx%" da. Die Batterie wird jedoch weiterhin entladen.
Das passiert bei einem Ladeplan (Abfahrt um x:xx Uhr), beim Laden mit Preisschwelle (laden wenn Preis <x ct) sowie im Modus Schnell.

### Steps to reproduce

1. Haken bei "Verhindere Entladung..." im Hausbatteriedialog setzen
2. Geplantes laden oder preisoptimiertes laden oder Modus "Schnell" einstellen
-> Batterie wird entladen

### Configuration details

```yaml
E3DC etc. wurde über das UI konfiguriert. Ich kann das yaml aber gerne nachreichen.
```

### Log details

```text
[main  ] INFO 2024/10/06 15:20:22 evcc 0.130.13
[main  ] INFO 2024/10/06 15:20:22 using config file: /etc/evcc.yaml
[db    ] INFO 2024/10/06 15:20:22 using sqlite database: /root/.evcc/evcc.db
[main  ] INFO 2024/10/06 15:20:23 listening at :7070
[site  ] INFO 2024/10/06 15:20:24 site config:
[site  ] INFO 2024/10/06 15:20:24   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/10/06 15:20:24     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/10/06 15:20:24     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/10/06 15:20:24     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/10/06 15:20:24   vehicles:
[site  ] INFO 2024/10/06 15:20:24     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2024/10/06 15:20:24 loadpoint 1:
[lp-1  ] INFO 2024/10/06 15:20:24   mode:        pv
[lp-1  ] INFO 2024/10/06 15:20:24   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/10/06 15:20:24   meters:      charge ✓
[lp-1  ] INFO 2024/10/06 15:20:24     charge:    power ✓ energy ✗ currents ✓
[site  ] DEBUG 2024/10/06 15:20:24 set buffer soc: 90
[site  ] DEBUG 2024/10/06 15:20:24 set buffer start soc: 0
[site  ] DEBUG 2024/10/06 15:20:24 set priority soc: 60
[site  ] DEBUG 2024/10/06 15:20:24 set battery discharge control: true
[lp-1  ] DEBUG 2024/10/06 15:20:25 set smart cost limit: 0.2
[lp-1  ] DEBUG 2024/10/06 15:20:25 phase timer inactive
[lp-1  ] DEBUG 2024/10/06 15:20:25 pv timer inactive
[lp-1  ] INFO 2024/10/06 15:20:25 vehicle updated: unknown -> Model Y
[lp-1  ] DEBUG 2024/10/06 15:20:25 set charge mode: pv
[site  ] WARN 2024/10/06 15:20:25 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/10/06 15:20:25 ----
[lp-1  ] DEBUG 2024/10/06 15:20:25 charge power: 11622W
[lp-1  ] DEBUG 2024/10/06 15:20:25 charge currents: [16.2 16.2 16.1]A
time="2024-10-06T15:20:25+02:00" level=info msg="Connecting to 192.168.1.16:5033"
time="2024-10-06T15:20:25+02:00" level=info msg="successfully connected to 192.168.1.16:5033"
time="2024-10-06T15:20:25+02:00" level=info msg="hiding auth request for security, use debug >= 99 to debug authentication"
time="2024-10-06T15:20:25+02:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
time="2024-10-06T15:20:25+02:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
[site  ] DEBUG 2024/10/06 15:20:25 pv power: 2862W
time="2024-10-06T15:20:25+02:00" level=info msg="Connecting to 192.168.1.16:5033"
time="2024-10-06T15:20:25+02:00" level=info msg="successfully connected to 192.168.1.16:5033"
time="2024-10-06T15:20:25+02:00" level=info msg="hiding auth request for security, use debug >= 99 to debug authentication"
time="2024-10-06T15:20:25+02:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
time="2024-10-06T15:20:25+02:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-10-06T15:20:25+02:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
[site  ] DEBUG 2024/10/06 15:20:25 battery soc: 43%
[site  ] DEBUG 2024/10/06 15:20:25 battery power: 0W
time="2024-10-06T15:20:25+02:00" level=info msg="Connecting to 192.168.1.16:5033"
time="2024-10-06T15:20:25+02:00" level=info msg="successfully connected to 192.168.1.16:5033"
time="2024-10-06T15:20:25+02:00" level=info msg="hiding auth request for security, use debug >= 99 to debug authentication"
time="2024-10-06T15:20:25+02:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
time="2024-10-06T15:20:25+02:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
[site  ] DEBUG 2024/10/06 15:20:25 grid meter: 9189W
[site  ] DEBUG 2024/10/06 15:20:25 site power: 9289W
[... 368 lines cut ...]
[site  ] DEBUG 2024/10/06 15:23:05 battery power: 4683W
time="2024-10-06T15:23:05+02:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
[site  ] DEBUG 2024/10/06 15:23:05 grid meter: 4786W
[site  ] DEBUG 2024/10/06 15:23:05 site power: 9569W
[lp-1  ] DEBUG 2024/10/06 15:23:05 charge voltages: [240 240 240]V
[lp-1  ] DEBUG 2024/10/06 15:23:05 detected connected phases: 3p
[lp-1  ] DEBUG 2024/10/06 15:23:05 detected active phases: 3p
[lp-1  ] DEBUG 2024/10/06 15:23:05 charger status: C
[lp-1  ] DEBUG 2024/10/06 15:23:05 soc estimated: 74.39% (vehicle: 74.00%)
[lp-1  ] DEBUG 2024/10/06 15:23:05 vehicle soc: 74%
[lp-1  ] DEBUG 2024/10/06 15:23:05 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/10/06 15:23:05 vehicle range: 295km
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.130.13

andig, 2024-10-07:
Mhhhm. Wieso schreibt es bei dir eine MaxDischargePower von 4,6kW bei Hold? Sehr seltsam.

Dirk77777, 2024-10-07:
Weiß nicht? :-) 
Ich habe testweise mal im E3DC-UI händisch eine Entladesperre eingerichtet, dann wird nicht entladen.

andig, 2024-10-08:
Du hast anscheinend 4,5kW `dischargelimit` konfiguriert. Da darfst Du Dich auch nicht wundern.

closed 2024-10-08 by andig

docolli, 2024-10-08:
Ich finde das Feature `dischargelimit` durchaus sinnvoll, da so während des Ladens des Autos noch weiterhin Energie aus dem Hausakku entnommen wird. Man kann sich das auf einen Wert setzen, der z.B. den normalen Hausverbrauch entspricht.
Wenn man einen Hausakku hat, dann sollte der idealerweise jeden Tag vollständig genutzt werden. Also über Tag auf 100% geladen und über Nacht (bis die PV am nächsten Tag wieder genug Leistung bringt) bis auf wenige Prozent Rest entladen werden.
Setzen wir die Entladeleistung des Hausakku generell auf 0, dann verbleibt (je nach Ladedauer des Autos) am nächsten Morgen ein größerer Rest im Hausakku und man kann an diesem Tag nicht mehr soviel selbst erzeugte PV Energie in den Hausakku speichern, wie wenn der Hausakku noch zusätzlich entladen worden wäre.

Das ganze Problem hier kommt aber doch eher von einer Fehlkonfiguration, die übers GUI gemacht wurde. Hier würde ich ansetzen und schauen, warum @Dirk77777 hier 4,5kW gesetzt hat. Eventuell muss man im GUI den Konfigurationspunkt besser erklären. Ich arbeite noch mit der evcc.yaml, aber wie sieht denn aktuell die Erstinstallation von evcc aus? Kommt da inzwischen ein GUI Dialog, über den man evcc einrichtet? Kommt da bereits die Frage nach dem `discharge limit`?

Dirk77777, 2024-10-08:
Danke für den Hinweis. Probiere ich aus, sobald das Auto wieder hier ist 

Zum Hintergrund bei mir: Ich habe kürzlich alle Konfigurationsoptionen, die es im UI gibt, aus dem yaml entfernt und über das UI neu eingerichtet. Weil ich es ausprobieren wollte, und weil ihr sagt, dass es in diese Richtung gehen wird (Konfig über UI, nicht Konfigfile).

Und ja, ich habe dort in den Advanced Options das Entladelimit auf 4500W gesetzt. Meine Interpretation dieser Option war, dass das nach oben begrenzt, also "max. 4500 entladen", und keine Mindestentladeleistung ist. 4500 ist in meinem Fall sinnlos, weil mein System eh nur maximal 4500 liefert. Aber ich dachte ich fülle es halt mal aus... 

Ich denke, im UI sollte die Beschreibung der Option verbessert werden,. z. B. "Wieviel soll die Batterie im Modus ... beitragen". Die momentane Beschreibung "Limitiert die Entladeleistung im 'Halten' Batterie Modus" klingt für mich nach Obergrenze.

docolli, 2024-10-08:
Deine Interpretation ist schon richtig, dieser Wert begrenzt die Entladeleistung des Hausakku auf den eingestellten Wert. Mehr Leistung liefert der Akku dann nicht, auch wenn das Haus mal 2000W braucht.

Also wenn du 1000W setzt und du lädst dein Auto über Nacht mit 6kW und dein Haus braucht 500W, dann wird (solange der Akku Ladung hat) 1000W aus dem Akku kommen, 500W braucht das Haus, die restlichen 500W gehen in die Wallbox und 5,5kW kommen aus dem Stromnetz. Lädst du 10h lang, dann sind das 10kWh Leistung aus dem Akku, von denen 5kWh für die Wallbox verwendet wurden.

Ich selber verwende aber den Haken bei "Verhindere Entladung im Schnell-Modus und geplante Laden" nicht, sondern nutze die im E3DC HKW vorhandene dynamische Ladepriorisierung.
[image]

So unterstützt der Hausakku die Wallbox voll (bis zu 4500W) bis zum eingestellten SOC (aktuell 75%, im Sommer gehen ich weiter runter ca. 40-50%). Danach regelt das HKW so, dass der Strom der Wallbox komplett aus dem Stromnetz kommt, aber der Hausverbrauch dynamisch weiterhin aus dem Hausakku kommt. Also wenn die Waschmaschine mal 2000W braucht, dann liefert der Akku das auch (im Gegensatz zum starren Wert in evcc).

Dirk77777, 2024-10-09:
Vielen Dank für die Erklärung, habs jetzt verstanden. Habe leider keine E3/DC-Wallbox (war mir zu teuer), daher helfen die Einstellungen im E3DC-UI bei mir nicht.
Deinen Wiki-Beitrag habe ich auch gelesen, sehr interessant. Auch interessante Links auf andere Projekte, schau ich mir bei Gelegenheit mal an. Es gibt immer was zu basteln :)