evcc-io/evcc #16563: 1 million mqtt messages within 5 minute

chiefjoejj, 2024-10-08:
### Describe the bug

My EVCC-installation pushes out a crazy amount of mqtt-messages which causes a high cpu load within my docker environment. Otherwise I wouldn't care because it is working - apart from the constant error messages telling me that there are "no message IDs availabe" because I guess my mqtt-broker is completely giving up... Tried a couple of modifications within my evcc.yaml but no luck so far to fix it. help, please! :)

### Steps to reproduce

1. start container
2. wait a couple of seconds
3. watch the mqtt messages skyrocket
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info

# unique installation id
plant: 1234567890123456789012345678901234567890123456789012345678xxxxxx

interval: 30 # control cycle interval

# trial token, valid until 2024-10-09
#sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
  - name: SolarEdge
    type: template
    template: victron-energy
    usage: pv
    host: 192.168.3.11 # IP address or hostname
    port: 502 # Port (optional)
#  - name: pv
#    type: custom
#    power: # Leistung (W)
#      source: mqtt
#      topic: victron/solaredge/power_w
#    energy: # Energie (kWh)
#      source: mqtt
#      topic: victron/solaredge/energy_kwh
#      scale: 1.000
#  - name: Stromnetz
#    type: template
#    template: victron-energy
#    usage: grid
#    host: 192.168.3.11 # IP address or hostname
#    port: 502 # Port (optional)
#    meterid: # Grid-Meter VRM instance- can be read out in VRM portal or via remoteUI. (optional)
  - name: Stromnetz
    type: custom
    power: # Leistung (W) - Forward
      source: mqtt
      topic: victron/grid/power_forward_w
      scale: -1
    energy: # Energie (kWh)
      source: mqtt
      topic: victron/grid/imported_kwh
      scale: 1
  - name: Hausspeicher
    type: template
    template: victron-energy
[... 290 lines cut ...]
[site  ] DEBUG 2024/10/08 10:21:00 pv power: 5229W
[site  ] DEBUG 2024/10/08 10:21:00 battery soc: 40%
[site  ] DEBUG 2024/10/08 10:21:00 battery power: -4022W
[site  ] DEBUG 2024/10/08 10:21:00 grid meter: 3W
[site  ] DEBUG 2024/10/08 10:21:00 site power: -3719W
[lp-2  ] DEBUG 2024/10/08 10:21:00 charge total import: 0.035kWh
[lp-2  ] DEBUG 2024/10/08 10:21:00 charger status: A
[site  ] DEBUG 2024/10/08 10:21:00 ----
[lp-1  ] DEBUG 2024/10/08 10:21:00 charge power: 0W
[lp-1  ] DEBUG 2024/10/08 10:21:00 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/10/08 10:21:00 charge power: 0W
[lp-2  ] DEBUG 2024/10/08 10:21:00 charge currents: [0 0 0]A
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.130.13

RenatusRo, 2024-10-08:
Are you sure about the health of your broker?
I've got this once every so many months, to be fixed by throwing away the broker and re-installing it.

chiefjoejj, 2024-10-08:
The broker is up and running - completely normal behaviour. All other clients connect perfectly without issues. Updated the mqtt-broker - same probleme as before. Tried a test-mqtt-broker - same probleme.

Only when I set the ip-adresses for the OpenWB Pros to nirvana then the probleme with millions of mqtt messages disappears. Updated the firmware of my two OpenWB Pros a few minutes ago but same issue...

andig, 2024-10-08:
Dein Broker hat einen Schluckauf und anscheinend funktioniert das Reconnect nicht, deshalb stauen die sich die Nachrichten. Leider kann ich das Reconnect-Issue (das wir auch schon in https://github.com/volkszaehler/mbmd/issues/346 gesehen) nicht reproduzieren. Ohne Reproducer gibts leider keinen Fix. Im Zweifel musst Du versuchen die Ursache lokal mit Deinem Broker zu identifizieren, ein kleines Go-Tool bauen und ggf. ein Ticket bei https://github.com/eclipse/paho.mqtt.golang auf machen. Ich hab leider keinen Anhaltspunkt wo ich suchen könnte.

chiefjoejj, 2024-10-08:
I tried another MQTT-broker in my network - the one hosted on my Venus OS installation. Same probleme here - My wild guess is still that it has to do something with the openWB Pros. As soon as EVCC connects to the wallboxes the probleme starts...

andig, 2024-10-08:
OWBPro has nothing to do with mqtt.

michiproep, 2024-10-10:
@chiefjoejj Can you see any other messages in your broker?
I also struggled with this on my docker environment until I found out it's a docker network issue.
e.g. running containers in different network modes (host, macvlan, etc.). In my case, I had to create 2 networks: macvlan for udp/tcp broadcasts and one for connecting between containers

chiefjoejj, 2024-10-11:
Thanks for all your input guys! In the end I was able to  solve the probleme. It had nothing to do with the docker environment and network modes but with my buggy config.yaml file. I completely started from scratch using the terminal-config-builder to create a minimal setup that allowed EVCC to start up and then I recursively modified/added the desired sections. After each step I checked the mqtt-behavior and after a perceived infinity all worked again perfectly. So glad I invested the time and got the issue fixed and can continue excess charging with the expected behaviour...

closed 2024-10-11 by chiefjoejj