evcc-io/evcc #16568: Embed the new RealTime Data from Solax Cloud in pv-meter-template "solax-inverter-cloud"

prinzartus, 2024-10-08:
I cannot get reasonable values for my configured PV-Meter using the template "solax-inverter-cloud"; 

I understand that solax-inverter-cloud does provide real-time-data as of lately. However the structure of the API might have been changed as well as the data-update-frequency. I think this change is not refelected in that evcc.template, hence I get unreasonable values.  

Be awarre, I just guessing, why I do not get correct values for my pv-meter, at all.

andig, 2024-10-08:
Es braucht Informationen zur Änderung. Ohne ist keine Umsetzung möglich.

VolkerK62, 2024-10-08:
the actual API document ist from 2022.
https://www.solaxcloud.com/user_api/SolaxCloud_User_Monitoring_API_V6.1.pdf

closed 2024-10-12 by andig