evcc-io/evcc #16569: Zwei oder mehrere Ladevorgänge (rechnerisch) zusammenführen - collate user selected seperate charging events()

prinzartus, 2024-10-08:
If charging events are interrrupted (by any means), it would be nice to be able to select - let's say three charging events - and collate them to a single charging event on the charging event screen. The "energie charged" and cost, duration to be summed up and the averages to be recalculated accordingly, just as if it was one uninterrupted charging event.

Maybe just during runtime only, while keeping the separate charging events in the db.

naltatis, 2024-10-09:
Yes, merging two entries should be possible without dataloss.

> Maybe just during runtime only, while keeping the separate charging events in the db.

If we do this, we should definitely merge both entries in the database once and not move this into runtime to reduce complexity. This one has not a high priority for me, but I see the value. I'm glad to support if someone wants to give it a try.

Steph-74, 2024-10-28:
If we talk about merge, we should also consider the possibility to split a session. When I unplug one car and plug in the other car before evcc gets an update from the wallbox (go-e Gemini), evcc stores everything in one session. I need to split the session manually in excel based on the log from the wallbox, as I get energy for one car reimbursed.