evcc-io/evcc #16575: Batterie "starte automatisch" - kleinere Schritte

michiproep, 2024-10-09:
Aktuell kann man die "starte automatisch Funktion" in 5% Schritten einstellen.
Ich würde eigentlich gerne die 100% wählen, was aber bei meinem SMA HomeStorage wegen dem bekannten "99 auf 100 Problem" keine gute Idee ist.
Ich hab vie API schon versucht auf 99% zu stellen, es wird aber auf 95% gesetzt.

Gäbe es eine Möglichkeit wenigstens via API, ohne Änderung der UI, die Ein-Prozent-Schritte zuzulassen?

Danke schon mal.

VolkerK62, 2024-10-09:
das ist ein reines Anzeigeproblem.
Du kannst z.B. 98% per API schicken. Wenn man per /api/state kontrolliert, ist es dort auch so eingetragen.
Lediglich im WebUI werden 95% angezeigt.
cc/ @naltatis

naltatis, 2024-10-09:
Ich hab hier jetzt mal die Anzeige gefixt ☝️. Damit zeigen wir immer den "echten" Wert an. Die Auswahloptionen in der UI würde ich erstmal so lassen.

michiproep, 2024-10-09:
Perfekt! danke

closed 2024-10-09 by michiproep